OpenIntelWireless/IntelBluetoothFirmware #369: AX201 on Z690 Alder Lake not supported

phantom-8, 2021-12-19:
**Description**
IntelFirmware failed to initialize due to "Intel version event size mismatch".  This is probably due some changes in the AX201 under the new Alder Lake CPU.

**Environment**
AX201 on MSI Pro Z690-A WIFI DDR4 m/b with Intel i7-12700K CPU running Monterey 12.0.1

- Kext Version:  IntelBluetoothFirmware v2.0.1, BlueToolFixup v2.6.1
- WiFi Card Model: AX201 bundled with m/b
- Product ID: 0x8087:0x0026
- macOS Version: Monterey 12.0.1

**Info in System Report - Bluetooth**

```
  Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Product ID:	0x0001
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)

```

**Logs**

```log
log show --last boot | grep IntelFir
< empty output >
```

**dmesg Logs**

```
[ 1315.747125]: IntelFirmware: Driver init()
[ 1315.747130]: IntelFirmware: Driver Probe()
[ 1315.747315]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[ 1315.747355]: IntelFirmware: Driver Start()
[ 1315.747403]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[ 1315.747406]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[ 1315.747414]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[ 1315.747433]: IntelFirmware: set configuration to 1
[ 1315.747716]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[ 1315.747719]: IntelFirmware: Found interface!!!
[ 1315.747727]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[ 1315.747728]: IntelFirmware: Found Interrupt endpoint!
[ 1315.748233]: IntelFirmware: Found Bulk out endpoint!
[ 1315.748715]: IntelFirmware: Found Bulk in endpoint!
[ 1315.749227]: IntelFirmware: BT init succeed
[ 1315.750272]: IntelFirmware: Intel version event size mismatch (act: 6, ver: 10)
[ 1315.750274]: IntelFirmware: Intel Read version failed
[ 1315.750275]: IntelFirmware: resetToBootloader
[ 1315.900787]: IntelFirmware: Clean up...
```

**Screenshots**

phantom-8, 2021-12-19:
Just compiled the IntelBluetoothFirmware from latest source date Dec 8, 2021 (aaf4247).  Same problem as the official v2.0.1 release.

zxystd, 2021-12-21:
Please disable all these drivers, do a fully shutdown, then boot to Windows or Linux, then reboot to macOS, to check if it is the same.

phantom-8, 2021-12-21:
I'm not sure if I understand completely.  Are the following procedures correct?

1) Boot into macOS, disable both IntelBluetoothFirmware.kext & BlueToolFixup.kext
2) Shutdown the PC, remove power cord and wait for a while
3) Boot to Windows or Linux.
4) Shutdown and reboot to macOS.
5) Check result.

Since both IntelBluetoothFirmware.kext & BlueToolFixup.kext are disabled, what's the purpose of checking the result after rebooting to macOS as these drivers aren't even loaded?

phantom-8, 2021-12-21:
I've performed the above testing.  The Bluetooth Info in system report is the same as previous.  As for dmesg log, there is no entry for InterFirmware as expected.  If I misunderstood your testing procedures, please let me know and I'll re-run them.

tonynesss, 2021-12-24:
same problem in my laptop

mdeimert, 2021-12-26:
Same problem here: Aorus Pro Wifi Z390

BCM_4350C2

Bluetooth Controller:
  Address: NULL
  State: Off.
  Chipset: BCM_4350C2
  Discoverable: Off.

vit9696, 2021-12-29:
It works on macOS 11. Just macOS 12 does not work from our testing.

williambj1, 2021-12-29:
> It works on macOS 11. Just macOS 12 does not work from our testing (master branch).

@vit9696 Sounds like an incompatibility issue with the new bluetoothd in Monterey with Bluetooth 5.2. Firmware for AX201 in this repo is downgraded to use Bluetooth 5.1 instead, make sure to fully shutdown the system if it was previously booted into another OS.

Not sure if newer batches that come with Z690 are using Bluetooth modules the same as AX210 (not able to downgrade to Bluetooth 5.1) though. Screenshot of Sysreport -> Bluetooth from Big Sur and logs from bluetoothd & IntelBluetoothFirmware in Monterey will help for identifying the issue.

The logs from @phantom-8 are quite odd and theoretically would not work for Big Sur as well.

---

> Since both IntelBluetoothFirmware.kext & BlueToolFixup.kext are disabled, what's the purpose of checking the result after rebooting to macOS as these drivers aren't even loaded?

I think @zxystd might forget to mention that this process should be done in macOS Big Sur. The firmware uploaded by another OS will stay in the device if a warm reboot is performed.

vit9696, 2021-12-29:
@williambj1 it was the chip that cannot be downgraded to BT 5.1 indeed, the chips were AX210 or something.

vit9696, 2021-12-29:
cc @vandroiy2013, could you please provide some logs on our Z690 build, just in case they happen to be useful?

mdeimert, 2021-12-29:
It started working for me after changing order of loading Kexts in Monterey.

All latest kexts for Lilu, Airport, Bluetooth, BlueToolFixup and Brcm

Now it displays as THIRD_PARTY_DONGLE in System raport and works great on 12.1 (Without airdrop - but this has never worked for me)

[image]

beelzebozox86, 2022-02-03:
> It started working for me after changing order of loading Kexts in Monterey.
> 
> All latest kexts for Lilu, Airport, Bluetooth, BlueToolFixup and Brcm
> 
> Now it displays as THIRD_PARTY_DONGLE in System raport and works great on 12.1 (Without airdrop - but this has never worked for me)
> 
> [image]

This is about z690, not some older SoC.

laggykiller, 2022-05-04:
I got the same issue with [MSI PRO B660M-A WIFI DDR4](https://www.msi.com/Motherboard/PRO-B660M-A-WIFI-DDR4). I am using i7-12700F and Monterey 12.3. The USB mapping is correct and the bluetooth internal USB device is detected.
The IntelFirmware log message repeats many time during boot before giving up and boot into desktop environment.
The dmesg is exactly the same.
I have even tried to boot Big Sur installation media, and I keep seeing error messages about bluetooth in installer log.

zxystd, 2022-05-04:
@laggykiller can you please upload the log or dmesg here?

laggykiller, 2022-05-04:
Since I got the same part of logs mentioned in the top post, I am providing my full dmesg log, ioreg and opencore log

Logs created while booting into Monterey 12.3

[dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621802/dmesg.txt)
[msi-pro-b660m-a-wifi-ddr4.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621803/msi-pro-b660m-a-wifi-ddr4.ioreg.txt)
[opencore-2022-05-04-145906.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621804/opencore-2022-05-04-145906.txt)

I will try to install Catalina and Big Sur on a spare hard drive, then upload the results

williambj1, 2022-05-05:
```log
[   70.853115]: IntelFirmware: Driver init()
[   70.853126]: IntelFirmware: Driver Probe()
[   70.853407]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[   70.853433]: IntelFirmware: Driver Start()
[   70.853477]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[   70.853481]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[   70.853492]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[   70.853497]: IntelFirmware: set configuration to 1
[   70.854131]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[   70.854136]: IntelFirmware: Found interface!!!
[   70.854149]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[   70.854152]: IntelFirmware: Found Interrupt endpoint!
[   70.854539]: IntelFirmware: Found Bulk out endpoint!
[   70.854931]: IntelFirmware: Found Bulk in endpoint!
[   70.855317]: IntelFirmware: BT init succeed
[   70.856395]: IntelFirmware: Intel version event size mismatch (act: 6, ver: 10)
[   70.856399]: IntelFirmware: Intel Read version failed
[   70.856401]: IntelFirmware: resetToBootloader
[   70.939553]: Sandbox: CommCenter(435) deny(1) mach-lookup com.apple.lsd.mapdbSystem Policy: CommCenter(435) deny(1) file-read-data /private/var/folders/kj/8_7f3ddj5md8x7vrzwmyf93r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSystem Policy: CommCenter(435) deny(1) file-read-metadata /private/var/folders/kj/8_7f3ddj5md8x7vrzwmyf93r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstore2 duplicate reports for System Policy: CommCenter(435) deny(1) file-read-metadata /private/var/folders/kj/8_7f3ddj5md8x7vrzwmyf93r0000gn/0/com.apple.LaunchServices.dv/com.apple.LaunchServices-3027-v2.csstoreSandbox: CommCenter(435) deny(1) mach-lookup com.apple.lsd.modifydb000070.997672 HS10@14a00000: AppleUSBHostPort::terminateDevice: destroying 0x8087/0026/0002 (IOUSBHostDevice): connect change interrupt
[   71.006858]: IntelFirmware: Clean up...
[   71.006862]: IntelFirmware: virtual void BtIntel::free()
[   71.006866]: IntelFirmware: virtual void USBDeviceController::free()
[   71.006890]: IntelFirmware: Driver Stop()
[   71.007021]: IntelFirmware: Driver free()
```

CC @zxystd 

> Since I got the same part of log mentioned in the top post, I am providing my full dmesg log, ioreg and opencore log
> 
> Logs created while booting into Monterey 12.3
> 
> [dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621802/dmesg.txt) [msi-pro-b660m-a-wifi-ddr4.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621803/msi-pro-b660m-a-wifi-ddr4.ioreg.txt) [opencore-2022-05-04-145906.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8621804/opencore-2022-05-04-145906.txt)
> 
> I will try to install Catalina and Big Sur on a spare hard drive, then upload the results

laggykiller, 2022-05-05:
For Catalina and Big Sur, I disabled BlueToolFixup.kext and added IntelBluetoothInjector.kext.
**Catalina**
[opencore-2022-05-05-042506-catalina-nopatch.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628788/opencore-2022-05-05-042506-catalina-nopatch.txt)
[dmesg-catalina-nopatch.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628789/dmesg-catalina-nopatch.txt)
[msi-pro-b660m-a-wifi-ddr4-catalina-nopatch.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628790/msi-pro-b660m-a-wifi-ddr4-catalina-nopatch.ioreg.txt)

**Big Sur**
[opencore-2022-05-05-044901-bigsur-nopatch.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628792/opencore-2022-05-05-044901-bigsur-nopatch.txt)
[dmesg-bigsur-nopatch.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628793/dmesg-bigsur-nopatch.txt)
[msi-pro-b660m-a-wifi-ddr4-bigsur-nopatch.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628794/msi-pro-b660m-a-wifi-ddr4-bigsur-nopatch.ioreg.txt)

---------------

I have also tried to add IntelBTPatcher.kext
**Catalina (With IntelBTPatcher.kext)**
[opencore-2022-05-05-043035-catalina-patched.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628795/opencore-2022-05-05-043035-catalina-patched.txt)
[dmesg-catalina-patched.txt
(https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628796/dmesg-catalina-patched.txt)
[msi-pro-b660m-a-wifi-ddr4-catalina-patched.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628797/msi-pro-b660m-a-wifi-ddr4-catalina-patched.ioreg.txt)

**Big Sur (With IntelBTPatcher.kext)**
[opencore-2022-05-05-043912-bigsur-patched.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628798/opencore-2022-05-05-043912-bigsur-patched.txt)
[dmesg-bigsur-patched.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628799/dmesg-bigsur-patched.txt)
[msi-pro-b660m-a-wifi-ddr4-bigsur-patched.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8628800/msi-pro-b660m-a-wifi-ddr4-bigsur-patched.ioreg.txt)
]

zxystd, 2022-05-05:
@laggykiller
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8631222/IntelBluetoothFirmware.kext.zip)
Please check this one.

laggykiller, 2022-05-05:
> @laggykiller [IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8631222/IntelBluetoothFirmware.kext.zip) Please check this one.

Tested on Monterey, bluetooth stil not working, but the error message has changed, plus IntelFirmware does not loop many times during boot.

[opencore-2022-05-05-201948.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635857/opencore-2022-05-05-201948.txt)
[msi-pro-b660m-a-wifi-dd4.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635858/msi-pro-b660m-a-wifi-dd4.ioreg.txt)
[dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635859/dmesg.txt)

```
$ sudo dmesg | grep IntelFirmware
[    2.225941]: IntelFirmware: Driver init()
[    2.225944]: IntelFirmware: Driver Probe()
[    2.225946]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[    2.225950]: IntelFirmware: Driver Start()
[    2.225962]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    2.225963]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    2.225967]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    2.226643]: IntelFirmware: set configuration to 1
[    2.226897]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    2.226901]: IntelFirmware: Found interface!!!
[    2.226907]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    2.226908]: IntelFirmware: Found Interrupt endpoint!
[    2.227075]: IntelFirmware: Found Bulk out endpoint!
[    2.227298]: IntelFirmware: Found Bulk in endpoint!
[    2.227497]: IntelFirmware: BT init succeed
[    2.230312]: IntelFirmware: Intel boot parameters size mismatch
[    2.230321]: IntelFirmware: Clean up...
[    2.230325]: IntelFirmware: virtual void BtIntel::free()
[    2.230329]: IntelFirmware: virtual void USBDeviceController::free()
[    2.230343]: IntelFirmware: Driver Stop()
[    2.230359]: IntelFirmware: Driver free()
```

laggykiller, 2022-05-05:
> @laggykiller [IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8631222/IntelBluetoothFirmware.kext.zip) Please check this one.

Tested on Monterey, bluetooth still not working, but the error message has changed, plus IntelFirmware does not loop many times during boot.

[msi-pro-b660m-a-wifi-ddr4.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635930/msi-pro-b660m-a-wifi-ddr4.ioreg.txt)
[opencore-2022-05-05-203157.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635931/opencore-2022-05-05-203157.txt)
[dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8635932/dmesg.txt)

```
$ sudo dmesg | grep IntelFirmware
[    2.122615]: IntelFirmware: Driver init()
[    2.122619]: IntelFirmware: Driver Probe()
[    2.122622]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[    2.122626]: IntelFirmware: Driver Start()
[    2.122640]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    2.122641]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    2.122646]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    2.123205]: IntelFirmware: set configuration to 1
[    2.123400]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    2.123403]: IntelFirmware: Found interface!!!
[    2.123410]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    2.123411]: IntelFirmware: Found Interrupt endpoint!
[    2.123554]: IntelFirmware: Found Bulk out endpoint!
[    2.123746]: IntelFirmware: Found Bulk in endpoint!
[    2.123892]: IntelFirmware: BT init succeed
[    2.126247]: IntelFirmware: Intel boot parameters size mismatch
[    2.126249]: IntelFirmware: Clean up...
[    2.126250]: IntelFirmware: virtual void BtIntel::free()
[    2.126251]: IntelFirmware: virtual void USBDeviceController::free()
[    2.126263]: IntelFirmware: Driver Stop()
[    2.126287]: IntelFirmware: Driver free()
```

zxystd, 2022-05-05:
@laggykiller Well, Intel mixed the gen2 and gen3 BT variants with the same USB id, need to refactor and write compat code.  Please wait a moment, I am busy right now.

zxystd, 2022-05-08:
@laggykiller please test this one.
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646278/IntelBluetoothFirmware.kext.zip)

laggykiller, 2022-05-08:
Boot freezes, but each time at different location
[image]
[image]
[image]
[image]

I have also tried using itlwm.kext instead of AirportItlwm.kext, but still freezes
[image]

zxystd, 2022-05-08:
@laggykiller
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646560/IntelBluetoothFirmware.kext.zip)
Little modifications.

laggykiller, 2022-05-08:
Getting closer, it successfully booted, and I can 'turn on' bluetooth. However, I cannot see and pair any devices, and it takes quite a long time to open the bluetooth setting page. It also seems like bluetooth does not want to be turned off (From GUI).

Here are the logs:

[dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646855/dmesg.txt)
[bluetoothd.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646856/bluetoothd.log)
[opencore-2022-05-08-104343.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646857/opencore-2022-05-08-104343.txt)
[ioreg.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8646858/ioreg.ioreg.txt)

zxystd, 2022-05-08:
@laggykiller You are using Monterey right? This chip must be BT5.2 and is not currently supported by Monterey, what the firmware kext's work is to upload the firmware to the BT chip. 

I suggest to test it on BigSur or Catalina.

laggykiller, 2022-05-09:
Tested to work in Big Sur (Using IntelBluetoothFirmware.kext + IntelBluetoothInjector.kext)

Here are the logs for reference

[dmesg-bigsur.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8648504/dmesg-bigsur.txt)
[ioreg-bigsur.ioreg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8648505/ioreg-bigsur.ioreg.txt)
[opencore-2022-05-09-023656-bigsur.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8648507/opencore-2022-05-09-023656-bigsur.txt)
[bluetoothd-bigsur.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/8648502/bluetoothd-bigsur.txt)

zxystd, 2022-05-09:
@laggykiller Ok, thank you! I will commit the changes tonight.

tanyangxf, 2022-05-21:
@zxystd Are there plans to support AX201,BT5.2 on Monterey?

MaydayAlaska, 2022-05-22:
So, if my understanding is correct, for the AX201 working on Monterey we just need apple to update the OS and making bt5.2 compatible, right?

laggykiller, 2022-05-22:
> So, if my understanding is correct, for the AX201 working on Monterey we just need apple to update the OS and making bt5.2 compatible, right?

But from what I know, AX200 (With bluetooth 5.2) works under Monterey?

MaydayAlaska, 2022-05-22:
> > So, if my understanding is correct, for the AX201 working on Monterey we just need apple to update the OS and making bt5.2 compatible, right?
> 
> But from what I know, AX200 (With bluetooth 5.2) works under Monterey?

I don't know, honestly. But does the AX201 work on other chipsets such as Z490 or z590? Or even older ones? Os the issue related to the wlan module or only to the z690 chipset?
Also, does tlit work on z690 chipsets but with older macOS versions like BigSur?

beelzebozox86, 2022-06-08:
Has the macOS Ventura beta bluetooth stack been looked at yet?

beelzebozox86, 2022-06-08:
Latest M2 MacBook specs for Wireless/BT:
Wi-Fi

802.11ax Wi-Fi 6 wireless networking
IEEE 802.11a/b/g/n/ac compatible
Bluetooth

Bluetooth 5.0 wireless technology

hit-zhimu, 2022-06-17:
> Latest M2 MacBook specs for Wireless/BT: Wi-Fi
> 
> 802.11ax Wi-Fi 6 wireless networking IEEE 802.11a/b/g/n/ac compatible Bluetooth
> 
> Bluetooth 5.0 wireless technology
in China main hackintosh forum pcbeta.com i just do not get any info about ax201 with bt5.2 `s 
compatibility with macOS Ventura, hope to get some results from global sites.
that`s quite bad news, b cause just bought a asus TUF b660m -wifi d4 motherboard & intel i5-12600KF to update my pc,  looks cant`t use for quite a while.

zxystd, 2022-06-17:
BT5.2 is also not supported yet on Ventura like on Monterey, it seems that the bluetooth stack are the same.

jamesst20, 2022-06-17:
> BT5.2 is also not supported yet on Ventura like on Monterey, it seems that the bluetooth stack are the same.

Thanks for the update and the hard work. 

May I ask if BT5.2 support has to come from Apple or if it's something you're planning to work on yourself? I'm also facing the same situation with a B660-i ASUS ITX Wifi Motherboard

marlonmsilva, 2022-06-23:
I haven't had any success either.
I have this stage.

Setup:
Monterey 12.4
Intel Core i5 12600k
Gigabyte Aero G Z690 DDR5
32GB RAM XPG Lancer DDR5 5200MHz
SSD SN570 1TB, NVMe M.2

Anyone with the same setup working bluetooth?

[image]
[image]

beelzebozox86, 2022-06-27:
> I haven't had any success either. I have this stage.
> 
> Setup: Monterey 12.4 Intel Core i5 12600k Gigabyte Aero G Z690 DDR5 32GB RAM XPG Lancer DDR5 5200MHz SSD SN570 1TB, NVMe M.2
> 
> Anyone with the same setup working bluetooth?
> 
> [image] [image]

Try Big Sur, not Monterey.

kyriosli, 2022-07-07:
So.. According to your update, this won't be fixed in a near future, till Apple supports 5.2 again? Can we replace the adapter with an older version of AX201, or use a USB external bluetooth adapter?

kyriosli, 2022-07-07:
> So.. According to your update, this won't be fixed in a near future, till Apple supports 5.2 again? Can we replace the adapter with an older version of AX201, or use a USB external bluetooth adapter?

Will ee1a80071 solve this problem?

williambj1, 2022-07-07:
> So.. According to your update, this won't be fixed in a near future, till Apple supports 5.2 again? Can we replace the adapter with an older version of AX201, or use a USB external bluetooth adapter?

All correct, alternatively you can consider using Big Sur.

> Will [ee1a800](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/ee1a800713b6719f2e1d9925df280f039df0431d) solve this problem?

Unfortunately it doesn't. The project is only a firmware uploader written as a kernel extension. The issue is caused by Apple's bluetoothd which runs in user-space.

zxystd, 2022-07-19:
@laggykiller @marlonmsilva @tanyangxf So now we are supposed to use BT5.2 and BT5.3 on Monterey/Ventura after patching the stack, you can now add this kext to your EFI and see if the BT is working. Also please update the `Lilu.kext` to the latest 1.6.2-beta one and the latest beta `IntelBluetoothFirmware.kext`. Waiting for yours testing result.
[IntelBTPatcher.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/9138493/IntelBTPatcher.kext.zip)

laggykiller, 2022-07-19:
@zxystd The kext works! I used the `IntelBTPatcher.kext` you provided here, `IntelBluetoothFirmware.kext` compiled from latest master, and `Lilu.kext` version 1.6.1

zxystd, 2022-07-20:
@laggykiller OK, thank you.

andreszerocross, 2022-07-20:
> @laggykiller OK, thank you.

May i ask?? 

So with new version of kext we only need IntelBtPatcher.kext + IntelBluetoothFirmware.kext??? No need for BluetoolFixup.kext???

And how about Big sur below?? Still we need IntelBluetoothInjector.kext? or just replace it with IntelBTPatcher.kext??

I ask because no documentation yet for this new kext

zxystd, 2022-07-20:
> May i ask??
>
> So with new version of kext we only need IntelBtPatcher.kext + IntelBluetoothFirmware.kext??? No need for BluetoolFixup.kext???
>
> And how about Big sur below?? Still we need [...]

In addition to inject the IntelBTPatcher.kext, extra usage are the same as before.

andreszerocross, 2022-07-20:
It doesn't work in Asus ROG Maximus Z690 Hero with Intel Intel Wifi 6E AX210 (I have set BluetoolFixup.kext + IntelBTPatcher.kext + IntelBluetoothFirmware.kext, all is latest commit version. I built it with XCode)

andreszerocross, 2022-07-20:
[image]

Another detail is, it's need more time to open Bluetooth Preferences (just this menu) then no any device is detected

jamesst20, 2022-07-23:
@zxystd 

IntelBtPatcher + latest lilu from sources + latest IntelBluetoothFirmware from sources works perfectly on  B660-i ASUS ITX Wifi Motherboard. 

Thanks a lot ! So happy to have native bluetooth!! :)

Attaching my kexts in case it allows more people to test

[Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/9172924/Kexts.zip)

ocean-bit, 2022-07-23:
@zxystd 
Thanks for your hard work!  Perfectly on MSI-PRO-Z690-A-WIFI-D4-12700K-Monterey 12.5.
I wonder which time would you release the next version of both **IntelBluetoothFirmware** and the **IntelBTPatcher**？
 We could only build from source or fetch from this page now.

[image]

Thanks again, just **niubi** to you.

andreszerocross, 2022-07-23:
> @zxystd
> 
> IntelBtPatcher + latest lilu from sources + latest IntelBluetoothFirmware from sources works perfectly on B660-i ASUS ITX Wifi Motherboard on Monterrey 12.4
> 
> Thanks a lot ! So happy to have native bluetooth!! :)
> 
> Attaching my kexts in case it allows more people to test
> 
> [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/9172924/Kexts.zip)

You didn't use BluetoolFixup.kext???

andreszerocross, 2022-07-23:
It looks still problem in AX210 (and good for AX201)

williambj1, 2022-07-23:
@andreszerocross Hi, can you attach logs from IntelBTPatcher and bluetoothd?

Logs from IntelBTPatcher requires DebugEnhancer to be installed first, then run  `sudo dmesg | grep BTPatcher > ~/Desktop/BTPatcher.log`
Logs from bluetoothd can be obtained by running `log show --last boot | grep bluetoothd > ~/Desktop/bluetoothd.log`

james-em, 2022-07-23:
> You didn't use BluetoolFixup.kext???

Yes I did, it's included in the zip.

jamesst20, 2022-07-23:
> You didn't use BluetoolFixup.kext???

Yes I did it's included in the zip

beelzebozox86, 2022-07-23:
@williambj1  WOW, this is some accomplishment!!!

Installed the kexts from the zip provided and Bluetooth is almost 100% working on my z690-AERO-G-DDR4.

I tested it with a Logitech T651 trackpad.  
Handoff works as well.
What doesn't work is Airdrop.  It will find my iPhone but file transfers don't work atm.  My iPhone also doesn't see my computer in Airdrop.  

Logs and IOREG attached.
[Archive.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/9173954/Archive.zip)

Excellent work!

LawrenceUp, 2022-08-05:
> @zxystd
>
> IntelBtPatcher + latest lilu from sources + latest IntelBluetoothFirmware from sources works perfectly on B660-i ASUS ITX Wifi Motherboard on Monterrey 12.4
>
> Thanks a lot ! So happy to have [...]

@jamesst20 Thanks for your share.
I successfully drive my bluetooth.

@zxystd 感谢作者的无私付出 

ASUS TUF GAMING Z690-PLUS DDR5 WIFI 
Intel Wi-Fi 6E AX211 160MHz
[image]

499978920, 2022-08-06:
睡眠唤醒后有一定几率无法连接，需要重开关蓝牙才可以，比如刚刚睡眠一晚上。无法连接时 launchd，bluetoohd 会占用很高 CPU

Thamizhan, 2022-08-07:
> 

I did used same Kexts but My BT is not fully working. It doesn't show scan results for my Realme Bud Q and Mivi Roam 2 Headset. But shows my Samsung F62 smart phone and I can connect to the phone and transfer files. Only issues with the headphones or speakres. All 3 devices are on Bluetooth 5.0.

Mobo: GIgabyte Z690 UD AX with Wi-Fi 6E AX211 and Bluetooth 5.2

[5 more replies]

closed 2022-08-31 by williambj1