OpenIntelWireless/IntelBluetoothFirmware #372: AX210 not working on Monterey

MartB, 2021-12-25:
**Description**
See title, seems like the card is just "offline" but the wifi switch is enabled, as my wifi works correctly.

**Environment**

- Kext Version: 2.0.1 & 2.1.0 RC1
- WiFi Card Model: Intel AX210
- Product ID: 0x8087 0x0032
- macOS Version: 12.1

**Info in System Report - Bluetooth**
```
Bluetooth-HCI:

  Produkt-ID:	0x0032
  Hersteller-ID:	0x8087  (Intel Corporation)
  Version:	0.00
  Geschwindigkeit:	Bis zu 12 MBit/s
  Standort-ID:	0x14300000 / 1
  Verfügbare Stromstärke (mA):	500
  Erforderliche Stromstärke (mA):	100
  Zusätzlicher Betriebsstrom (mA):	0
  Integriert:	Ja
```

**Logs**
```log
---
```

**Screenshots**
[image]
[image]

williambj1, 2021-12-26:
@MartB 

Unfortunately, the new bluetoothd in Monterey does not support Bluetooth 5.2. Unlike other cards, there is no way to downgrade AX210's Bluetooth to 5.1.

However, the old IOBluetooth drivers from Big Sur and older is compatible with Bluetooth 5.2.

Right now there are only two options available:

1. Purchase a USB Bluetooth dongle and use Monterey
2. Downgrade the system to Big Sur or older

Hope this helps

MartB, 2021-12-26:
@williambj1 
Thanks you for this explanation! 
Worthy of having this in the FAQ somewhere, i absolutely thought im doing something wrong.

Do you think the AX200 would work?
(I got pretty much all recentish intel cards just trying to find the "best")

closed 2021-12-26 by MartB

dnlzzxz, 2022-03-26:
> @MartB
>
> Unfortunately, the new bluetoothd in Monterey does not support Bluetooth 5.2. Unlike other cards, there is no way to downgrade AX210's Bluetooth to 5.1.
>
> However, the old IOBluetooth drivers [...]

Thank you very much for the explanation now at least I have some peace of mind knowing that is not me the culprit. So for now the only move is to buy a broadcom chip...

LanderlYoung, 2022-03-27:
> 

Same thought here. And is did simple file replace from AX200 to AX210.

Found the file name relations here.
AX210 >> ibt-0041-0041.sfi
AX200 >> ibt-20-0-3.sfi
AX200 >> ibt-20-1-3.sfi
AX200 >> ibt-20-1-4.sfi

Sad thing is that does not work, the Bluetooth icon on the System menu bar becomes a turned-off state.
Didn't know if my operation is wrong.

Jiagu1218, 2022-06-22:
@williambj1 My AX210 Bluetooth is working, after system sleep.Does it help to fix it
[image]

[image]

alexnfsc175, 2022-06-27:
Hi @Jiagu1218 , could you share your configuration, what kexts are you using and in what order?

Jiagu1218, 2022-07-11:
@alexnfsc175 That is my configuration. Using [AirportItlwm](https://github.com/OpenIntelWireless/itlwm/releases) ,[IntelBluetoothFirmware](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/releases)  and follow this
Make sure IntelBluetoothFirmware.kext v1.1.3 or newer is used.
Exclude the injection of IntelBluetoothInjector.kext for macOS Monterey from the Bootloader's config
Include BlueToolFixup.kext from [acidanthera/BrcmPatchRAM](https://github.com/acidanthera/BrcmPatchRAM)
[image]

aleixsr, 2023-10-08:
Fully confirmed, AX210 works fine on Ventura but doesn't on Monterey (checked both macOS using same EFI)