OpenIntelWireless/IntelBluetoothFirmware #373: Bluetooth firmware will not upload on Intel AC7265 (0x8087, 0x0a2a)

Overc1ocker, 2022-01-07:
-MacOS Monterey (this issue was present on older macOS versions as well)
-Verified kext is loaded via kextstat

How to reproduce:
-Shut computer down completely
-Attempt to boot MacOS. Check system profiler for bluetooth device.

Result: No bluetooth device shows up. Bluetooth does not work.

How to verify this kext is at fault:
-Power up computer, boot into linux
-After Linux boots, REBOOT the computer
-Attempt to boot MacOS. Check system profiler for bluetooth device.
-Intel Bluetooth device shows up; devices connect via bluetooth normally

------------------------------------------------
Note: In MacOS monterey, the intel bluetooth device seems to lock up the IOUSB kext, causing slow bootup and unusable USB ports on initial boot. A partial fix is to put the computer to sleep, then wake it back up to fix the USB ports and bluetooth. (This assumes you have rebooted from linux). This most likely has nothing to do with this kext, but I wanted to add this to see if anyone has experienced it.

zxystd, 2022-01-07:
Attach logs would be fine.

Overc1ocker, 2022-01-14:
There are no logs to attach: Kext will not load AT ALL if Linux is not loaded first.

If Linux is loaded first, then MacOS Monterey via a reboot

```
$kextstat |grep IntelBluetooth  
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
  115    0 0                  0x6d5000   0x6d5000   com.zxystd.IntelBluetoothFirmware (2.1.0) EB72D6E1-12D7-3035-8E82-2EC10ED3F7A9 <24 7 6 3>
```

If MacOS Monterey is loaded first.

```
kextstat |grep IntelBluetooth  
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
```
Same behavior with BlueToolFixup.

When booting MacOS first, Bluetooth HCI is also missing from "System Information"

Overc1ocker, 2022-01-14:
IOreg: Cold boot (I booted directly into macos without rebooting from linux
[ioreg-cold-boot.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7871672/ioreg-cold-boot.log)

IOreg: Linux (I booted MacOS after using linux to initialize the intel bluetooth chip.)

[ioreg-boot-after-linux.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7871679/ioreg-boot-after-linux.log)
Note: I also remapped my usb ports and verified that the port the bluetooth chip is on is present in the USBmap.kext

I'm using a HP DA0014dx laptop

williambj1, 2022-01-15:
@Overc1ocker Looks like the bluetooth device is completely dead if a cold boot is performed, USB port mapping looks good though. Does bluetooth work if IntelBluetoothFirmware is removed? (IntelBluetoothFirmware only patches the ram on 1st gen devices, they should already have a working builtin BT4.0 firmware)

Urberzelda, 2022-01-16:
Intel AX210 bluetooth not worming either :)

Overc1ocker, 2022-02-01:
No. It doesn't seem to. The card remains dead

Overc1ocker, 2022-02-01:
I just made some ssdt edits completely unrelated to USB or bluetooth (I redid my EC ssdt and added the brightness keys kext.) Now bluetooth doesnt work at all. Still works fine in linux though. Kext does not load and no bluetooth HCI shows.

Overc1ocker, 2022-02-01:
Ok. My laptop is a confusing mess. 
-- Intel Bluetooth firmware will not load if I do not boot linux first. Otherwise "bluetooth HCI" does not even show up.

--My EC (embedded controller) cannot be enabled if I want bluetooth to work. If I enable it using a SSDT, "bluetooth HCI" does not show up and kext does not load

So 2 things cannot be true if I want this kext to work
1. No cold booting
2. EC cannot be enabled (which is almost a requirement for a laptop)

Overc1ocker, 2022-02-03:
I will make another post to explain the issue. Turns out I had some things wrong

closed 2022-02-03 by Overc1ocker