OpenIntelWireless/IntelBluetoothFirmware #378: bluetoothd crashes after wake from sleep - Mac OS Monterey 12.1

RTHPJM, 2022-01-23:
**Description**
I have noticed that Apple's bluetoothd will crash after a sleep/wake cycle. Sometimes will restart okay, sometimes it will continue to crash.

If bluetoothd recovers, then Mac OS will operate as expected with devices attached.

Symptoms when bluetoothd has crashed and will not recover include:
 - no devices attached (or registered)
 - System Preferences - Bluetooth panel fails to load
 - Menu Bar - Bluetooth gadget toggle switch has no effect
 - Typically more crash reports appear in the log files, viewed via the Console app.

A hard reboot (shutdown, and power off at the power supply, wait for residual power to drain, aka full power-down) will recover the situation. 

**Environment**

(I have redacted some of the UIDs...)

- Kext Version: com.zxystd.IntelBluetoothFirmware (2.1.0) with as.acidanthera.BlueToolFixup (2.6.1)
- WiFi Card Model: Intel® Wireless-AC 9260
- Product ID: 0x0025
- macOS Version: 12.1

**OpenCore Version:**
REL-077-2022-01-10

Kernel Extensions:
Name (Version)
as.vit9696.Lilu (1.5.9)
as.vit9696.AppleALC (1.6.8)
org.acidanthera.NVMeFix (1.0.9)
as.acidanthera.BlueToolFixup (2.6.1)
as.vit9696.RestrictEvents (1.0.6)
wtf.spinach.AMDRyzenCPUPowerManagement (0.7)
as.vit9696.VirtualSMC (1.2.8)
wtf.spinach.SMCAMDProcessor (1)
com.zxystd.AirportItlwm (2.1.0)
com.insanelymac.RealtekRTL8111 (2.4.2)
com.zxystd.IntelBluetoothFirmware (2.1.0)
org.virtualbox.kext.VBoxDrv (6.1.30)
org.virtualbox.kext.VBoxUSB (6.1.30)
org.virtualbox.kext.VBoxNetFlt (6.1.30)
org.virtualbox.kext.VBoxNetAdp (6.1.30)

**System Information**
Hardware Overview:

  Model Name:	Mac Pro
  Model Identifier:	MacPro7,1
  Processor Name:	AMD Ryzen 9 3900X 12-Core Processor            
  Processor Speed:	3.8 GHz
  Number of Processors:	1
  Total Number of Cores:	12
  L2 Cache (per Core):	512 KB
  L3 Cache:	64 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
  System Firmware Version:	1715.60.5.0.0
  OS Loader Version:	540.60.2~89

 Motherboard: ASUS TUF GAMING X570-PLUS (WI-FI)

**Info in System Report - Bluetooth**

```
Bluetooth Controller:
  Address:	__:__:__:__:__:F6
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v256 c256
  Product ID:	0x0001
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
  Paired Bluetooth Devices:
  MX Master:
  Address:	__:__:__:__:__:BD
  Connected:	Yes
  Vendor ID:	0x046D
  Product ID:	0xB01E
  Firmware Version:	MPM12.10_0005
  Minor Type:	Mouse
  Services:	0x400000 < BLE >

```

**Logs**

```log
N/A
```

**Kernel Panic Logs**

```
N/A
```

**Screenshots**

N/A

** Crash Report**
[bluetoothd-2022-01-23-104744.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/7920337/bluetoothd-2022-01-23-104744.txt)

williambj1, 2022-01-23:
Thank you for the report. Unfortunately we have absolutely zero control since we are only uploading the firmware. Although this doesn't make much sense, it would be better to report this to Apple.

closed 2022-01-23 by williambj1

RTHPJM, 2022-01-23:
Thanks williambj1,

I sort of knew this would be the case. I wondered if the issue might be higher in the stack, maybe bluetoolfixup from the Acidanthera team. I thought I should report here first, but I’m happy to have the issue closed here, I take the report forward to bluetoolfixup.

Thanks for you quick response.