OpenIntelWireless/IntelBluetoothFirmware #379: intel n2230 Wi-Fi可以 蓝牙无法工作

wei134102, 2022-01-27:
请问大家，intel n2230 Wi-Fi可以 蓝牙无法工作，怎么查看kext是否加载呢？
下面是我的报告！

Mac  ~  sudo dmesg | grep Bluetooth
[IOBluetoothFamily][start] -- completed
**** [IOBluetoothHostControllerUSBTransport][start] -- completed -- result = TRUE -- 0x2800 ****
[IOBluetoothFamily][staticBluetoothTransportShowsUp] -- Received Bluetooth Controller register service notification -- 0x2800
[IOBluetoothHostController][start] -- completed
PMRD: joinAggressiveness IOBluetoothHostControllerUSBTransport 0xfd56b94bfe1f4ac1
[IOBluetoothHostController::setConfigState] calling registerService
**** [IOBluetoothFamily][ProcessBluetoothTransportShowsUpActionWL] -- calling IOBluetoothFamily's registerService() -- 0xfcf0 -- 0x8000 -- 0x2800 ****
**** [IOBluetoothFamily][ProcessBluetoothTransportShowsUpActionWL] -- Connected to the transport successfully -- 0xfcf0 -- 0x8000 -- 0x2800 ****
**** [IOBluetoothHostController][RequestPowerStateChangeFromUserSpace] -- ON -> ON (thread: 0x5a40)
**** [IOBluetoothHostControllerUSBTransport][RequestTransportPowerStateChange] -- entering -- ON -> ON -- this = 0x2800 ****
**** [IOBluetoothHostControllerUSBTransport][RequestTransportPowerStateChange] -- exiting -- result = SUCCESS -- this = 0x2800 ****
**** [IOBluetoothHostController][RequestPowerStateChangeFromUserSpace] -- ON -> ON (thread: 0x5a40) -- returning 0x0000 (kIOReturnSuccess)
failed to lookup target storage class: 2Sandbox: mdwrite(446) System Policy: allow file-write-xattr /System/Library/PreferencePanes/Bluetooth.prefPanefailed to lookup target storage class: 2ACM: deleteCredentialSet: Deleting credential set, CS[104].
failed to lookup target storage class: 2failed to lookup target storage class: 2Sandbox: 1 duplicate report for mdwrite allow file-write-xattr /System/Library/PreferencePanes/Bluetooth.prefPaneSandbox: swcd(393) deny(1) ipc-posix-shm-read-data /tmp/com.apple.csseed.118failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2failed to lookup target storage class: 2Sandbox: suggestd(401) deny(1) file-read-data /Users/wei/Library/Preferences/.GlobalPreferences.plistSandbox: swcd(393) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: SCIM_Extension(371) deny(1) mach-lookup com.apple.proactive.PersonalizationPortrait.QuickTypeSandbox: parsecd(312) deny(1) mach-lookup com.apple.lsd.mapdbSandbox: parsecd(312) deny(1) file-write-create /private/var/folders/wv/k_1b34jn7cs35xhp_wmyy97h0000gn/0/com.apple.parsecdSandbox: parsecd(312) deny(1) mach-lookup com.apple.lsd.modifydbSandbox: 20 duplicate reports for parsecd deny(1) mach-lookup com.apple.lsd.modifydbSandbox: parsecd(312) deny(1) mach-lookup com.apple.lsd.mapdbSandbox: 4 duplicate reports for parsecd deny(1) mach-lookup com.apple.lsd.mapdbSandbox: commerce(320) deny(1) file-read-data /Users/wei/Library/Preferences/MobileMeAccounts.plistSandbox: softwareupdated(456) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox: fmfd(322) deny(1) file-write-data /Users/wei/Library/Caches/com.apple.icloud.fmfd/Cache.db-walSandbox: suggestd(401) deny(1) file-read-data /Users/wei/Library/Preferences/com.apple.CoreDuet.plistPMRD: assertionsUser 0x40

wei134102, 2022-01-27:
追加 好像没有加载上，怎么处理呢
 wei@weis-iMac  ~  kextstat | grep -v "com.apple"
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   38    3 0xffffff7f82fef000 0x27000    0x27000    as.vit9696.Lilu (1.5.9) B051298C-0FFC-32A7-B0D6-CB0DE0B793C0 <8 6 5 3 2 1>
   39    0 0xffffff7f83016000 0x5000     0x5000     as.lvs1974.DebugEnhancer (1.0.5) F399B689-9363-3801-B8B5-5F71BD203F83 <38 8 6 5 3 2 1>
   41    0 0xffffff7f8301f000 0x182000   0x182000   as.vit9696.AppleALC (1.6.8) F804EAD7-3E72-3B4F-B46F-12EB34C22D91 <38 13 8 6 5 3 1>
   42    0 0xffffff7f831a1000 0x11000    0x11000    org.netkas.driver.FakeSMC (1800) FC7031CC-C949-35D1-AC1A-CA8DB349793E <12 8 6 5 3 1>
   43    0 0xffffff7f831b2000 0x4000     0x4000     com.rehabman.driver.USBInjectAll (0.7.1) A5767EA0-3760-3131-B996-28EE85139CF5 <13 12 5 3>
   44    0 0xffffff7f831b7000 0x64000    0x64000    as.vit9696.WhateverGreen (1.2.9) DA17A3EC-EE26-3298-B1C3-9975F0E50513 <38 13 8 6 5 3 2 1>
   60    0 0xffffff7f8321b000 0x14000    0x14000    com.insanelymac.IntelMausiEthernet (2.5.0d0) 302385A6-DCF6-3D8D-898A-160607F4EFCA <16 13 6 5 3 1>
   67    0 0xffffff7f83264000 0x18000    0x18000    net.osx86.kexts.GenericUSBXHCI (1.2.11) no UUID <66 13 8 6 5 3>

zxystd, 2022-01-27:
蓝牙的USB产品ID是多少?在系统信息-USB里面可以看到，目测这款蓝牙是csr的，自带固件，不需要firmware驱动，只需要injector，firmware驱动没加载正常。

wei134102, 2022-01-27:
蓝牙 HCI：

  产品 ID：	0x07da
  厂商 ID：	0x8087  (Intel Corporation)
  版本：	78.69
  速度：	最高可达 12 Mb/秒
  位置 ID：	0x1d120000 / 2
  可用电流 (mA)：	500
  所需电流 (mA)：	0
  额外的操作电流 (mA)：	0

wei134102, 2022-01-27:
USB 2.0 总线：

  主控制器驱动器：	AppleUSBEHCIPCI
  PCI 设备 ID：	0x1c26 
  PCI 修订版 ID：	0x0005 
  PCI 供应商 ID：	0x8086 

集线器：

  产品 ID：	0x0024
  厂商 ID：	0x8087  (Intel Corporation)
  版本：	0.00
  速度：	最高可达 480 Mb/秒
  位置 ID：	0x1d100000 / 1
  可用电流 (mA)：	500
  所需电流 (mA)：	0
  额外的操作电流 (mA)：	0

蓝牙 HCI：

  产品 ID：	0x07da
  厂商 ID：	0x8087  (Intel Corporation)
  版本：	78.69
  速度：	最高可达 12 Mb/秒
  位置 ID：	0x1d120000 / 2
  可用电流 (mA)：	500
  所需电流 (mA)：	0
  额外的操作电流 (mA)：	0
请帮看看，怎么处理一下，谢谢 ！

zxystd, 2022-01-27:
只加载injector就可以了

wei134102, 2022-01-27:
> 只加载injector就可以了
马上试试。这个型号可以用IntelBluetoothInjector.kext吗？

wei134102, 2022-01-27:
> 只加载injector就可以了

无法搜索到设备，手机有一次搜到黑苹果的蓝牙mac地址 不显示名称，链接不上，后来搜不到了！

closed 2022-01-27 by wei134102