OpenIntelWireless/IntelBluetoothFirmware #381: macOS Monterey 无法使用蓝牙

hello8693hello, 2022-02-02:
**Description**
RT，按照文档的说明开启了蓝牙，可是在系统信息里usb的地方没有看见蓝牙设备，虽然系统偏好设置里的蓝牙能进去，但是打不开。冷启动没用。我的机器是NUC8i5BEH，使用的是 @weachy 大佬的EFI（OC 0.7.4）。

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: Intel® Wireless-AC 9560
- Product ID: 这个我也不知道，如果说是黑果模拟的机型，那就是mac mini8,1
- macOS Version: 12.2

**Info in System Report - Bluetooth**
NOTHING

**Logs**
none

**Screenshots**
[image]

[image]

williambj1, 2022-02-02:
Map USB ports.

closed 2022-02-02 by williambj1

hello8693hello, 2022-02-02:
咋整啊，求大佬指教，本人黑苹果小白

CenYunhan, 2022-07-15:
要定制USB 自行到b站搜索教程

CenYunhan, 2022-07-15:
网卡是不可能驱动蓝牙的 macOS不支持蓝牙5.2且ax210不能降级蓝牙版本 现在只有两个选择：要么降级到Big Sur 要么换ax200 如果有机会 那就等苹果官方支持蓝牙5.2