OpenIntelWireless/IntelBluetoothFirmware #382: [Solution] Bluetooth device missing when booting macos but present in other OSes

Overc1ocker, 2022-02-03:
Recently I had an issue where the intel bluetooth device or "Bluetooth HCI" would not show up in macOS. The result is that the IntelBluetoothFIrmware.kext would not load and `kextstat` confirmed this. This is not a kext problem. It's most likely an issue with your machine. So before blaming the kext, check these 3 things!

1. All USB ports must be properly [mapped](https://dortania.github.io/OpenCore-Post-Install/usb/)
2. If you still don't see the bluetooth device, some firmwares block it unless you are booting a known OS, such as WINDOWS or LINUX. Therefore you must tell your laptop you are booting a OS it recognizes. Use` [SSDT-XOSI](https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/manual.html#osi-to-xosi)` to do this.
3. If you are Monterey, you must NOT have intel bluetooth firmware injector kext enabled. It WILL cause problems!!

1509098778, 2022-02-04:
intel ax210 bluetooth is still not working on macos 12.2
[image]
[image]
[image]
[image]
[image]

1509098778, 2022-02-04:
hope the following link helps why ax210 bluetooth is not working
[IntelBluetoothFirmware/issues/372#issuecomment-1001098124](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/372#issuecomment-1001098124)