OpenIntelWireless/IntelBluetoothFirmware #383: IntelBluetoothFirmware.kext not working on ax210 on monterey on x570 amd

v01der, 2022-02-10:
i tried IntelBluetoothFirmware.kext with BlueToolFixup.kext as per the guide i also have maped usb, the bluetooth HCI is showing in hardware-->USB with a valid product id and vendor id and location id with the both kexts and without the both kexts, tried after cold boot and tried on 12.2 plus 12.3 beta 1 and beta 2 same result for ax210 tried to get issue by the comand sudo dmesg | grep IntelFirmware w/w-out DebugEnhancer.kext no out from terminal with the said command,
[image]
can do some test if u want

LanderlYoung, 2022-02-10:
Me too.

I went through the doc https://openintelwireless.github.io/IntelBluetoothFirmware/FAQ.html#what-does-this-kext-do

but can't figure out how to `Exclude the injection of IntelBluetoothInjector.kext for macOS Monterey from the Bootloader's config`, don't know if it's related.

williambj1, 2022-02-10:
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/372#issuecomment-1001098124

closed 2022-02-10 by williambj1