OpenIntelWireless/IntelBluetoothFirmware #384: handoff works only one way

Infinity-1729, 2022-02-24:
**Description**

my Mac is able to receive handoff data like clipboard from my iPhone but not able to send handoff data to iPhone,
if i copy something from iPhone and pasting into Mac works fine
but if I copy something in Mac and and pasting into iPhone doesn't work
same with the iMessage or safari or any other app
if I open in iPhone it pops up in my mac
but not the other way around
it's the problem with BlueToolFixup.kext
because I tried in Big Sur with with IntelBluetoothInjector.kext
everything works like a charm

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: intel(R) Wireless-AC 9650 160MHz
- Vendor ID:0x8087
- Product ID:0x0aaa
- macOS Version:12.2.1 

**Info in System Report - Bluetooth**

```
 Bluetooth Controller:
  Address:	24:EE:9A:E9:34:1D
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	On
  Firmware Version:	v256 c256
  Product ID:	0x0001
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

alansleep, 2022-02-27:
I agree, same happens with Asus PCE-AX3000

Infinity-1729, 2022-03-01:
> I agree, same happens with Asus PCE-AX3000

inform me if you find out any solution for this; for now, I m waiting for the update of kext

williambj1, 2022-03-03:
This issue has existed for nearly 1 year and it's fair to assume that Apple has intentionally disabled continuity on Tx for third party bluetooth. Unfortunately there is nothing we can do with a firmware uploader.

closed 2022-03-03 by williambj1