OpenIntelWireless/IntelBluetoothFirmware #385: Stuck on boot till the card is disconnected/reconnected

Thatisjigen, 2022-03-01:
I'm using latest opencore and Airportiwlan+Intelbluetooth firmware, bluetooth is working without the kext but isn't toogleable so I've added it to avoid wifi+bt known issues
Btw once intelbluetooth firmware is enabled it stucks on boot and I have to disable it (pressing the airplane mode button) to manage to boot, once booted all is working as expected.
If required I can send logs obviously
HW: Zbook G1 15, Intel Centrino Advanced-N 6235

williambj1, 2022-03-03:
@Thatisjigen Please provide a picture with `-v` enabled when the issue happens.

Thatisjigen, 2022-03-05:
[image]
Here you have, I can't find nothing useful tho

xiaoyan0113, 2022-05-04:
hello,I use same Wireless card(Intel Centrino Advanced-N 6235)
USB ID is:**0x8087,0x07DA**
I checked instructions and find this USB ID not Compatibility.
Please Compatibility this ID,Thx a lot.

williambj1, 2022-05-05:
Might be related to https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/aaf42472824865f553eeb7e17c7fa1c024da1305

zxystd, 2022-05-08:
@Thatisjigen For Intel Centrino Advanced-N 6235, the bluetooth chip doesn't need to be patched or download firmware, so the `IntelBluetoothFirmware.kext` is not needed, what you need is the `IntelBluetoothInjector.kext`, how did you load the kexts?

Thatisjigen, 2022-05-09:
> @Thatisjigen For Intel Centrino Advanced-N 6235, the bluetooth chip doesn't need to be patched or download firmware, so the `IntelBluetoothFirmware.kext` is not needed, what you need is the `IntelBluetoothInjector.kext`, how did you load the kexts?

Hi, yeah, firmware is not needed on Big Sur, but that way bluetooth is not toggleable, that's all.
Adding the firmware the behaviour is described above ( chips needs to be shut down/up with the airplane button to pass the macos loading screen)

I'm loading the kext via opencore, no firmware injection

On Monterey, with firmware injection kext the issue disappears at all

I know that functionality is still there and toggling it on or off isn't that of a big deal, still if I can help to throubleshoot the cause of why it behaves like that when a firmware is loaded I'll be happy to provide logs and infos.

vincerutabaga, 2022-05-21:
It seems that my ax201 bluetooth card stop working after monterey 12.4 update (gigabyte z490).

Thatisjigen, 2022-07-22:
> @Thatisjigen @xiaoyan0113 is it fixed?

I got no idea, my laptop died last month (rip) so I moved to a thinkpad and here it's fine (Monterey)

closed 2023-06-09 by zxystd