OpenIntelWireless/IntelBluetoothFirmware #386: In opencore 0.6.7, the driver cannot start the system, and the interface is stuck at the user avatar.

xinjiajuan, 2022-03-09:
**Description**

**Environment**

- Kext Version: 2.1.0
- WiFi Card Model: Intel AX210
- Product ID: 
- macOS Version: BigSur 11.6.4

In the old version of the driver, after clearing the opencore startup cache, the bluetooth is normal. After restarting again, the bluetooth device cannot be found.
After replacing the 2.1.0 version driver, it cannot be turned on and stops at the user's avatar

[image]**Kernel Panic Logs**

xinjiajuan, 2022-03-09:
There are no binaries in IntelBluetoothjector.kext not so leave it blank

williambj1, 2022-03-09:
Use other sources for EFI config support.

closed 2022-03-09 by williambj1

xinjiajuan, 2022-03-10:
> Use other sources for EFI config support.
It works fine today. Maybe IntelBluetoothjector.kext has no binary file, but the old version has it. In the configuration efi file, the original configuration is still retained, and the binary configuration is removed.thanks