OpenIntelWireless/IntelBluetoothFirmware #402: bluetooth not working ax201 b660

meowwThecat, 2022-07-01:
### Have you read the docs?

Yes

### macOS Version

12.4

### Kext Version

2.1.0 / bluetoolfix: 2.6.1

### Wireless Adapter Model and USB Product ID

vendorID=0x8087, productID=0x0026

### Description

**bluetooth not working**  wifi works 

**config**

                    1. 12400
                    2. gig 660ds3 ax201
                    3. 32 gb 2666
                    4. 1 tb 970 evo plus
                    5. 6600xt

> i checked may times before purchasing all the parts , so its quite frustrating that bluetooth of mine is not working.
(and theres only 1 issue about this ax201 model one can easily overlook :( )

seeking your kind help 

**about mac info ::**

[image]

**about mac:: usb info:** 

"bluetooth hcl" appears all time without any bluetooth kext, if i  include intelbluetoothfirmware and bluetool fixup kext, all the shenanigans's start (looping same verbose msg over and over again and also sleep breaks ,(in hackintool : iousbflash device flashing constantly)) . i will post the usb config later below.

 **inlcuded kexts** 

[image]

**kext load order** 

[image]

**kextstat output** 

[image]

**usb map  info** 
[image]

### Info in System Report - Bluetooth

```shell
boot verbose mode shows  same msg many time :

[  304.363364]: IntelFirmware: Clean up...
[  304.363373]: IntelFirmware: virtual void BtIntel::free()
[  304.363377]: IntelFirmware: virtual void USBDeviceController::free()
[  304.363449]: IntelFirmware: Driver Stop()
[  304.363499]: IntelFirmware: Driver free()
[  304.591067]: IntelFirmware: Driver init()
[  304.591076]: IntelFirmware: Driver Probe()
[  304.591079]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[  304.591085]: IntelFirmware: Driver Start()
[  304.591102]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[  304.591104]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[  304.591112]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[  304.591668]: IntelFirmware: set configuration to 1
[  304.592031]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[  304.592035]: IntelFirmware: Found interface!!!
[  304.592045]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[  304.592046]: IntelFirmware: Found Interrupt endpoint!
[  304.592415]: IntelFirmware: Found Bulk out endpoint!
[  304.592808]: IntelFirmware: Found Bulk in endpoint!
[  304.593177]: IntelFirmware: BT init succeed
[  304.597104]: IntelFirmware: Intel version event size mismatch (act: 6, ver: 10)
[  304.597108]: IntelFirmware: Intel Read version failed
[  304.597109]: IntelFirmware: resetToBootloader
[  304.748339]: IntelFirmware: Clean up...
[  304.748357]: IntelFirmware: virtual void BtIntel::free()
.......................................
```

### Relevant log output

```shell
sudo dmesg | grep -i IntelBluetooth
Password:
[   52.484849]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  141.883997]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  190.936049]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  202.081773]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  239.194480]: IntelBluetoothFirmware::handlePowerDomainDidChangeTo not in power tree
[  356.459892]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  409.107041]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  413.724779]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  424.419756]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  430.553157]: IntelBluetoothFirmware::handlePowerDomainDidChangeTo not in power tree
[  585.682203]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  642.878359]: IntelBluetoothFirmware::handlePowerDomainDidChangeTo not in power tree
[  670.925542]: IntelBluetoothFirmware::handlePowerDomainWillChangeTo not in power tree
[  766.941470]: IntelBluetoothFirmware::handlePowerDomainDidChangeTo not in power tree
```

williambj1, 2022-07-02:
Looks like you have one of those weird OEM customized AX201 models with AX201's Wi-Fi and AX210's gen3 bluetooth. Unfortunately this means the minimal version of bluetooth available is 5.2 and is only functional in macOS Big Sur or earlier as Apple messed up support for Bluetooth 5.2 starting from Monterey. See #369 for more details.

The bluetooth firmware upload issue should be fixed in the recent commits and the compiled binary is available in the artifacts section from the latest Ci run.

Since this is a duplicate of #369 and we prefer to have the same issue discussed in one place, I will close this for now. Feel free to continue the discussion in #369.

williambj1, 2022-07-02:
Duplicate of #369

closed 2022-07-02 by williambj1