OpenIntelWireless/IntelBluetoothFirmware #403: Add support for 0x7FF8 or also known as BCM4350C2

RyzenDew, 2022-07-02:
### Have you read the docs?

Yes

### macOS Version

macos 12

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

0x7FF8 AX200

### Description

looking to see if you could add support for my devices
i am willing to get any info you need from within my hack

### Info in System Report - Bluetooth

```shell
Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	UART
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
will grab soon
```

closed 2022-07-03 by williambj1

RyzenDew, 2022-07-03:
This isnt done my device does not work at all so confused at least add notes or something

Space-Clustercompany, 2022-07-03:
where did you get the v2.2.0