OpenIntelWireless/IntelBluetoothFirmware #405: 9560NGW didn't work on Monterey 12.4

whitestarlau, 2022-07-11:
### Have you read the docs?

Yes

### macOS Version

macOS 12.4

### Kext Version

v2.1.0

### Wireless Adapter Model and USB Product ID

intel 9560AC 0x0aaa

### Description

I have load IntelBluetoothFirmware.kext and BlueToolFixup.kext in my opencore config,and I make usb port mapping.But I still cannot drive my bluetooth.And the wireless is work very well.

### Info in System Report - Bluetooth

```shell
In USB,the HCI is corret:
蓝牙HCI：

  产品ID：	0x0aaa
  厂商ID：	0x8087  (Intel Corporation)
  版本：	0.02
  速度：	最高可达12 Mb/秒
  位置ID：	0x14800000 / 1
  可用电流(mA)：	500
  所需电流(mA)：	100
  额外的操作电流(mA)：	0
  内建：	是

In bluetooh,it is not ok:

  蓝牙控制器：
  地址：	NULL
  状态：	关闭
  芯片组：	BCM_4350C2
  可被发现：	关闭
  固件版本：	v0 c0
  产品ID：	0x0001
  支持的服务：	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  传输：	UART
  供应商ID：	0x004C (Apple)
```

### Relevant log output

```shell
whitelau@whitedeiMac ~ % sudo dmesg | grep IntelFirmware
Password:
[    0.883129]: IntelFirmware: Driver init()
[    0.883134]: IntelFirmware: Driver Probe()
[    0.883137]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
[    0.883147]: IntelFirmware: Driver Start()
[    0.883160]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    0.883163]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    0.883170]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    0.883828]: IntelFirmware: set configuration to 1
[    0.884195]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    0.884199]: IntelFirmware: Found interface!!!
[    0.884209]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    0.884212]: IntelFirmware: Found Interrupt endpoint!
[    0.884404]: IntelFirmware: Found Bulk out endpoint!
[    0.884608]: IntelFirmware: Found Bulk in endpoint!
[    0.884758]: IntelFirmware: BT init succeed
[    0.885757]: IntelFirmware: Firmware revision 0.1 build 20 week 53 2019
[    0.885874]: IntelFirmware: Found device firmware ibt-17-16-1.sfi 
[    0.889708]: IntelFirmware: firmwareVersion
[    0.889740]: IntelFirmware: Boot Address: 0x40800
[    0.889742]: IntelFirmware: Firmware Version: 20-53.19
[    0.889744]: IntelFirmware: Firmware already loaded
[    0.889797]: IntelFirmware: Frimware is already running, finishing
[    0.890776]: IntelFirmware: Clean up...
[    0.890779]: IntelFirmware: virtual void BtIntel::free()
[    0.890782]: IntelFirmware: virtual void USBDeviceController::free()
```

whitestarlau, 2022-07-11:
[image]
[image]
[image]

whitestarlau, 2022-07-11:
Any thing help?

whitestarlau, 2022-07-12:
Now I Modify my usb mapping,however the buletooth still not work.But I got longer log,like this:

> [    0.872163]: IntelFirmware: Driver init()
> [    0.872169]: IntelFirmware: Driver Probe()
> [    0.872172]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0AAA
> [    0.872180]: IntelFirmware: Driver Start()
> [    0.872195]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
> [    0.872198]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
> [    0.872205]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
> [    0.872866]: IntelFirmware: set configuration to 1
> [    0.873246]: IntelFirmware: virtual bool USBDeviceController::findInterface()
> [    0.873251]: IntelFirmware: Found interface!!!
> [    0.873268]: IntelFirmware: virtual bool USBDeviceController::findPipes()
> [    0.873271]: IntelFirmware: Found Interrupt endpoint!
> [    0.873457]: IntelFirmware: Found Bulk out endpoint!
> [    0.873634]: IntelFirmware: Found Bulk in endpoint!
> [    0.873824]: IntelFirmware: BT init succeed
> [    0.874550]: IntelFirmware: Bootloader revision 0.1 build 42 week 52 2015
> [    0.875532]: IntelFirmware: Device revision is 2
> [    0.875535]: IntelFirmware: Secure boot is enabled
> [    0.875536]: IntelFirmware: OTP lock is enabled
> [    0.875538]: IntelFirmware: API lock is enabled
> [    0.875539]: IntelFirmware: Debug lock is disabled
> [    0.875541]: IntelFirmware: Minimum firmware build 1 week 10 2014
> [    0.875668]: IntelFirmware: Found device firmware ibt-17-16-1.sfi 
> [    0.879544]: IntelFirmware: firmwareVersion
> [    0.879577]: IntelFirmware: Boot Address: 0x40800
> [    0.879579]: IntelFirmware: Firmware Version: 20-53.19
> [    0.879581]: IntelFirmware: send firmware header
> [    0.879966]: IntelFirmware: send firmware header done
> [    0.879970]: IntelFirmware: send firmware pkey
> [    0.880595]: IntelFirmware: send firmware pkey done
> [    0.880599]: IntelFirmware: send firmware signature
> [    0.881432]: IntelFirmware: send firmware signature done
> [    0.881435]: IntelFirmware: send firmware payload
> [    2.165555]: IntelFirmware: send firmware payload done
> [    2.166535]: IntelFirmware: Notify: Firmware download done
> [    2.179543]: IntelFirmware: Notify: Device reboot done
> [    2.179549]: IntelFirmware: Found device firmware ibt-17-16-1.ddc 
> [    2.179555]: IntelFirmware: Load DDC config: ibt-17-16-1.ddc 14
> [    2.182519]: IntelFirmware: Load DDC config done
> [    2.183526]: IntelFirmware: Read debug features done
> [    2.184533]: IntelFirmware: Set debug features done
> [    2.185524]: IntelFirmware: Firmware revision 0.1 build 20 week 53 2019
> [    2.186525]: IntelFirmware: Clean up...
> [    2.186528]: IntelFirmware: virtual void BtIntel::free()
> [    2.186530]: IntelFirmware: virtual void USBDeviceController::free()

zxystd, 2022-07-13:
Are you using the latest BlueToolFixup.kext?
If confirm, maybe you can try this new kext
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/9100881/IntelBluetoothFirmware.kext.zip)

closed 2022-07-13 by whitestarlau