OpenIntelWireless/IntelBluetoothFirmware #406: Bluetooth not working in 2.2 beta with Monterey

peddy22, 2022-08-02:
### Have you read the docs?

Yes

### macOS Version

12.5 Monterey on 2.2 RC

### Kext Version

v2.20

### Wireless Adapter Model and USB Product ID

Intel AX210 (8086)

### Description

   59    0 0                  0xb000     0xb000     com.zxystd.IntelBTPatcher (2.2.0) F7AF8FB7-D140-3524-BC56-EA82168AAB48 <52 24 9 7 6 3 2 1>

main driver does not appear to load
blue tool does not appear to load
usb inject all has been used

kernel messages
`sudo dmesg | grep -i bluetooth             
[    1.374871]: **** [IOBluetoothFamily][start] -- completed successfully for HIDShim
**** [BroadcomBluetooth20703USBTransport][init] -- entering ****
[    9.980218]: **** [BroadcomBluetooth20703USBTransport][init] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.980239]: **** [BroadcomBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.980258]: **** [CSRBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][init] -- entering ****
[    9.980276]: **** [IOBluetoothHostControllerPCIeTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][init] -- entering ****
[    9.980294]: **** [IOBluetoothHostControllerTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[    9.980311]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.980328]: **** [IOBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothPacketLogger][init] -- entering ****
[    9.980345]: **** [IOBluetoothPacketLogger][init] -- exiting -- success ****
**** [IOBluetoothSerialManager][init] -- entering ****
[    9.980361]: **** [IOBluetoothSerialManager][init] -- exiting -- success ****
**** [BroadcomBluetooth20703USBTransport][start] -- entering ****
[   10.210179]: **** [BroadcomBluetooth20703USBTransport][start] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][start] -- entering ****
[   10.210191]: **** [BroadcomBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][start] -- entering ****
[   10.210204]: **** [CSRBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][start] -- entering ****
[   10.210218]: **** [IOBluetoothHostControllerPCIeTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][start] -- entering ****
[   10.210242]: **** [IOBluetoothHostControllerTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[   10.210255]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerUSBTransport][start] -- entering ****
[   10.210266]: **** [IOBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [IOBluetoothPacketLogger][start] -- entering ****
[   10.210275]: **** [IOBluetoothPacketLogger][start] -- exiting -- success ****
**** [IOBluetoothSerialManager][start] -- entering ****
[   10.210288]: **** [IOBluetoothSerialManager][start] -- exiting -- success ****
[   11.572288]: **** [IOBluetoothFamily][newUserClient] -- created userclient for HIDShim
[   11.572359]: **** [IOBluetoothFamily][AddHCIEventNotification] -- Added notification port for HIDShim
[   11.572416]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created input shared memory for HIDShim
[   11.572459]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created output shared memory for HIDShim
[   11.715376]: Sandbox apply: bluetoothd[136] <bytes>AppleKeyStore:7456:136: unexpected session: 100000 uid: -1 requested by: 136
[   11.733358]: Sandbox: bluetoothd(136) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox apply: secinitd[288] <bytes>setVIRTUAL_IF_CREATE role=1, bsd_name=, mac=0e:b3:bc:21:c9:06, unk1=0
[   11.791403]: 1 duplicate report for Sandbox: bluetoothd(136) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: com.apple.Ambien(286) deny(1) mach-lookup com.apple.windowserver+ IOAudioEngine[<ptr>]::init(0)
[... 5 lines cut ...]
[   45.222011]: 3 duplicate reports for Sandbox: bluetoothd(136) deny(1) mach-lookup com.apple.network.EAPOLControllerSandbox: searchpartyd(264) deny(1) mach-lookup com.apple.timed.xpcuipc_accept: peer disconnected unp_gencnt 94itlwm: setSCANCACHE_CLEAR
[   73.364967]: 1 duplicate report for Sandbox: bird(425) deny(1) file-read-data /Library/Application Support/CrashReporter/DiagnosticMessagesHistory.plistSandbox: ContextStoreAgen(462) deny(1) mach-lookup com.apple.SystemConfiguration.configdSandbox: passd(534) deny(1) mach-lookup com.apple.CoreLocation.agentSandbox: passd(534) deny(1) mach-lookup com.apple.server.bluetooth.le.att.xpcSandbox: passd(534) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistAppleSSE::hasBridgeSSE - SSE not present (neither KR nor SEPMgr)
[  166.133720]: 2 duplicate reports for Sandbox: accessoryupdater(647) allow mach-lookup com.apple.commcenter.coretelephony.xpcSandbox: bluetoothd(136) deny(1) mach-lookup com.apple.network.EAPOLControlleritlwm: getSYNC_ENABLED
[  166.828470]: Sandbox apply: netbiosd[651] <bytes>1 duplicate report for Sandbox: bluetoothd(136) deny(1) mach-lookup com.apple.network.EAPOLControllerSandbox: icdd(491) allow network-outbound /private/var/run/mDNSResponderuipc_accept: peer disconnected unp_gencnt 736uipc_accept: peer disconnected unp_gencnt 738uipc_accept: peer disconnected unp_gencnt 740uipc_accept: peer disconnected unp_gencnt 7423 duplicate reports for Sandbox: icdd(491) allow network-outbound /private/var/run/mDNSResponderSandbox: icdd(491) allow file-read-data /Library/Image Capture/Devicesen1: DAD complete for fe80:5::46c:7fa5:9a5e:bd03 - no duplicates found .
sep@MacBook-Pro ~ % 
`

### Info in System Report - Bluetooth

```shell
"NULL"  placeholder under Bluetooth
usb devices don't show any hcl device
```

### Relevant log output

```shell
see above
```

### Kernel Panic Logs

```shell
no panics here
```

peddy22, 2022-08-04:
Some more kernel logs
`[   15.317519]: IntelFirmware: Driver init()
[   15.317546]: IntelFirmware: Driver Probe()
[   15.317565]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0032
[   15.317597]: IntelFirmware: Driver Start()
[   15.317682]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[   15.317695]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[   15.317724]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[   15.318795]: IntelFirmware: set configuration to 1
[   15.320075]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[   15.320094]: IntelFirmware: Found interface!!!
[   15.320137]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[   15.320148]: IntelFirmware: Found Interrupt endpoint!
[   15.320366]: IntelFirmware: Found Bulk out endpoint!
[   15.320438]: IntelFirmware: Found Bulk in endpoint!
[   15.320520]: IntelFirmware: BT init succeed
[   15.324750]: IntelFirmware: Firmware timestamp 2021.39 buildtype 1 build 31878
[   15.325252]: IntelFirmware: Found device firmware ibt-0041-0041.sfi 
[   15.341031]: IntelFirmware: Found device firmware: ibt-0041-0041.sfi
[   15.341143]: IntelFirmware: Boot Address: 0x100800
[   15.341154]: IntelFirmware: Firmware Version: 134-39.21
[   15.341162]: IntelFirmware: Firmware already loaded
[   15.341443]: IntelFirmware: Frimware is already running, finishing
[   15.343764]: IntelFirmware: Clean up...
[   15.343780]: IntelFirmware: virtual void BtIntel::free()
[   15.343788]: IntelFirmware: virtual void USBDeviceController::free()
[   28.272727]: itlwm: iwx_load_firmware
[   28.514854]: itlwm: iwx_rx_pkt: firmware alive, size=128
[   28.515221]: itlwm: : load firmware ok
[   28.515337]: itlwm: load firmware iwlwifi-ty-a0-gf-a0.pnvm done 27456
`

williambj1, 2022-08-07:
Proper Usb port mapping is required.

peddy22, 2022-08-07:
> Proper Usb port mapping is required.

got it, do you have a guide for this? is it just a matter of setting the port type to internal in USBMap? (which I've tried)

williambj1, 2022-08-07:
A quick Google search will show plenty of decent guides and tools. Closed since this is not a bug of this project.

closed 2022-08-07 by williambj1

peddy22, 2022-08-07:
Thanks. I agree since there are many reports of AX210 working. I think I'll
go with the Windows tool since USBMap hasn't given me joy even when setting
the port type to 255.

There is a cosmetic bug in the kernel messages - "frimware" that's about it.