OpenIntelWireless/IntelBluetoothFirmware #407: Strange Issues on AX211 and Bluetooth 5.2

Thamizhan, 2022-08-07:
### Have you read the docs?

Yes

### macOS Version

macOS 12.4

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel 6E AX211

### Description

I used latest kexts and patches from this on my Gigabyte Z690 UD AX with AX211 and Bluetooth 5.2 https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/369#issuecomment-1193041516

But My BT is not fully working. It doesn't show scan results for my Realme Bud Q and Mivi Roam 2 Headset. But shows my Samsung F62 smart phone and I can connect to the phone and transfer files. Only issues with the headphones or speakres. All 3 devices are on Bluetooth 5.0. 

[image]

Mobo: GIgabyte Z690 UD AX with Wi-Fi 6E AX211 and Bluetooth 5.2

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	<My BT Mac - shown correct value>
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v13446 c13446
  Product ID:	0x0001
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
-
```

williambj1, 2022-08-07:
Unfortunately there’s nothing we can do from our side since we’re using Apple’s Bluetooth stack. Closing as this is neither a firmware uploading nor a LE Scan initialization workaround issue.

Thamizhan, 2022-08-07:
The problem is with my Antena received from mobo. Tested with other antenna cables and everything works great.

closed 2022-08-07 by williambj1