OpenIntelWireless/IntelBluetoothFirmware #412: ax200 use IntelBTPatcher.kext sometime can't shutdown normally

Kwokping, 2022-09-13:
### Have you read the docs?

Yes

### macOS Version

macOS 13

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel ax 200

### Description

ax200 use IntelBTPatcher.kext sometime can't shutdown normally , shutdown the machine,but become reboot. remove IntelBTPatcher.kext become normal.

### Info in System Report - Bluetooth

```shell
Sep 13 10:08:36 ac12039f5f6f shutdown[3895]: SHUTDOWN_TIME: 1663034916 488566
Sep 13 10:09:03 localhost bootlog[0]: BOOT_TIME 1663034943 0
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.cdscheduler" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.install" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.authd" sharing output destination "/var/log/asl" with ASL Module "com.apple.asl".
	Output parameters from ASL Module "com.apple.asl" override any specified in ASL Module "com.apple.authd".
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.authd" sharing output destination "/var/log/system.log" with ASL Module "com.apple.asl".
	Output parameters from ASL Module "com.apple.asl" override any specified in ASL Module "com.apple.authd".
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.authd" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.eventmonitor" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.mail" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.performance" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.iokit.power" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.contacts.ContactsAutocomplete" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.mkb" sharing output destination "/private/var/log/keybagd.log" with ASL Module "com.apple.mkb.internal".
	Output parameters from ASL Module "com.apple.mkb.internal" override any specified in ASL Module "com.apple.mkb".
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.mkb" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: Configuration Notice:
	ASL Module "com.apple.MessageTracer" claims selected messages.
	Those messages may not appear in standard system log files or in the ASL database.
Sep 13 10:09:08 localhost syslogd[117]: ASL Sender Statistics
Sep 13 10:09:16 iMac-Pro loginwindow[155]: Clearing phase info
Sep 13 10:09:16 iMac-Pro loginwindow[155]: calling progress app remoteObjectProxy
Sep 13 10:09:16 iMac-Pro loginwindow[155]: Creating connection
Sep 13 10:09:16 iMac-Pro loginwindow[155]: Connection = 0x60000022c3c0, callers = (
	    "<IASUPCaller: 0x600003134910>"
[... 200 lines cut ...]
Sep 13 10:15:12 iMac-Pro loginwindow[154]: ISAP: hide progress UI called
Sep 13 10:15:12 iMac-Pro loginwindow[154]: client 0x600001dbddc0: phaseName = "loginwindow Boot" is being released
Sep 13 10:15:12 iMac-Pro loginwindow[154]: Releasing Connection
Sep 13 10:15:17 iMac-Pro loginwindow[154]: The connection was interrupted, calling interruption handlers
Sep 13 10:15:18 iMac-Pro loginwindow[154]: USER_PROCESS: 154 console
Sep 13 10:15:57 ac12039f5f6f LemonMonitor[621]: DataTransport::TransportMgr +++++
Sep 13 10:15:57 ac12039f5f6f LemonDaemon[645]: DataTransport::TransportMgr +++++
Sep 13 10:16:11 ac12039f5f6f loginwindow[154]: There is still an active connection
Sep 13 10:16:43 --- last message repeated 1 time ---
Sep 13 10:23:37 ac12039f5f6f login[980]: USER_PROCESS: 980 ttys000
Sep 13 10:27:35 ac12039f5f6f syslogd[117]: ASL Sender Statistics
```

### Relevant log output

```shell
null
```

### Kernel Panic Logs

```shell
{"bug_type":"210","timestamp":"2022-09-06 11:21:29.00 +0800","os_version":"macOS 13.0 (22A5331f)","roots_installed":0,"incident_id":"47666274-9BE9-4388-B5CA-5B3D44EF370D"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBvciBzbGVlcC93YWtlIGZhaWx1cmUgcGFuaWMgc3RhY2tzaG90IGZvdW5k","macOSPanicString":"panic(cpu 4 caller 0xffffff8013d4f163): Kernel trap at 0xffffff8016990c98, type 14=page fault, registers:\nCR0: 0x000000008001003b, CR2: 0x0000000000000048, CR3: 0x000000001ffeb000, CR4: 0x00000000001626e0\nRAX: 0xffffff8016990c94, RBX: 0xffffff955a2d0900, RCX: 0x0000000000000000, RDX: 0x00000000e00002eb\nRSP: 0xfffffffffde63bb0, RBP: 0xfffffffffde63bb0, RSI: 0x0000000000000000, RDI: 0xffffff8bc28b8200\nR8:  0xffffff8018c6b0e0, R9:  0x0000000000005649, R10: 0x0000000000000005, R11: 0xffffffc0f3564000\nR12: 0xffffff80153e8f5a, R13: 0x0000000000000000, R14: 0xffffff8bc2583300, R15: 0xffffff955a2d0900\nRFL: 0x0000000000010282, RIP: 0xffffff8016990c98, CS:  0x0000000000000008, SS:  0x0000000000000010\nFault CR2: 0x0000000000000048, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 1\n\nPanicked task 0xffffff908e3fb218: 237 threads: pid 0: kernel_task\nBacktrace (CPU 4), panicked thread: 0xffffff90886ad0c8, Frame : Return Address\n0xfffffffffde63590 : 0xffffff8013bf4a4d \n0xfffffffffde635e0 : 0xffffff8013d5fa06 \n0xfffffffffde63620 : 0xffffff8013d4ec90 \n0xfffffffffde63670 : 0xffffff8013b8f951 \n0xfffffffffde63690 : 0xffffff8013bf4d2d \n0xfffffffffde63780 : 0xffffff8013bf43d9 \n0xfffffffffde637e0 : 0xffffff80143e1a9b \n0xfffffffffde638d0 : 0xffffff8013d4f163 \n0xfffffffffde63a50 : 0xffffff8013d4ee52 \n0xfffffffffde63aa0 : 0xffffff8013b8f951 \n0xfffffffffde63ac0 : 0xffffff8016990c98 \n0xfffffffffde63bb0 : 0xffffff8018c6b2e7 \n0xfffffffffde63c20 : 0xffffff80169dfb4f \n0xfffffffffde63df0 : 0xffffff80153e8fad \n0xfffffffffde63e20 : 0xffffff8014318044 \n0xfffffffffde63e80 : 0xffffff80153e8f27 \n0xfffffffffde63ea0 : 0xffffff8013c4d568 \n0xfffffffffde63ee0 : 0xffffff8013c4e42e \n0xfffffffffde63fa0 : 0xffffff8013b8f19e \n      Kernel Extensions in backtrace:\n         com.apple.driver.usb.AppleUSBCommon(1.0)[16524148-F9F7-3453-843B-F23820AE5F40]@0xffffff80153e6000->0xffffff80153e9fff\n         com.apple.iokit.IOUSBHostFamily(1.2)[3C9477AB-8008-3555-B2DC-5A3BB36E39F9]@0xffffff801698c000->0xffffff8016a00fff\n            dependency: com.apple.driver.AppleBusPowerController(1.0)[9B3DA1D5-0140-333D-84C9-5FF2413FD369]@0xffffff8014ec6000->0xffffff8014ec9fff\n            dependency: com.apple.driver.AppleSMC(3.1.9)[5C075FAA-709A-329C-9099-6C64607C141E]@0xffffff801525e000->0xffffff801527afff\n            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[16524148-F9F7-3453-843B-F23820AE5F40]@0xffffff80153e6000->0xffffff80153e9fff\n            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[30CF4C8D-D8C2-3001-B22F-B07C6240EE49]@0xffffff8016a62000->0xffffff8016a62fff\n            dependency: com.apple.iokit.IOACPIFamily(1.4)[62C52EB0-4D0B-3A00-B4B0-A1841EA15AB4]@0xffffff801617d000->0xffffff801617efff\n         com.zxystd.IntelBTPatcher(2.2)[F092EAC7-8693-3E95-ADCD-EFD18C44142F]@0xffffff8018c68000->0xffffff8018c76fff\n            dependency: as.vit9696.Lilu(1.6.2)[264B15BE-8923-3A33-A9F5-8F0FFBB80595]@0xffffff80179f8000->0xffffff8017a23fff\n            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[3C9477AB-8008-3555-B2DC-5A3BB36E39F9]@0xffffff801698c000->0xffffff8016a00fff\n\nProcess name corresponding to current thread (0xffffff90886ad0c8): kernel_task\nBoot args: -wegnoigpu debug=0x100 -lilubetaall\n\nMac OS version:\n22A5331f\n\nKernel version:\nDarwin Kernel Version 22.1.0: Mon Aug 15 20:06:46 PDT 2022; root:xnu-8792.40.29.161.2~1\/RELEASE_X86 [... cut ...] 090423000\n  . VM    : 0xffffff8090423000 - 0xffffff855c756000\n  . RO    : 0xffffff855c756000 - 0xffffff86f5dbc000\n  . GEN0  : 0xffffff86f5dbc000 - 0xffffff8bc20ef000\n  . GEN1  : 0xffffff8bc20ef000 - 0xffffff908e422000\n  . GEN2  : 0xffffff908e422000 - 0xffffff955a755000\n  . GEN3  : 0xffffff955a755000 - 0xffffff9a26a88000\n  . DATA  : 0xffffff9a26a88000 - 0xffffffa08c422000\n  Metadata: 0xffffff803fcc2000 - 0xffffff805fcc2000\n  Bitmaps : 0xffffff805fcc2000 - 0xffffff806bcc2000\n\n\n\n"}
```

williambj1, 2022-09-13:
Please add `keepsyms=1` to boot-args and post the panic log again. Thank you.

Kwokping, 2022-09-13:
@williambj1 
panic(cpu 0 caller 0xffffff801834d373): Kernel trap at 0xffffff801af98c98, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000048, CR3: 0x00000000245f3000, CR4: 0x00000000001626e0
RAX: 0xffffff801af98c94, RBX: 0xffffff99de17b300, RCX: 0x0000000000000000, RDX: 0x00000000e00002eb
RSP: 0xffffffcb8dd0bbb0, RBP: 0xffffffcb8dd0bbb0, RSI: 0x0000000000000000, RDI: 0xffffff86add7fe00
R8:  0xffffff801d9530e0, R9:  0x0000000000000167, R10: 0x0000000000000005, R11: 0xffffffcb8def9000
R12: 0xffffff80199ecf5a, R13: 0x0000000000000000, R14: 0xffffff95121369c0, R15: 0xffffff99de17b300
RFL: 0x0000000000010282, RIP: 0xffffff801af98c98, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000048, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Panicked task 0xffffff90461ba218: 234 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff95126f3b30, Frame : Return Address
0xffffffcb8dd0b590 : 0xffffff80181f1bfd mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffcb8dd0b5e0 : 0xffffff801835dc16 mach_kernel : _kdp_i386_trap + 0x116
0xffffffcb8dd0b620 : 0xffffff801834cea0 mach_kernel : _kernel_trap + 0x3e0
0xffffffcb8dd0b670 : 0xffffff801818c951 mach_kernel : _return_from_trap + 0xc1
0xffffffcb8dd0b690 : 0xffffff80181f1edd mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffcb8dd0b780 : 0xffffff80181f1589 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffffcb8dd0b7e0 : 0xffffff80189e11cb mach_kernel : _panic + 0x84
0xffffffcb8dd0b8d0 : 0xffffff801834d373 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffcb8dd0ba50 : 0xffffff801834d062 mach_kernel : _kernel_trap + 0x5a2
0xffffffcb8dd0baa0 : 0xffffff801818c951 mach_kernel : _return_from_trap + 0xc1
0xffffffcb8dd0bac0 : 0xffffff801af98c98 com.apple.iokit.IOUSBHostFamily : __ZN31AppleUSBHostInterfaceUserClient17requestCompletionEP37AppleUSBHostUserClientAsyncParamBlockij + 0x4
0xffffffcb8dd0bbb0 : 0xffffff801d9532e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207
0xffffffcb8dd0bc20 : 0xffffff801afe7b4f com.apple.iokit.IOUSBHostFamily : __ZN17AppleUSBIORequest8completeEv + 0xa45
0xffffffcb8dd0bdf0 : 0xffffff80199ecfad com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter30completeRequestThreadCallGatedEP15AppleUSBRequest + 0x53
0xffffffcb8dd0be20 : 0xffffff80189175b4 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffcb8dd0be80 : 0xffffff80199ecf27 com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter25completeRequestThreadCallEPNS_20tRequestCompleteDataE + 0x2f
0xffffffcb8dd0bea0 : 0xffffff801824ad08 mach_kernel : _thread_call_delayed_timer + 0x478
0xffffffcb8dd0bee0 : 0xffffff801824bbce mach_kernel : _thread_call_delayed_timer + 0x133e
0xffffffcb8dd0bfa0 : 0xffffff801818c19e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.driver.usb.AppleUSBCommon(1.0)[8C2D8B58-4EA1-35C7-9A82-B85D4F740E48]@0xffffff80199ea000->0xffffff80199edfff
         com.apple.iokit.IOUSBHostFamily(1.2)[E4B707B6-99D2-3AB9-B7D3-DDC9733B7001]@0xffffff801af94000->0xffffff801b008fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[9B3DA1D5-0140-333D-84C9-5FF2413FD369]@0xffffff80194c6000->0xffffff80194c9fff
            dependency: com.apple.driver.AppleSMC(3.1.9)[5C075FAA-709A-329C-9099-6C64607C141E]@0xffffff8019862000->0xffffff801987efff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[8C2D8B58-4EA1-35C7-9A82-B85D4F740E48]@0xffffff80199ea000->0xffffff80199edfff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[AB38344E-8B98-3574-A571-B8A82E8C9D9F]@0xffffff801b06a000->0xffffff801b06afff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[62C52EB0-4D0B-3A00-B4B0-A1841EA15AB4]@0xffffff801a785000->0xffffff801a786fff
         com.zxystd.IntelBTPatcher(2.2)[F092EAC7-8693-3E95-ADCD-EFD18C44142F]@0xffffff801d950000->0xffffff801d95efff
            dependency: as.vit9696.Lilu(1.6.2)[264B15BE-8923-3A33-A9F5-8F0FFBB80595]@0xffffff801c000000->0xffffff801c02bfff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[E4B707B6-99D2-3AB9-B7D3-DDC9733B7001]@0xffffff801af94000->0xffffff801b008fff

Process name corresponding to current thread (0xffffff95126f3b30): kernel_task
Boot args: -wegnoigpu debug=0x100 keepsyms=1 -lilubetaall

Mac OS version:
22A5342f

Kernel version:
Darwin Kernel Version 22.1.0: Mon Aug 29 20:09:50 PDT 2022; root:xnu-8792.40.74.0.1~20/RELEASE_X86_64
Kernel UUID: 4303EAC8-C588-3279-8486-671BDE11D04D
roots installed: 0
KernelCache slide: 0x0000000017e00000
KernelCache base:  0xffffff8018000000
Kernel slide:      0x0000000017edc000
Kernel text base:  0xffffff80180dc000
__HIB  text base: 0xffffff8017f00000
System model name: iMacPro1,1 (Mac-7BA5B2D9E42DDD94)
System shutdown begun: YES
Panic diags file available: NO (0xe00002bc)
Hibernation exit count: 0

System uptime in nanoseconds: 37926943865
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x00000008d49f3f5c
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000e6591a740 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff8044216000 - 0xffffffa044216000
  . PGZ   : 0xffffff8044216000 - 0xffffff8048217000
  . VM    : 0xffffff8048217000 - 0xffffff851454a000
  . RO    : 0xffffff851454a000 - 0xffffff86adbb0000
  . GEN0  : 0xffffff86adbb0000 - 0xffffff8b79ee3000
  . GEN1  : 0xffffff8b79ee3000 - 0xffffff9046216000
  . GEN2  : 0xffffff9046216000 - 0xffffff9512549000
  . GEN3  : 0xffffff9512549000 - 0xffffff99de87c000
  . DATA  : 0xffffff99de87c000 - 0xffffffa044216000
  Metadata: 0xffffffc0a4666000 - 0xffffffc0c4666000
  Bitmaps : 0xffffffc0c4666000 - 0xffffffc0d0666000

Kwokping, 2022-09-22:
@zxystd can the problem be fixed in the next version？

zxystd, 2022-09-23:
@Kwokping Firstly thank you for the feedback, I will make some changes and post here when I have free time, I am busy these days. Maybe this weekend?

Kwokping, 2022-09-29:
@zxystd thanks.

zxystd, 2022-12-01:
@Kwokping  Hi, sorry for use a long time to reply... please try this one patcher, and try to reproduce the issue, thank you.
[IntelBTPatcher.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/10134352/IntelBTPatcher.kext.zip)

Kwokping, 2022-12-01:
@zxystd Cloudn‘t download the file

zxystd, 2022-12-02:
[IntelBTPatcher.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/10137310/IntelBTPatcher.kext.zip)
@Kwokping Update the link

Kwokping, 2022-12-02:
@zxystd it seem to shutdown normally after updated the new kext, thank you!

zxystd, 2022-12-02:
> @zxystd it seem to shutdown normally after updated the new kext, thank you!

Ok, thank you, if you can confirm that it is fixed, I will push the commit now.

Kwokping, 2022-12-02:
@zxystd ok, it is fixed

closed 2023-06-09 by zxystd