OpenIntelWireless/IntelBluetoothFirmware #415: AX211 bluetooth module doesn't work in Monterey.

duckjpsgaintas, 2022-10-02:
### Have you read the docs?

Yes

### macOS Version

Monterey

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AX211

### Description

I set up the IntelBluetoothFirmware.kext / IIntelBTPatcher.kext / Bluetoolfixup.kext just following the guiding document and the bluetooth doesn't work. The bluetooth device information on the system report shows the address is a NULL and the chipset is BCM_4350C2 and the vendorID is 0x004C(apple). I saw some other users also met this problem. 

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Product ID:	0x0001
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
already installed DebugEnhancer but still didn't show anything after typing "sudo dmesg | grep -i IntelBluetooth"
```

mianmian7, 2022-10-18:
try to Making USB mapping

sth0409, 2022-10-26:
尝试修改一下顺序
[image]

zxystd, 2022-12-01:
user mistake.

closed 2022-12-01 by zxystd