OpenIntelWireless/IntelBluetoothFirmware #416: Random Panics

naveenkrdy, 2022-10-23:
### Have you read the docs?

Yes

### macOS Version

macOS 12.6

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

WIFI 6 AX200

### Description

System panics randomly. Mostly during sleep. This started happening since updating to latest BT kexts. And happens even when my BT is off

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	24:41:8C:64:4F:32
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v256 c256
  Product ID:	0x0001
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
N/A
```

### Kernel Panic Logs

```shell
panic(cpu 9 caller 0xffffff8016fd0833): Kernel trap at 0xffffff8017597685, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000038, CR3: 0x0000000028527000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff801c826020, RCX: 0x0000000000000000, RDX: 0x00000000e00002eb
RSP: 0xffffffffd764ba30, RBP: 0xffffffffd764ba40, RSI: 0xffffff801c826020, RDI: 0xffffffa544caf550
R8:  0xffffff801c1550e0, R9:  0x00000000fffeffff, R10: 0x0000000000000003, R11: 0xffffffa078e94540
R12: 0xffffffa078d35400, R13: 0xaaaaaaaaaaaaaaaa, R14: 0xffffffa544caf550, R15: 0xffffffa545aa8b01
RFL: 0x0000000000010286, RIP: 0xffffff8017597685, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000038, Error code: 0x0000000000000000, Fault CPU: 0x9, PL: 0, VF: 1

Panicked task 0xffffff96e028f670: 184 threads: pid 0: kernel_task
Backtrace (CPU 9), panicked thread: 0xffffffa078e94540, Frame : Return Address
0xffffffffd764b3e0 : 0xffffff8016e7fdad mach_kernel : _handle_debugger_trap + 0x41d
0xffffffffd764b430 : 0xffffff8016fe10d6 mach_kernel : _kdp_i386_trap + 0x116
0xffffffffd764b470 : 0xffffff8016fd0443 mach_kernel : _kernel_trap + 0x4d3
0xffffffffd764b4c0 : 0xffffff8016e1fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffffd764b4e0 : 0xffffff8016e8017d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffffd764b600 : 0xffffff8016e7f936 mach_kernel : _panic_trap_to_debugger + 0x2b6
0xffffffffd764b660 : 0xffffff8017715093 mach_kernel : _panic + 0x84
0xffffffffd764b750 : 0xffffff8016fd0833 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffffd764b8d0 : 0xffffff8016fd0516 mach_kernel : _kernel_trap + 0x5a6
0xffffffffd764b920 : 0xffffff8016e1fa70 mach_kernel : _return_from_trap + 0xe0
0xffffffffd764b940 : 0xffffff8017597685 mach_kernel : __ZN15OSMetaClassBase12safeMetaCastEPKS_PK11OSMetaClass + 0x15
0xffffffffd764ba40 : 0xffffff801c165cd9 com.zxystd.IntelBluetoothFirmware : __ZN19USBDeviceController16interruptHandlerEPvS0_ij + 0x29
0xffffffffd764ba90 : 0xffffff801c1552e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207
0xffffffffd764bb00 : 0xffffff8019e25a5e com.apple.iokit.IOUSBHostFamily : __ZN17AppleUSBIORequest8completeEv + 0x1048
0xffffffffd764bcd0 : 0xffffff80187e5aef com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter20completeRequestQueueEP11queue_entry + 0x2d1
0xffffffffd764bee0 : 0xffffff80187e5801 com.apple.driver.usb.AppleUSBCommon : __ZN24AppleUSBRequestCompleter12checkForWorkEv + 0x1ed
0xffffffffd764bf20 : 0xffffff801764399e mach_kernel : __ZN10IOWorkLoop15runEventSourcesEv + 0x13e
0xffffffffd764bf60 : 0xffffff8017642fc7 mach_kernel : __ZN10IOWorkLoop10threadMainEv + 0x37
0xffffffffd764bfa0 : 0xffffff8016e1f19e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.driver.usb.AppleUSBCommon(1.0)[2C01FB21-7D31-31DC-865D-C4781169215C]@0xffffff80187e3000->0xffffff80187e6fff
         com.apple.iokit.IOUSBHostFamily(1.2)[A40D8C37-5C5A-31AE-B3F6-056733605B6F]@0xffffff8019db8000->0xffffff8019e48fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[E995B5AE-4200-3B66-8850-FB49BF27BC11]@0xffffff80182d9000->0xffffff80182dcfff
            dependency: com.apple.driver.AppleSMC(3.1.9)[EE1310F2-B1FA-3BF6-B86B-65F1EA049CE7]@0xffffff801866f000->0xffffff8018687fff
            dependency: com.apple.driver.usb.AppleUSBCommon(1.0)[2C01FB21-7D31-31DC-865D-C4781169215C]@0xffffff80187e3000->0xffffff80187e6fff
            dependency: com.apple.driver.AppleUSBHostMergeProperties(1.2)[4A25A0A2-8E4E-3330-A9F5-F3E4D9A8945A]@0xffffff8019ebc000->0xffffff8019ebcfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CD3677CB-1370-3425-AB61-064B4F0508E6]@0xffffff801958c000->0xffffff801958dfff
         com.zxystd.IntelBTPatcher(2.2)[F092EAC7-8693-3E95-ADCD-EFD18C44142F]@0xffffff801c152000->0xffffff801c160fff
            dependency: as.vit9696.Lilu(1.6.2)[264B15BE-8923-3A33-A9F5-8F0FFBB80595]@0xffffff801ae64000->0xffffff801ae8ffff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[A40D8C37-5C5A-31AE-B3F6-056733605B6F]@0xffffff8019db8000->0xffffff8019e48fff
         com.zxystd.IntelBluetoothFirmware(2.2)[43887626-85E4-3BE5-B383-5D25CA48815C]@0xffffff801c163000->0xffffff801c836fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[A40D8C37-5C5A-31AE-B3F6-056733605B6F]@0xffffff8019db8000->0xffffff8019e48fff

Process name corresponding to current thread (0xffffffa078e94540): kernel_task
Boot args: keepsyms=1 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
21G115

Kernel version:
Darwin Kernel Version 21.6.0: Mon Aug 22 20:17:10 PDT 2022; root:xnu-8020.140.49~2/RELEASE_X86_64
Kernel UUID: 06A10466-36D7-3A0B-9EE7-A9053A8991A3
KernelCache slide: 0x0000000016c00000
KernelCache base:  0xffffff8016e00000
Kernel slide:      0x0000000016c10000
Kernel text base:  0xffffff8016e10000
__HIB  text base: 0xffffff8016d00000
System model name: MacBookPro16,1 (Mac-E1008331FDC96864)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 22632385126019
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000014958346a35f
  Sleep   : 0x0000149552c6333b 0x00000000f507fb94 0x00001489d20dec71
  Wake    : 0x000014957cbcb7fd 0x00000000f4ae9204 0x0000149559729e11
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 1 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff90781ba000 - 0xffffffb0781ba000
  . PGZ   : 0xffffff90781ba000 - 0xffffff907a1bb000
  . VM    : 0xffffff907a1bb000 - 0xffffff95469ba000
  . RO    : 0xffffff95469ba000 - 0xffffff96e01ba000
  . GEN0  : 0xffffff96e01ba000 - 0xffffff9bac9ba000
  . GEN1  : 0xffffff9bac9ba000 - 0xffffffa0791ba000
  . GEN2  : 0xffffffa0791ba000 - 0xffffffa5459ba000
  . GEN3  : 0xffffffa5459ba000 - 0xffffffaa121ba000
  . DATA  : 0xffffffaa121ba000 - 0xffffffb0781ba000
  Metadata: 0xfffffff28651e000 - 0xfffffff2a651e000
  Bitmaps : 0xfffffff2a651e000 - 0xfffffff2ac51e000
```

songkea, 2022-10-24:
macos 12.6 
ax201

BlueToolFixup.kext
IntelBluetoothFirmware.kext
IntelBTPatcher.kext

start system info:  iokit daemon（kernelmanagerd） stall[0] (60s) IOUSBHostDevice ‘HDAS’
Wait four minutes to get to the desktop

Bluetooth can not open automatically, manually click bluetooth open, wait for ten minutes, Bluetooth can be used normally, can relay, search to other Bluetooth signals. I look forward to hearing from you. Thank you

hacker1024, 2022-10-26:
This happens on Ventura too. Same backtrace.

naveenkrdy, 2022-11-05:
No more panics after disabling power nap in power settings.

zxystd, 2022-12-01:
@naveenkrdy Hi, sorry for use a long time to reply... please try this one patcher, and try to reproduce the issue, thank you.
[IntelBTPatcher.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/10134352/IntelBTPatcher.kext.zip)

zxystd, 2022-12-02:
[IntelBTPatcher.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/10137313/IntelBTPatcher.kext.zip)
Update the link.

Andrupka, 2023-05-06:
This happens to me too. Will that updated patcher help?

closed 2023-06-09 by zxystd