OpenIntelWireless/IntelBluetoothFirmware #417: Wake from sleep is an automatic machine restart

duxphp, 2022-10-26:
### Have you read the docs?

Yes

### macOS Version

macos  13

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

ax201

### Description

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               bluetoothd [5108]
Path:                  /usr/sbin/bluetoothd
Identifier:            bluetoothd
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               0

Date/Time:             2022-10-25 16:56:15.8233 +0800
OS Version:            macOS 13.0 (22A380)
Report Version:        12
Anonymous UUID:        4B3C2C9B-9276-D257-5134-E1D8DD526CE1

Sleep/Wake UUID:       F22C9021-CBE0-443C-B534-5635881E28AC

Time Awake Since Boot: 6200 seconds
Time Since Wake:       2 seconds

System Integrity Protection: enabled

Crashed Thread:        4

Exception Type:        EXC_GUARD
Exception Codes:       GUARD_TYPE_USER
Exception Codes:       0x6000000000000012, 0x0000000000000002

Termination Reason:    Namespace LIBSYSTEM, Code 2 Application Triggered Fault

Thread 0:
0   libsystem_kernel.dylib        	    0x7ff80af8c6a2 mach_msg2_trap + 10
1   libsystem_kernel.dylib        	    0x7ff80af9a67d mach_msg2_internal + 82
2   libsystem_kernel.dylib        	    0x7ff80af9371a mach_msg_overwrite + 723
3   libsystem_kernel.dylib        	    0x7ff80af8c989 mach_msg + 19
4   CoreFoundation                	    0x7ff80b0a5d0e __CFRunLoopServiceMachPort + 145
5   CoreFoundation                	    0x7ff80b0a477a __CFRunLoopRun + 1360
6   CoreFoundation                	    0x7ff80b0a3bb0 CFRunLoopRunSpecific + 560
7   CoreFoundation                	    0x7ff80b1287ea CFRunLoopRun + 40
8   bluetoothd                    	       0x107a54841 0x107a2d000 + 161857
9   dyld                          	    0x7ff80ac9a310 start + 2432

Thread 1:
0   libsystem_kernel.dylib        	    0x7ff80af8dd96 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	    0x7ff80afc7d47 _pthread_wqthread + 426
2   libsystem_pthread.dylib       	    0x7ff80afc6c67 start_wqthread + 15

Thread 2:
0   libsystem_kernel.dylib        	    0x7ff80af8dd96 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	    0x7ff80afc7d47 _pthread_wqthread + 426
[... 224 lines cut ...]
],
  "sharedCache" : {
  "base" : 140703308972032,
  "size" : 21474836480,
  "uuid" : "3e6c1144-e4b6-381c-92d1-0a3396dd3bb0"
},
  "vmSummary" : "ReadOnly portion of Libraries: Total=546.8M resident=0K(0%) swapped_out_or_unallocated=546.8M(100%)\nWritable regions: Total=499.6M written=0K(0%) resident=0K(0%) swapped_out=0K(0%) unallocated=499.6M(100%)\n\n                                VIRTUAL   REGION \nREGION TYPE                        SIZE    COUNT (non-coalesced) \n===========                     =======  ======= \nActivity Tracing                   256K        1 \nDispatch continuations           160.0M        2 \nIOKit                               16K        2 \nKernel Alloc Once                    8K        1 \nMALLOC                           206.3M       72 \nMALLOC guard page                   48K       10 \nMALLOC_MEDIUM (reserved)         120.0M        1         reserved VM address space (unallocated)\nSQLite page cache                  384K        3 \nSTACK GUARD                       56.0M        9 \nStack                             12.1M        9 \nStack (reserved)                   520K        1         reserved VM address space (unallocated)\nStack Guard                          4K        1 \nVM_ALLOCATE                        212K        6 \n__CTF                               756        1 \n__DATA                            20.0M      450 \n__DATA_CONST                      25.7M      291 \n__DATA_DIRTY                      1283K      172 \n__FONT_DATA                        2352        1 \n__LINKEDIT                       182.2M        5 \n__OBJC_RO                         65.1M        1 \n__OBJC_RW                         1982K        2 \n__TEXT                           364.7M      468 \ndyld private memory                256K        1 \nmapped file                       29.3M        7 \nshared memory                      584K        7 \n===========                     =======  ======= \nTOTAL                              1.2G     1524 \nTOTAL, minus reserved VM space     1.1G     1524 \n",
  "legacyInfo" : {
  "threadTriggered" : {

  }
}
}

### Info in System Report - Bluetooth

```shell
无
```

### Relevant log output

```shell
无
```

lavluda, 2022-11-15:
faced same issue

phplaw, 2022-11-20:
Ventura got same issue.

guliujian, 2022-11-25:
also got same issue in Ventura

liweilong, 2022-11-28:
Same issue with Ventura 13.0.1 AX211 and 2.2.0 kext

liweilong, 2022-11-29:
I got same issue, but I use bluesnooze fix this https://github.com/odlp/bluesnooze
This app can close bluetooth before mac sleep, and open it after awake.

zxystd, 2022-12-02:
Please try this version https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/412#issuecomment-1334642864

liweilong, 2022-12-03:
I have download the new ItelBTPatcher.text.zip, unzip it and replace it in EFI/OC/Kexts, but the bluetooth not even work after reboot.
Sorry, I'm a beginner of opencore. @zxystd

zxystd, 2022-12-04:
> I have download the new ItelBTPatcher.text.zip, unzip it and replace it in EFI/OC/Kexts, but the bluetooth not even work after reboot. Sorry, I'm a beginner of opencore. @zxystd

I don't know which `Lilu` version you are using, but I have push the changes to the master branch, you can download from the Action artifacts.
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/suites/9625333631/artifacts/459755927

superbogie, 2023-03-13:
> I don't know which `Lilu` version you are using, but I have push the changes to the master branch, you can download from the Action artifacts. [...]

Same issue for me under Ventura 13.2.1. I get a system lock-up at about 12000 seconds of uptime followed by a reboot. Have tried updating from v2.2.0 to the v2.3.0 IntelBluetoohPatcher.kext and while BT is shown as on after rebooting, in the System Settings as well as the menubar, it doesn't recognize my mouse nor my BT earbuds no matter how many times I turn either of them on or off. Also attempted to DL the above "master branch" version but get an error message at the top of the page stating "Failed to generate URL to download artifact."

weekly71, 2023-06-08:
Having same issue after upgrading to Ventura 13.4. My card is Intel Dual-Band AC3160. I've tried fixed IntelBTPatcher from https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/412#issuecomment-1334642864, which fixes sleep and shutdown but breaks bluetooth. Artifact link above doesn't work.

closed 2023-06-09 by zxystd