OpenIntelWireless/IntelBluetoothFirmware #418: Bluetooth not able to connect in Ventura

gnpaone, 2022-10-27:
### Have you read the docs?

Yes

### macOS Version

macOS 13

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AX200 (PID - 0x29, VID - 0x8087)

### Description

I have recently updated to Ventura latest public non-beta release, previously in Monterey Bluetooth used to work but now it's just detecting devices but couldn't connect (same behaviour happened with intel wifi - airportitlwm too - in fact connecting wifi gave kernel panic - just detected but bluetooth no KP)(Many kexts including whatevergreen not worked in Ventura but I updated all kexts to latest (all of them has changelogs macOS 13 constants added) including Bluetool, IntelBluetoothFirmware, and newly added IntelBTPatch (same issue with or without this new kext).) I have Debugenhancer latest kext only, no dmesg output for IntelBluetooth.

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	On
  Firmware Version:	v256 c256
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
sudo dmesg | grep -I IntelBluetooth no output
sudo dmesg | grep -I Bluetooth output:

[    0.615678]: **** [IOBluetoothFamily][start] -- completed successfully for HIDShim
**** [BroadcomBluetooth20703USBTransport][init] -- entering ****
[   10.135285]: **** [BroadcomBluetooth20703USBTransport][init] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][init] -- entering ****
[   10.135303]: **** [BroadcomBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][init] -- entering ****
[   10.135321]: **** [CSRBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][init] -- entering ****
[   10.135345]: **** [IOBluetoothHostControllerPCIeTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][init] -- entering ****
[   10.135355]: **** [IOBluetoothHostControllerTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[   10.135371]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUSBTransport][init] -- entering ****
[   10.135388]: **** [IOBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothPacketLogger][init] -- entering ****
[   10.135399]: **** [IOBluetoothPacketLogger][init] -- exiting -- success ****
**** [IOBluetoothSerialManager][init] -- entering ****
[   10.135410]: **** [IOBluetoothSerialManager][init] -- exiting -- success ****
**** [BroadcomBluetooth20703USBTransport][start] -- entering ****
[   10.137957]: **** [BroadcomBluetooth20703USBTransport][start] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][start] -- entering ****
[   10.137965]: **** [BroadcomBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][start] -- entering ****
[   10.137974]: **** [CSRBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][start] -- entering ****
[   10.138066]: **** [IOBluetoothHostControllerPCIeTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][start] -- entering ****
[   10.138071]: **** [IOBluetoothHostControllerTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[... 110 lines cut ...]
[   69.495796]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.525069]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.525109]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.536345]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.575738]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.576243]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.576290]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.579831]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.620990]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.621830]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.624281]: 1 duplicate report for Sandbox: analyticsd(179) deny(1) file-issue-extension target:/private/var/db/analyticsd/Library/Preferences/analyticsd.plist class:com.apple.cfprefsd.read-writeSandbox: systemstats(116) deny(1) mach-lookup com.apple.tccdmemorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.657598]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.658008]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.669191]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   69.669285]: memorystatus: assertion priority 40 overrides priority 0 for Bluetooth:686
[   69.695948]: memorystatus: assertion priority 40 overrides priority 30 for Bluetooth:686
[   75.602890]: 2 duplicate reports for Sandbox: siriactionsd(509) deny(1) user-preference-read com.apple.universalaccessSandbox: cloudpaird(564) deny(1) mach-lookup com.apple.tccdSandbox: cloudpaird(564) deny(1) system-fsctl (_IO "h" 47)Sandbox: cloudpaird(564) deny(1) mach-lookup com.apple.metadata.mds1 duplicate report for Sandbox: cloudpaird(564) deny(1) mach-lookup com.apple.metadata.mdsSandbox: cloudpaird(564) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plist1 duplicate report for Sandbox: cloudpaird(564) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: bluetoothd(169) deny(1) file-read-metadata /Library/Application SupportSandbox: bluetoothd(169) deny(1) file-read-metadata /Library/Application Support/BTServerUnable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving up.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving up.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving up.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving up.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving up.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Delaying for 1000 u.s. to see if the queue drains.Unable to send CoreAnalytics event. Giving u [... cut ...] ookup com.apple.tccdSandbox: systemstats(116) deny(1) mach-lookup com.apple.tccd.systemSandbox: systemstats(116) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(116) deny(1) mach-lookup com.apple.tccd.systemSandbox: passd(705) deny(1) mach-lookup com.apple.server.bluetooth.le.att.xpcSandbox: passd(705) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistAppleSSE::hasBridgeSSE - SSE not present (neither KR nor SEPMgr)
```

### Kernel Panic Logs

```shell
No Kernel Panic
```

ngobach, 2022-11-27:
Hello, using following kext combo fixed Bluetooth on my Intel AC 7260:

```
IntelBluetoothFirmware.kext (V2.1.0)
BlueToolFixup.kext (V2.6.4)
```

For **BlueToolFixup.kext** you can grab one from [here](https://dortania.github.io/builds/?product=BrcmPatchRAM&viewall=true).

gabrie1lira, 2022-11-29:
> Hello, using following kext combo fixed Bluetooth on my Intel AC 7260:
> 
> ```
> IntelBluetoothFirmware.kext (V2.1.0)
> BlueToolFixup.kext (V2.6.4)
> ```
> 
> For **BlueToolFixup.kext** you can grab one from [here](https://dortania.github.io/builds/?product=BrcmPatchRAM&viewall=true).

Thx, you are friend!

closed 2022-12-02 by williambj1

gnpaone, 2022-12-02:
> https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/419#issuecomment-1296216116

Ohh ok 🥺

williambj1, 2022-12-02:
> > [#419 (comment)](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/419#issuecomment-1296216116)
> 
> Ohh ok 🥺
> 
> PS: This is not a duplicate issue, I checked if anyone else reported before creating issue, maybe some other label suits for this 😑

The culprit is the same as #419. With that being said, is there anything I missed that makes this not a duplicate?