OpenIntelWireless/IntelBluetoothFirmware #419: Can't connect mouse on Ventura

stantravels, 2022-10-30:
### Have you read the docs?

Yes

### macOS Version

MacOS 13

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AC 9560 (0xaaa)

### Description

Hi! I started experiencing troubles in connecting my Huawei bluetooth mouse after upgrading to Ventura. The previous macOS versions including Monterey were working good with it, completely without any connectivity issues. The bluetooth settings see the mouse along with other BT devices nearby, my Galaxy Buds Live headphones are even connecting and working good, but the mouse is not - like in the issue #418 the symptoms are the same, I get the infinite connection loop, then the mouse disappears from the device list but continues to blink with connection signal unless I turn off bluetooth in macOS. I have the latest OC 0.8.5 and all the latest kexts installed. Can provide any additional info if needed.

### Info in System Report - Bluetooth

```shell
Контроллер Bluetooth:
  Адрес:	64:5D:86:BA:1C:46
  Состояние:	On
  Набор микросхем:	THIRD_PARTY_DONGLE
  Видимо для других устройств:	Off
  Версия прошивки:	v256 c256
  Поддерживаемые службы:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Передача данных:	USB
  ID производителя:	0x004C (Apple)
  Не подключено:
  Galaxy Buds Live (FBC7):
  Адрес:	6C:DD:BC:3F:FB:C7
  ID производителя:	0x0075
  ID продукта:	0xA013
  Версия прошивки:	0.0.1
  Дополнительный тип:	Headset
```

### Relevant log output

```shell
None
```

### Kernel Panic Logs

```shell
None
```

williambj1, 2022-10-30:
Sorry this is out of our control since the entire bluetooth stack is now running in userspace..

closed 2022-10-30 by williambj1

stantravels, 2022-10-30:
Sorry, but what does it mean from the user prospective? Is there anything I can do with it? If the headphones are working there must be a way to make the mouse work as well.

williambj1, 2022-10-30:
> Sorry, but what does it mean from the user prospective? Is there anything I can do with it? If the headphones are working there must be a way to make the mouse work as well.

It means whether the device can be connected or not is totally up to Apple's bugs/decisions. There's nothing we can do.