OpenIntelWireless/IntelBluetoothFirmware #421: Mouse not working

hjhrwerv2423, 2022-12-10:
### Have you read the docs?

Yes

### macOS Version

13.0.1 (22A400)

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AX201 (vendor: 0x8087, dev-id: 0x0026)

### Description

I have upgraded from 12.4 to 13.0.1 and my bluetooth mouse stopped working. At the same time, bluetooth with other devices works fine, I checked the bluetooth headphones and they work. But can't even connect to a mouse.
Mouse: bt 4.2
Headphones: 5.2

Same error as reported [here](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/51#issuecomment-1186527078)

### Info in System Report - Bluetooth

```shell
USB 3.1 Bus:

  Host Controller Driver:	AppleUSBXHCIPCI
  PCI Device ID:	0x34ed 
  PCI Revision ID:	0x0030 
  PCI Vendor ID:	0x8086 

Bluetooth USB Host Controller:

  Product ID:	0x0026
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.02
  Speed:	Up to 12 Mb/s
  Location ID:	0x14700000 / 3
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
  Built-In:	Yes
 

 Bluetooth Controller:
  Address:	0C:DD:24:CE:0D:0F
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v256 c256
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
  Not Connected:
  Haylou W1:
  Address:	9C:19:C2:35:C2:DA
  Minor Type:	Headset
```

### Relevant log output

```shell
[    6.077556]: IntelFirmware: Driver init()
[    6.077560]: IntelFirmware: Driver Probe()
[    6.077563]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0026
[    6.077568]: IntelFirmware: Driver Start()
[    6.077579]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    6.077580]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    6.077619]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    6.077621]: IntelFirmware: set configuration to 1
[    6.078165]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    6.078169]: IntelFirmware: Found interface!!!
[    6.078179]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    6.078181]: IntelFirmware: Found Interrupt endpoint!
[    6.078408]: IntelFirmware: Found Bulk out endpoint!
[    6.078657]: IntelFirmware: Found Bulk in endpoint!
[    6.078916]: IntelFirmware: BT init succeed
[    6.080299]: IntelFirmware: Read the legacy Intel version information
[    6.080303]: IntelFirmware: Firmware revision 0.0 build 111 week 53 2019
[    6.080308]: IntelFirmware: Frimware is already running, finishing
[    6.081304]: IntelFirmware: Clean up...
[    6.081307]: IntelFirmware: virtual void BtIntel::free()
[    6.081308]: IntelFirmware: virtual void USBDeviceController::free()
```

wangjiangb, 2023-01-07:
I have the same issue with the same wifi/bt card. The mouse or keyboard cannot be paired. Headphones can be paired and it is working correctly.

BronzedOreo, 2023-02-02:
我在ventura 13.2 ax201网卡也是同样的问题，蓝牙可以连接其他蓝牙设备 但是不能连接蓝牙鼠标

phucho236, 2023-03-06:
I have the same issue

tienthanght96, 2023-03-29:
I have the same issue on MacOS 13.2, some devices can be found but cannot connect (ex: keydous NJ68 keyboard), headphone/mobile devices work perfectly

RaDianTBLaZe, 2023-04-18:
> I have the same issue on MacOS 13.2, some devices can be found but cannot connect (ex: keydous NJ68 keyboard), headphone/mobile devices work perfectly

Can we fix it ? Bcoz some headphones connect and some do not

closed 2023-06-09 by zxystd