OpenIntelWireless/IntelBluetoothFirmware #423: Won't compile on Xcode 14.x and/or MacOS 13.x

mhschaap, 2022-12-15:
### Have you read the docs?

Yes

### macOS Version

MacOS 13.1

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel 9560AC (0xa370)

### Description

Hello, Thanks for the great work! I use the compiled kext without issues on MacOS 13. 
However, I compile my kexts using the tool "Lilu and Friends" from CorpNewt. Compilation fails with MacOS 13.0.1/Xcode14.0 and MacOS13.1/Xcode14.2. With Monterey and Xcode 13 it compiles fine, as tested by @CorpNewt. Our interaction and some screenshots can be found here: [https://github.com/corpnewt/Lilu-and-Friends/issues/42](url)

### Info in System Report - Bluetooth

```shell
N.A.
```

### Relevant log output

```shell
N.A.
```

pigworlds, 2022-12-16:
I have no problem building this kext under Xcode 14.2 MacOS 13.1. All builds fine.

williambj1, 2022-12-16:
All targets seem to compile without issues for me with macOS 13.0.1 (22A400) + Xcode 14.1 (14B47b). @mhschaap Could you try the following steps?

```sh
git clone --depth=1 https://github.com/OpenIntelWireless/IntelBluetoothFirmware
git clone --depth=1 https://github.com/Acidanthera/Lilu
git clone --depth=1 https://github.com/Acidanthera/MacKernelSDK
cd Lilu
ln -s ../MacKernelSDK ./MacKernelSDK
xcodebuild -target Lilu -configuration Debug
ln -s build/Debug/Lilu.kext ../Lilu.kext
cd ..
cd IntelBluetoothFirmware
ln -s ../Lilu.kext ./Lilu.kext
ln -s ../MacKernelSDK ./MacKernelSDK
xcodebuild -alltargets -configuration Debug
open build/Debug
```

corpnewt, 2022-12-16:
Should be fixed as of [this commit](https://github.com/corpnewt/Lilu-and-Friends/commit/e30eaa418a7ff4d1801753f42c69a46140836216) to Lilu and Friends, thanks @pigworlds 

-CorpNewt

closed 2022-12-16 by williambj1

mhschaap, 2022-12-16:
> ~@mhschaap Could you try the following steps?~

Many thanks to all of you! I would have been happy to follow these steps. However, it was night here, and you solved it meanwhile. Great combined effort!