OpenIntelWireless/IntelBluetoothFirmware #427: Crash on shutdown (auto restart)

firedevel, 2023-02-09:
### Have you read the docs?

No

### macOS Version

macOS 12.6.2

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

Intel 7260 (0x08B1)

### Description

If it cannot shut down normally, click Shut Down and it will restart

### Info in System Report - Bluetooth

```shell
。
```

### Relevant log output

```shell
。
```

### Kernel Panic Logs

```shell
panic(cpu 0 caller 0xffffff8016fcfa67): Double fault at 0xffffff8016e99149, registers:
CR0: 0x000000008001003b, CR2: 0xfffffffffff6fff8, CR3: 0x00000000227f2000, CR4: 0x00000000001626e0
RAX: 0xfffffffffff700a0, RBX: 0xfffffffffff70190, RCX: 0x00000000ffffffff, RDX: 0x00000000fffffffe
RSP: 0xfffffffffff70000, RBP: 0xfffffffffff700d0, RSI: 0xfffffffffff70190, RDI: 0x0000000000000032
R8:  0x0000000000000000, R9:  0x0000000000000001, R10: 0xffffff8017768780, R11: 0x0000000000000010
R12: 0xfffffffffffffffe, R13: 0xfffffffffff7009e, R14: 0xffffff801bfa78ed, R15: 0xffffff8017483dd0
RFL: 0x0000000000010283, RIP: 0xffffff8016e99149, CS:  0x0000000000000008, SS:  0x0000000000000000
Error code: 0x0000000000000000
 @trap_native.c:168
Panicked task 0xffffff95917ac670: 152 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff9a5d9f7540, Frame : Return Address
0xffffff8016d0c240 : 0xffffff8016e7cead 
0xffffff8016d0c290 : 0xffffff8016fdf5d6 
0xffffff8016d0c2d0 : 0xffffff8016fce943 
0xffffff8016d0c320 : 0xffffff8016e1ca70 
0xffffff8016d0c340 : 0xffffff8016e7d27d 
0xffffff8016d0c460 : 0xffffff8016e7ca36 
0xffffff8016d0c4c0 : 0xffffff8017714ed3 
0xffffff8016d0c5b0 : 0xffffff8016fcfa67 
0xffffff8016d0c6c0 : 0xffffff801771ccb8 
0xffffff8016d0c6d0 : 0xffffff8016e1d2ef 
0xfffffffffff700d0 : 0xffffff8017483e97 
0xfffffffffff701c0 : 0xffffff801bfa54fd 
0xfffffffffff70220 : 0xffffff801bfa71b6 
0xfffffffffff70290 : 0xffffff801bfa72e7 
0xfffffffffff70300 : 0xffffff801bfa72e7 
0xfffffffffff70370 : 0xffffff801bfa72e7 
0xfffffffffff703e0 : 0xffffff801bfa72e7 
0xfffffffffff70450 : 0xffffff801bfa72e7 
0xfffffffffff704c0 : 0xffffff801bfa72e7 
0xfffffffffff70530 : 0xffffff801bfa72e7 
0xfffffffffff705a0 : 0xffffff801bfa72e7 
0xfffffffffff70610 : 0xffffff801bfa72e7 
0xfffffffffff70680 : 0xffffff801bfa72e7 
0xfffffffffff706f0 : 0xffffff801bfa72e7 
0xfffffffffff70760 : 0xffffff801bfa72e7 
0xfffffffffff707d0 : 0xffffff801bfa72e7 
0xfffffffffff70840 : 0xffffff801bfa72e7 
0xfffffffffff708b0 : 0xffffff801bfa72e7 
0xfffffffffff70920 : 0xffffff801bfa72e7 
0xfffffffffff70990 : 0xffffff801bfa72e7 
0xfffffffffff70a00 : 0xffffff801bfa72e7 
0xfffffffffff70a70 : 0xffffff801bfa72e7 
0xfffffffffff70ae0 : 0xffffff801bfa72e7 
0xfffffffffff70b50 : 0xffffff801bfa72e7 
0xfffffffffff70bc0 : 0xffffff801bfa72e7 
0xfffffffffff70c30 : 0xffffff801bfa72e7 
0xfffffffffff70ca0 : 0xffffff801bfa72e7 
0xfffffffffff70d10 : 0xffffff801bfa72e7 
0xfffffffffff70d80 : 0xffffff801bfa72e7 
0xfffffffffff70df0 : 0xffffff801bfa72e7 
0xfffffffffff70e60 : 0xffffff801bfa72e7 
0xfffffffffff70ed0 : 0xffffff801bfa72e7 
0xfffffffffff70f40 : 0xffffff801bfa72e7 
0xfffffffffff70fb0 : 0xffffff801bfa72e7 
0xfffffffffff71020 : 0xffffff801bfa72e7 
0xfffffffffff71090 : 0xffffff801bfa72e7 
0xfffffffffff71100 : 0xffffff801bfa72e7 
0xfffffffffff71170 : 0xffffff801bfa72e7 
0xfffffffffff711e0 : 0xffffff801bfa72e7 
0xfffffffffff71250 : 0xffffff801bfa72e7 
0xfffffffffff712c0 : 0xffffff801bfa72e7 
0xfffffffffff71330 : 0xffffff801bfa72e7 
0xfffffffffff713a0 : 0xffffff801bfa72e7 
0xfffffffffff71410 : 0xffffff801bfa72e7 
0xfffffffffff71480 : 0xffffff801bfa72e7 
0xfffffffffff714f0 : 0xffffff801bfa72e7 
0xfffffffffff71560 : 0xffffff801bfa72e7 
0xfffffffffff715d0 : 0xffffff801bfa72e7 
0xfffffffffff71640 : 0xffffff801bfa72e7 
0xfffffffffff716b0 : 0xffffff801bfa72e7 
0xfffffffffff71720 : 0xffffff801bfa72e7 
0xfffffffffff71790 : 0xffffff801bfa72e7 
0xfffffffffff71800 : 0xffffff801bfa72e7 
0xfffffffffff71870 : 0xffffff801bfa72e7 
0xfffffffffff718e0 : 0xffffff801bfa72e7 
0xfffffffffff71950 : 0xffffff801bfa72e7 
0xfffffffffff719c0 : 0xffffff801bfa72e7 
0xfffffffffff71a30 : 0xffffff801bfa72e7 
0xfffffffffff71aa0 : 0xffffff801bfa72e7 
0xfffffffffff71b10 : 0xffffff801bfa72e7 
0xfffffffffff71b80 : 0xffffff801bfa72e7 
0xfffffffffff71bf0 : 0xffffff801bfa72e7 
0xfffffffffff71c60 : 0xffffff801bfa72e7 
0xfffffffffff71cd0 : 0xffffff801bfa72e7 
0xfffffffffff71d40 : 0xffffff801bfa72e7 
0xfffffffffff71db0 : 0xffffff801bfa72e7 
0xfffffffffff71e20 : 0xffffff801bfa72e7 
0xfffffffffff71e90 : 0xffffff801bfa72e7 
0xfffffffffff71f00 : 0xffffff801bfa72e7 
0xfffffffffff71f70 : 0xffffff801bfa72e7 
	Backtrace continues...
      Kernel Extensions in backtrace:
         com.zxystd.IntelBTPatcher(2.3)[D1B747D5-0A22-322C-9CBE-12FBEBE27092]@0xffffff801bfa4000->0xffffff801bfb2fff
            dependency: as.vit9696.Lilu(1.6.3)[11494D76-D048-3265-AE83-DF430492AD93]@0xffffff801ae3c000->0xffffff801ae67fff
            dependency: com.apple.iokit.IOUSBHostFamily(1.2)[20EC3681-15C9-332F-98E7-941C5AEA0253]@0xffffff8019da6000->0xffffff8019e37fff

Process name corresponding to current thread (0xffffff9a5d9f7540): kernel_task
Boot args: gfxrst=1 -igfxbls

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 21.6.0: Sun Nov  6 23:31:16 PST 2022; root:xnu-8020.240.14~1/RELEASE_X86_64
Kernel UUID: F831C932-31CA-3F2C-9E4C-2304C5641587
KernelCache slide: 0x0000000016c00000
KernelCache base:  0xffffff8016e00000
Kernel slide:      0x0000000016c10000
Kernel text base:  0xffffff8016e10000
__HIB  text base: 0xffffff8016d00000
System model name: MacBookPro11,5 (Mac-06F11F11946D27C5)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization
Hibernation exit count: 0

System uptime in nanoseconds: 6026370519
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x0000000167332494
  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000
  Wake    : 0x0000000000000000 0x0000000c1c7b4f1f 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
  Zone map: 0xffffff80c3f31000 - 0xffffffa0c3f31000
  . PGZ   : 0xffffff80c3f31000 - 0xffffff80c5f32000
  . VM    : 0xffffff80c5f32000 - 0xffffff8592731000
  . RO    : 0xffffff8592731000 - 0xffffff872bf31000
  . GEN0  : 0xffffff872bf31000 - 0xffffff8bf8731000
  . GEN1  : 0xffffff8bf8731000 - 0xffffff90c4f31000
  . GEN2  : 0xffffff90c4f31000 - 0xffffff9591731000
  . GEN3  : 0xffffff9591731000 - 0xffffff9a5df31000
  . DATA  : 0xffffff9a5df31000 - 0xffffffa0c3f31000
  Metadata: 0xffffff807df21000 - 0xffffff809df21000
  Bitmaps : 0xffffff809df21000 - 0xffffff80a3f21000
```

usr-sse2, 2023-03-07:
I got a similar panic on wake, but I'm on 2.2.0:
```json
{"bug_type":"210","timestamp":"2023-03-07 20:01:19.00 +0300","os_version":"macOS 13.2.1 (22D68)","roots_installed":0,"incident_id":"1D1ADB26-C022-4CEC-9477-BA76608A88A2"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBwYW5pYyBzdGFja3Nob3QgZm91bmQgaW4gY29yZWZpbGU=","macOSPanicString":"panic(cpu 4 caller 0xffffff801c749357): Double fault at 0xffffff801c60b464, registers:\nCR0: 0x000000008001003b, CR2: 0xffffffb4701fffb8, CR3: 0x000000002ade1000, CR4: 0x00000000003626e0\nRAX: 0xffffffb4702000a0, RBX: 0xffffff905b3fb600, RCX: 0xffffffb470200150, RDX: 0xffffff801cb496a0\nRSP: 0xffffffb4701fffc0, RBP: 0xffffffb470200090, RSI: 0xffffffb470200160, RDI: 0xffffff80206118ea\nR8:  0x000000000000000a, R9:  0x0000000000000000, R10: 0xffffff80206118ea, R11: 0x0000000000000002\nR12: 0xffffff8b8f7ed980, R13: 0xffffff905b3fb600, R14: 0xffffff8b8f7ed9f0, R15: 0xffffff95283aae01\nRFL: 0x0000000000010286, RIP: 0xffffff801c60b464, CS:  0x0000000000000008, SS:  0x0000000000000000\nError code: 0x0000000000000000\n @trap_native.c:168\nPanicked task 0xffffff952815b208: 212 threads: pid 0: kernel_task\nBacktrace (CPU 4), panicked thread: 0xffffff99f43140c8, Frame : Return Address\n0xffffff801c30e270 : 0xffffff801c5eb38d mach_kernel : _handle_debugger_trap + 0x4ad\n0xffffff801c30e2c0 : 0xffffff801c758ed6 mach_kernel : _kdp_i386_trap + 0x116\n0xffffff801c30e300 : 0xffffff801c748120 mach_kernel : _kernel_trap + 0x3e0\n0xffffff801c30e350 : 0xffffff801c585951 mach_kernel : _return_from_trap + 0xc1\n0xffffff801c30e370 : 0xffffff801c5eb66d mach_kernel : _DebuggerTrapWithState + 0x5d\n0xffffff801c30e460 : 0xffffff801c5ead19 mach_kernel : _panic_trap_to_debugger + 0x1a9\n0xffffff801c30e4c0 : 0xffffff801cde072b mach_kernel : _panic + 0x84\n0xffffff801c30e5b0 : 0xffffff801c749357 mach_kernel : _sync_iss_to_iks_unconditionally + 0x167\n0xffffff801c30e6c0 : 0xffffff801cde8898 mach_kernel : _panic_double_fault64 + 0x27\n0xffffff801c30e6d0 : 0xffffff801c585e4f mach_kernel : _hndl_double_fault + 0xf\n0xffffffb470200090 : 0xffffff801cb49767 mach_kernel : _snprintf + 0xa7\n0xffffffb470200180 : 0xffffff802060f51d com.zxystd.IntelBTPatcher : __Z15_hexDumpHCIDataPhm + 0x8d\n0xffffffb4702001e0 : 0xffffff80206111b6 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0xd6\n0xffffffb470200250 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702002c0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200330 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702003a0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200410 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200480 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702004f0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200560 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702005d0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200640 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702006b0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200720 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200790 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200800 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200870 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702008e0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200950 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb4702009c0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200a30 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200aa0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200b10 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200b80 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200bf0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200c60 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200cd0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200d40 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200db0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200e20 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200e90 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470200f00 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : _ [... cut ...] pletionPvS_ij + 0x207\n0xffffffb470201de0 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207\n0xffffffb470201e50 : 0xffffff80\n\n"}
```

Unfortunately, I haven't copied the log from panic report window, so I have it only in JSON format.

Note the repeating line
```
0xffffffb470201590 : 0xffffff80206112e7 com.zxystd.IntelBTPatcher : __ZL17asyncIOCompletionPvS_ij + 0x207
```
I've posted into this issue because it's also infinite recursion, so it may be the same.

zxystd, 2023-06-09:
Fixed in latest code.

closed 2023-06-17 by zxystd