OpenIntelWireless/IntelBluetoothFirmware #432: After turning off bluetooth, can not turn on it .

ouonet, 2023-03-05:
### Have you read the docs?

Yes

### macOS Version

mac os 12

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

intel 9560AC

### Description

After turning off bluetooth, can not turn on it . 

### Info in System Report - Bluetooth

```shell
...
```

### Relevant log output

```shell
sudo dmesg | grep -i bluetooth
[ 3135.345111]: System Policy: mdwrite(633) deny(1) file-write-xattr /System/Library/PreferencePanes/Bluetooth.prefPaneitlwm: iwm_ampdu_tx_done sending bar ssn=3215 tid=0
[ 3382.714728]: 002423.434709 bluetoothd@: AppleUSBHostUserClient::openGated: could not open provider USB2.0-CRW. provider already opened for exclusive access by a kernel client
[ 3382.741068]: **** [IOBluetoothFamily][newUserClient] -- created userclient for HIDShim
[ 3382.741124]: **** [IOBluetoothFamily][AddHCIEventNotification] -- Added notification port for HIDShim
[ 3382.741201]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created input shared memory for HIDShim
[ 3382.741221]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created output shared memory for HIDShim
[ 3382.787217]: 002423.507198 bluetoothd@: AppleUSBHostUserClient::openGated: could not open provider USB2.0-CRW. provider already opened for exclusive access by a kernel client
[ 3382.804449]: 002423.524430 bluetoothd@: AppleUSBHostUserClient::openGated: could not open provider USB2.0-CRW. provider already opened for exclusive access by a kernel client
[ 3382.828841]: Sandbox apply: bluetoothd[1226] <bytes>AppleKeyStore:7456:1226: unexpected session: 100000 uid: -1 requested by: 1226
[ 3382.841327]: Sandbox: bluetoothd(1226) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plist1 duplicate report for Sandbox: bluetoothd(1226) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: bluetoothd(1226) deny(1) mach-lookup com.apple.network.EAPOLControllerSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: locationd(329) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.systemSandbox: locationd(329) deny(1) mach-lookup com.apple.tccd.systemSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: locationd(329) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.systemSandbox: locationd(329) deny(1) mach-lookup com.apple.tccd.systemPMRD: assertionsUser 0x40->0x41
[ 3382.919250]: 002423.639231 bluetoothd@: AppleUSBHostUserClient::openGated: could not open provider USB2.0-CRW. provider already opened for exclusive access by a kernel client
[ 3382.936136]: 002423.656117 bluetoothd@: AppleUSBHostUserClient::openGated: could not open provider USB2.0-CRW. provider already opened for exclusive access by a kernel client
[ 3383.004709]: Sandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.systemSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.systemSandbox: bluetoothd(1226) deny(1) mach-lookup com.apple.WirelessCoexManagerPMRD: setAggressiveness(0) kPMMinutesToSleep = 0
[ 3383.018848]: Sandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.systemSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(107) deny(1) mach-lookup com.apple.tccd.system**** [IOBluetoothFamily][RemoveHCIEventNotification] -- Removed notification port for HIDShim
[ 3383.132490]: **** [IOBluetoothFamily][RemoveHIDShimSharedMemoryRingBuffers] -- Removed input shared memory for HIDShim
[ 3383.132496]: **** [IOBluetoothFamily][RemoveHIDShimSharedMemoryRingBuffers] -- Removed output shared memory for HIDShim
[ 3383.132618]: **** [IOBluetoothFamily][DetachUserClients] -- HIDShim userclient has been terminated
```

Kwokping, 2023-04-04:
I have the same issue on Macos 13.3

LucaBarrella, 2023-04-04:
Same... you can fix (temporally) with this command "sudo pkill bluetoothd"

ouonet, 2023-04-08:
> Same... you can fix (temporally) with this command "sudo pkill bluetoothd"

I tried it , but doesn't work.

LucaBarrella, 2023-04-09:
Hi, unfortunately you have to do this every time you intend to reactivate the bluetooth, potentially you could create a script that turns off the bluetooth when the pc goes to sleep, which then reactivates it when the pc wakes up, with 12.2.1 you could use "bluesnooze".

LucaBarrella, 2023-04-09:
> > Same... you can fix (temporally) with this command "sudo pkill bluetoothd"
> 
> I tried it , but doesn't work.

Hi, unfortunately you have to do this every time you intend to reactivate the bluetooth, potentially you could create a script that turns off the bluetooth when the pc goes to sleep, which then reactivates it when the pc wakes up and restores it with 12.2.1 you could use "bluesnooze".

xjh169, 2023-04-17:
我更新 13.4 Beta 版(22F5037d) 上有同样的问题。（扩展版 v2.3.0）

RaDianTBLaZe, 2023-04-18:
> > > Same... you can fix (temporally) with this command "sudo pkill bluetoothd"
> 
> > 
> 
> > I tried it , but doesn't work.
> 
> 
> 
> Hi, unfortunately you have to do this every time you intend to reactivate the bluetooth, potentially you could create a script that turns off the bluetooth when the pc goes to sleep, which then reactivates it when the pc wakes up and restores it. In 12.2.1 you could use "bluesnooze" or hibernation 25.

My Bluetooth connects and disconnects immediately please help if you can , it works fine with some headphones but not with some .

VolokitaBoy, 2023-04-19:
I also have a Bluetooth when disconnected does not turn on at macos 13.3.1. On Monterey everything instantly turns off-turns on..AX201 intel.I also noticed. that the bluetooth module has stopped being displayed in the hackintool in the devices section
[image]

RaDianTBLaZe, 2023-04-21:
> I also have a Bluetooth when disconnected does not turn on at macos 13.3.1. On Monterey everything instantly turns off-turns on..AX201 intel.I also noticed. that the bluetooth module has stopped being displayed in the hackintool in the devices section [image]

i found out that it can be  issue with monterey and cant be fixed only way to make bluetooth work is going back to big sur , also the kext doesnt have any patches for ax201 , it has only for ax200 and ax210 . plus  an advice dont use intelbtpatcher .

VolokitaBoy, 2023-04-21:
> > У меня также есть Bluetooth, который при отключении не включается в macos 13.3.1. На Monterey все мгновенно отключается-включается ..AX201 intel.Я также заметил. что модуль Bluetooth перестал отображаться в hackintool в разделе устройств [image]
> 
> я узнал, что это может быть проблема с monterey и не может быть исправлено, единственный способ заставить Bluetooth работать - вернуться к big sur, также в kext нет исправлений для ax201, у него есть только для ax200 и ax210. плюс совет не использовать intelbtpatcher.

That's it!There are no problems at all in Monterey, everything works fine.And in Ventura, crutches

closed 2023-06-09 by zxystd