OpenIntelWireless/IntelBluetoothFirmware #433: 升级至Macos13.3后，intel蓝牙驱动异常

Kwokping, 2023-04-04:
### Have you read the docs?

Yes

### macOS Version

macOS 13.3

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel ax 200

### Description

升级至Macos13.3后，intel蓝牙驱动异常，启动后蓝牙设备无法在Hackintosh软件上识别，但能正常使用，但是关闭蓝牙后再打开蓝牙开关，蓝牙设备就无法正常使用。

### Info in System Report - Bluetooth

```shell
ranslated Report (Full Report Below)
-------------------------------------

Process:               bluetoothd [139]
Path:                  /usr/sbin/bluetoothd
Identifier:            bluetoothd
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               0

Date/Time:             2023-04-03 15:28:59.2552 +0800
OS Version:            macOS 13.3 (22E252)
Report Version:        12
Anonymous UUID:        597FDB42-E4F7-94DE-6EA6-CB6AF26BCF80

Time Awake Since Boot: 12 seconds

System Integrity Protection: enabled

Crashed Thread:        2

Exception Type:        EXC_GUARD
Exception Codes:       GUARD_TYPE_USER
Exception Codes:       0x6000000000000012, 0x0000000000000002

Termination Reason:    Namespace LIBSYSTEM, Code 2 Application Triggered Fault

Thread 0:
0   libsystem_kernel.dylib        	    0x7ff8169dcd2e __ulock_wait + 10
1   libsystem_platform.dylib      	    0x7ff816a45b07 _os_unfair_lock_lock_slow + 162
2   libobjc.A.dylib               	    0x7ff816687452 _finishInitializing(objc_class*, objc_class*) + 98
3   libobjc.A.dylib               	    0x7ff81668733c lockAndFinishInitializing(objc_class*, objc_class*) + 297
4   libobjc.A.dylib               	    0x7ff816686ebf initializeNonMetaClass + 831
5   libobjc.A.dylib               	    0x7ff81669b96b initializeAndMaybeRelock(objc_class*, objc_object*, locker_mixin<lockdebug::lock_mixin<objc_lock_base_t>>&, bool) + 221
6   libobjc.A.dylib               	    0x7ff8166868e1 lookUpImpOrForward + 778
7   libobjc.A.dylib               	    0x7ff81668601b _objc_msgSend_uncached + 75
8   bluetoothd                    	       0x105ae4d70 0x105abb000 + 171376
9   dyld                          	    0x7ff8166c041f start + 1903

Thread 1:
0   libsystem_kernel.dylib        	    0x7ff8169dd882 __psynch_mutexwait + 10
1   libsystem_pthread.dylib       	    0x7ff816a17b14 _pthread_mutex_firstfit_lock_wait + 76
2   libsystem_pthread.dylib       	    0x7ff816a15921 _pthread_mutex_firstfit_lock_slow + 214
3   libobjc.A.dylib               	    0x7ff816686bec initializeNonMetaClass + 108
4   libobjc.A.dylib               	    0x7ff81669b96b initializeAndMaybeRelock(objc_class*, objc_object*, locker_mixin<lockdebug::lock_mixin<objc_lock_base_t>>&, bool) + 221
5   libobjc.A.dylib               	    0x7ff8166868e1 lookUpImpOrForward + 778
[... 291 lines cut ...]
### Relevant log output

```shell
[    1.124779]: IntelFirmware: Driver init()
[    1.124788]: IntelFirmware: Driver Probe()
[    1.124795]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0029
[    1.124807]: IntelFirmware: Driver Start()
[    1.124857]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    1.124888]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    1.124908]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    1.124916]: IntelFirmware: set configuration to 1
[    1.125354]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    1.125364]: IntelFirmware: Found interface!!!
[    1.125384]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    1.125387]: IntelFirmware: Found Interrupt endpoint!
[    1.125641]: IntelFirmware: Found Bulk out endpoint!
[    1.125908]: IntelFirmware: Found Bulk in endpoint!
[    1.127095]: IntelFirmware: BT init succeed
[    1.128034]: IntelFirmware: Read the legacy Intel version information
[    1.128039]: IntelFirmware: Firmware revision 0.0 build 111 week 53 2019
[    1.128046]: IntelFirmware: Frimware is already running, finishing
[    1.129028]: IntelFirmware: Clean up...
[    1.129032]: IntelFirmware: virtual void BtIntel::free()
[    1.129035]: IntelFirmware: virtual void USBDeviceController::free()
```

LucaBarrella, 2023-04-06:
Hi, have you fixed?

Kwokping, 2023-04-06:
@LucaBarrella NO,wait for the kext update.

LucaBarrella, 2023-04-06:
Thx, I’m back to 13.2.1 for this reason.
@LucaBarrella NO,wait for the kext update.

anonymous-writer, 2023-04-08:
May be the same error even if with completely different hardware.
https://github.com/acidanthera/bugtracker/issues/2255

Kwokping, 2023-04-11:
这是重新打开蓝牙时的错误日志。
错误	15:48:37.642112+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:37.642164+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:37.646744+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:37.646804+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:38.718182+0800	cfprefsd	Couldn't open parent path due to [2: No such file or directory]
错误	15:48:38.727705+0800	cfprefsd	Couldn't open parent path due to [2: No such file or directory]
错误	15:48:39.438774+0800	symptomsd	Ignoring source 0x7fd529a32090, snapshot is from kernel kernel_task UDPSUBFLOW4 flow id 12958 (close) so first 1 pkts rx 1 tx 0, bytes 205 0 cell 0 0 wifi 0 0 wired 0 0 deltas 205 0 0 0 0 0 0 0
错误	15:48:39.441135+0800	symptomsd	Ignoring source 0x7fd529934b90, snapshot is from kernel kernel_task UDPSUBFLOW4 flow id 12856 (close) so first 1 pkts rx 9 tx 3, bytes 1301 1044 cell 0 0 wifi 0 0 wired 0 0 deltas 1301 1044 0 0 0 0 0 0
错误	15:48:39.442543+0800	symptomsd	Ignoring source 0x7fd529934b90, snapshot is from kernel kernel_task UDPSUBFLOW6 flow id 12959 (close) so first 1 pkts rx 2 tx 1, bytes 324 348 cell 0 0 wifi 0 0 wired 0 0 deltas 324 348 0 0 0 0 0 0
错误	15:48:39.443718+0800	symptomsd	Ignoring source 0x7fd529934b90, snapshot is from kernel kernel_task UDPSUBFLOW6 flow id 12857 (close) so first 1 pkts rx 9 tx 0, bytes 1301 0 cell 0 0 wifi 0 0 wired 0 0 deltas 1301 0 0 0 0 0 0 0
错误	15:48:39.541999+0800	cfprefsd	Couldn't open parent path due to [2: No such file or directory]
错误	15:48:39.659899+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:39.659961+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:39.664896+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:39.664948+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:39.757877+0800	cfprefsd	Couldn't open parent path due to [2: No such file or directory]
错误	15:48:40.673918+0800	kernel	11 duplicate reports for Sandbox: Family(775) deny(1) mach-lookup com.apple.contactsd.persistence
错误	15:48:40.673933+0800	kernel	Sandbox: com.apple.WebKit.WebContent(927) deny(1) mach-lookup com.apple.diagnosticd
错误	15:48:40.674756+0800	kernel	Sandbox: Family(775) deny(1) mach-lookup com.apple.contactsd.persistence
错误	15:48:40.700844+0800	Family	Persistent store service connection invalidated: failed at lookup with error 159 - Sandbox restriction
错误	15:48:40.708945+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:40.709023+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:40.713732+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:40.713777+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:40.715559+0800	Family	Error communicating with persistent store service proxy: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named com.apple.contactsd.persistence was invalidated: failed at lookup with error 159 - Sandbox restriction." UserInfo={NSDebugDescription=The connection to service named com.apple.contactsd.persistence was invalidated: failed at lookup with error 159 - Sandbox restriction.}
错误	15:48:40.715874+0800	Family	Error connecting to remote endpoint: (null)
错误	15:48:40.716159+0800	Family	fault: Unable to create token NSXPCConnection.  NSXPCStoreServerEndpointFactory 0x7f790f74a8d0 -newEndpoint returned nil
故障	15:48:40.716728+0800	Family	CoreData: Unable to create token NSXPCConnection.  NSXPCStoreServerEndpointFactory 0x7f790f74a8d0 -newEndpoint returned nil
错误	15:48:40.724126+0800	Family	error: Failed to create NSXPCConnection
错误	15:48:40.740134+0800	bluetoothd	LMP Routing is disabled
错误	15:48:40.735213+0800	BlueTool	Could not find btaddress param
故障	15:48:40.742433+0800	bluetoothd	Unknown chipset 0
错误	15:48:40.743499+0800	bluetoothd	Failed to register the vendor callback with result 101
错误	15:48:40.743650+0800	bluetoothd	Core Stack initialization failed!
错误	15:48:40.744061+0800	bluetoothd	Stack startup failed - 110
错误	15:48:40.765975+0800	bluetoothd	Mac Cache Migrator Received PowerStateChanged
错误	15:48:40.746284+0800	BlueTool	Could not find btaddress param
错误	15:48:40.797415+0800	Family	Persistent store service connection invalidated: failed at lookup with error 159 - Sandbox restriction
错误	15:48:40.798616+0800	Family	Error communicating with persistent store service proxy: Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named com.apple.contactsd.persistence was invalidated: failed at lookup with error 159 - Sandbox restriction." UserInfo={NSDebugDescription=The connection to service named com.apple.contactsd.persistence was invalidated: failed at lookup with error 159 - Sandbox restriction.}
[... cut ...]
错误	15:48:42.707612+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:42.707669+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)
错误	15:48:42.711479+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(-1) returned -1 22 (Invalid argument)
错误	15:48:42.711535+0800	runningboardd	memorystatus_control error: MEMORYSTATUS_CMD_CONVERT_MEMLIMIT_MB(0) returned -1 22 (Invalid argument)

zxystd, 2023-06-01:
https://github.com/acidanthera/BrcmPatchRAM

closed 2023-06-01 by zxystd