OpenIntelWireless/IntelBluetoothFirmware #434: 蓝牙关闭之后 无法再次 开启

xjh169, 2023-04-17:
### Have you read the docs?

No

### macOS Version

macOS 13.4 Beta版(22F5037d)

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

英特尔无线-AC 8265 （0x004C）

### Description

小米笔记本 Pro 2019 i7-8550U  MX250  KBL，蓝牙关闭之后 无法再次 开启。

（如果在控制中心那里把蓝牙 开关 打开 关机重启 也是无法打开蓝牙的，出现蓝牙无法打开的情况下 需要把控制中心的蓝牙开关 关闭， 重启才能打开蓝牙，）

### Info in System Report - Bluetooth

```shell
无
```

### Relevant log output

```shell
无
```

xjh169, 2023-04-17:
https://user-images.githubusercontent.com/28785981/232403202-442f874a-b842-48a9-b2b8-ab4fd8c385c7.mov

Kwokping, 2023-04-18:
同样情况

gaganmalvi, 2023-04-18:
`sudo pkill bluetoothd`should work as a workaround (which is what I do). Beta 2 is pretty stable except the occasional black screen that keeps happening on Mi Notebook 14 CML i3-10110U.

zxystd, 2023-05-30:
`BluetoolFixup.kext` should be updated, here is the updated one you can test(Only the `BluetoolFixup.kext` is needed).
https://github.com/zxystd/BrcmPatchRAM/suites/13231677116/artifacts/720906228

Kwokping, 2023-05-30:
@zxystd the BluetoolFixup.kext isn't work

xjh169, 2023-05-31:
> `BluetoolFixup.kext`应该更新，这是您可以测试的更新版本（仅`BluetoolFixup.kext`需要）。 https://github.com/zxystd/BrcmPatchRAM/suites/13231677116/artifacts/720906228

结合
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/435 我的问题蓝牙问题已经解决！谢谢🙏

closed 2023-05-31 by xjh169

malash, 2023-06-06:
This PR may work https://github.com/acidanthera/BrcmPatchRAM/pull/28