OpenIntelWireless/IntelBluetoothFirmware #435: 你好，我更新测试13.4第4测试版本后我的蓝牙无法启动，我有蓝牙型号是ax201,在第2测试版本使用没有问题，升级测试版本3和4后就是蓝牙无法启动

553051957, 2023-05-05:
### Have you read the docs?

Yes

### macOS Version

macos13

### Kext Version

v2.20

### Wireless Adapter Model and USB Product ID

intel ax201

### Description

你好，我更新测试13.4第4测试版本后我的蓝牙无法启动，我有蓝牙型号是ax201,在第2测试版本使用没有问题，升级测试版本3和4后就是蓝牙无法启动

### Info in System Report - Bluetooth

```shell
更新测试测试版本3和4后，蓝牙无法使用，在没有更新测试版本2以前没有任何的问题
```

### Relevant log output

```shell
无
```

### Kernel Panic Logs

```shell
无
```

xjh169, 2023-05-07:
一样

553051957, 2023-05-07:
> 同样情况

等待作者更新解决

HuaZoffice, 2023-05-09:
同样情况

HuaZoffice, 2023-05-12:
目前升级到了rc2，还是无法启动

xjh169, 2023-05-12:
我直接降级macOS 12，macOS 13 bug太多了

553051957, 2023-05-13:
> 我直接降级macOS 12，macOS 13 bug太多了

我没有怎么用，娱乐办公可以，现在蓝牙用不了

553051957, 2023-05-13:
> 目前升级到了rc2，还是无法启动

一样的情况，我以为更新就是可以启动，看来要等待作者更新

zxystd, 2023-05-13:
理论上IntelBluetoothFirmware.kext就算不更新了都可以一直用到黑苹果灭亡，出现这个问题要么是苹果的问题要么是需要BluetoolFixup.kext更新一些patch，但这已经不属于本仓库的范畴了。

553051957, 2023-05-13:
> 理论上IntelBluetoothFirmware.kext就算不更新了都可以一直用到黑苹果灭亡，出现这个问题要么是苹果的问题要么是需要BluetoolFixup.kext更新一些patch，但这已经不属于本仓库的范畴了。

对的，猜测是这个补丁的原因，苹果修改蓝牙的，BlutoolFixup.kext,当时蓝牙无法启动，你先前不是有这个补丁，IntelBTPatcher.kext

xjh169, 2023-05-13:
> > 理论上IntelBluetoothFirmware.kext就算不更新了都可以直接用到黑苹果灭亡，出现这个问题或者是苹果的问题或者是需要BluetoolFixup.kext更新一些补丁，但是这已经不属于本库的范围了。
> 
> 对的，猜测是这个补丁的原因，苹果修改蓝牙的，BlutoolFixup.kext，当蓝牙无法启动时，你先没有这个补丁，IntelBTPatcher.kext

有IntelBTPatcher.kext 2.6.7 不知道能不能使用。你可以测试一下  https://github.com/acidanthera/BrcmPatchRAM/actions

553051957, 2023-05-13:
> > > 理论上IntelBluetoothFirmware.kext就算不更新了都可以直接用到黑苹果灭亡，出现这个问题或者是苹果的问题或者是需要BluetoolFixup.kext更新一些补丁，但是这已经不属于本库的范围了。
> > 
> > 
> > 对的，猜测是这个补丁的原因，苹果修改蓝牙的，BlutoolFixup.kext，当蓝牙无法启动时，你先没有这个补丁，IntelBTPatcher.kext
> 
> 有BlutoolFixup.kex 2.6.7 不知道能不能使用。你可以测试一下 https://github.com/acidanthera/BrcmPatchRAM/actions

尝试过结果一样，我加入这个补丁，也是无法启动蓝牙，上条回复我打错字，不好意思，在13bete2之前我都是有三个蓝牙驱动
BlutoolFixup.kext ,IntelBluetoothFirmware.kext ,IntelBTPatcher.kext,才是可以驱动我的蓝牙，我的intelAX201网卡，不过也是谢谢你，搞不定，我就是换免驱苹果网卡用着先，到时在看下

553051957, 2023-05-15:
> > > 理论上IntelBluetoothFirmware.kext就算不更新了都可以直接用到黑苹果灭亡，出现这个问题或者是苹果的问题或者是需要BluetoolFixup.kext更新一些补丁，但是这已经不属于本库的范围了。
> > 
> > 
> > 对的，猜测是这个补丁的原因，苹果修改蓝牙的，BlutoolFixup.kext，当蓝牙无法启动时，你先没有这个补丁，IntelBTPatcher.kext
> 
> 有BlutoolFixup.kex 2.6.7 不知道能不能使用。你可以测试一下 https://github.com/acidanthera/BrcmPatchRAM/actions

我尝试下，结果一样，不过比目前好一点，点击开关有反应，就是没有驱动成功蓝牙驱动

HuaZoffice, 2023-05-16:
> 

你好请问下你用的是哪个型号的免驱网卡我也去整一个

HuaZoffice, 2023-05-16:
> 我尝试下，结果一样，不过比目前好一点，点击开关有反应，就是没有驱动成功蓝牙驱动

你好请问你用的是哪个型号的免驱网卡

xjh169, 2023-05-16:
我没有 免驱网卡，我的英特尔自带 Intel Wireless 8265，macOS 12.6.5 (21G531) 一切正常。 @HuaZoffice 
之前在JD.com 买了一个 USB-C免驱 有线网卡 可以接到路由器。

553051957, 2023-05-16:
> 我没用 免驱网卡，我的英特尔自带 Intel Wireless 8265，macOS 12.6.5 (21G531) 一切正常。 @HuaZoffice 之前在JD.com 买了一个 USB-C免驱 有线网卡 可以接到路由器。

我的都在13.3测试版本3和4版本就是蓝牙挂了，你给到我的网站也是只有版本2.6.6版本，没有你说到2.6.7版本，

xjh169, 2023-05-16:
https://github.com/acidanthera/BrcmPatchRAM/actions/runs/4919419340 这个 改动的beta   @553051957

553051957, 2023-05-16:
> https://github.com/acidanthera/BrcmPatchRAM/actions/runs/4919419340 这个 改动的beta @553051957

谢谢，我试下

553051957, 2023-05-16:
> https://github.com/acidanthera/BrcmPatchRAM/actions/runs/4919419340 这个 改动的beta @553051957

结果一样，无法启动蓝牙
[image]
[image]

xjh169, 2023-05-17:
实在没办法，退回macOS 12 吧   😂 @553051957

melxman, 2023-05-18:
我是ax210，某两个beta其间无法使用，但是rc1之前的那个beta莫名的好了。使用的蓝牙相关的kext也没更新过

HuaZoffice, 2023-05-18:
> 我是ax210，某两个beta其间无法使用，但是rc1之前的某个beta莫名的好了。使用的蓝牙相关的kext也没更新过 不过现在唯一的一点问题就是：蓝牙会时不时的断开 最开始是以为我连上看视频，然后视频暂停了一段时间没有声音发出。蓝牙基于某些规则自己断开节电了(我台式机)。但是后来发现是随机的，特别随机，经常听着声音就断开了

清除nvram以后蓝牙应该就不能用了

553051957, 2023-05-19:
> 我是ax210，某两个beta其间无法使用，但是rc1之前的某个beta莫名的好了。使用的蓝牙相关的kext也没更新过 不过现在唯一的一点问题就是：蓝牙会时不时的断开 最开始是以为我连上看视频，然后视频暂停了一段时间没有声音发出。蓝牙基于某些规则自己断开节电了(我台式机)。但是后来发现是随机的，特别随机，经常听着声音就断开了

从rc3以后就是蓝牙无法驱动

553051957, 2023-05-19:
> > 我是ax210，某两个beta其间无法使用，但是rc1之前的某个beta莫名的好了。使用的蓝牙相关的kext也没更新过 不过现在唯一的一点问题就是：蓝牙会时不时的断开 最开始是以为我连上看视频，然后视频暂停了一段时间没有声音发出。蓝牙基于某些规则自己断开节电了(我台式机)。但是后来发现是随机的，特别随机，经常听着声音就断开了
> 
> 清除nvram以后蓝牙应该就不能用了
蓝牙驱动的原因，断续的原因，反正更新rc3后就是无法使用蓝牙

qcTimb3rlandqc, 2023-05-20:
[image]
try that

553051957, 2023-05-20:
> [image] try that

ok,i'm try,thank you

qcTimb3rlandqc, 2023-05-20:
lets me no make sur to put data

qcTimb3rlandqc, 2023-05-20:
;)

553051957, 2023-05-20:
> lets me no make sur to put data

[image]
Try the solution you mentioned, and the results are the same. There are no changes. This solution is a solution that affects the OTA Bluetooth patch and upgrades the system, not a Bluetooth driver solution. Bluetooth can be driven in 13.2bete2, but after upgrading bete3, Bluetooth cannot be driven. Thank you, boss

qcTimb3rlandqc, 2023-05-20:
but is working for me when I instal the 13.4 full install so you are sur you have put it to data and is look like is workin for one other guy here is have close the issu

553051957, 2023-05-20:
> but is working for me when I instal the 13.4 full install so you are sur you have put it to data and is look like is workin for one other guy here is have close the issu

ok,This is the final method，thank you very much

Rofocale11, 2023-05-20:
> > [image] try that
> 
> 
> 
> ok,i'm try,thank you

Not work for me on 13.4

Samanosuke2020, 2023-05-21:
[image]
Thank you Working Fine On OC 9.2, macOS 13.4 (22F66), TPLink -UB400

xjh169, 2023-05-21:
我的 macOS 13.4（22F66）OpenCorev0.9.2 ，英特尔 板载蓝牙 BCM_4350C2，还是无法打开🥵
[image]
[image]

553051957, 2023-05-21:
> 我的 macOS 13.4（22F66）OpenCorev0.9.2 ，英特尔 板载蓝牙 BCM_4350C2，还是无法打开🥵 [image] [image]

我尝试下全新安装，目前有2个都是可以驱动的，以前旧驱动影响了，到时我试下全新安装

zxystd, 2023-05-21:
> [image] Try the solution you mentioned, and the results are the same. There are no changes. This solution is a solution that affects the OTA Bluetooth patch and upgrades the system, not a Bluetooth [...]

This is the positive solution. Apple writes this bug though...

qcTimb3rlandqc, 2023-05-21:
<key>bluetoothInternalControllerInfo</key>
				<data>AAAAAAAAAAAAAAAAAAA=</data>
				<key>bluetoothExternalDongleFailed</key>
				<data>AA==</data>

553051957, 2023-05-21:
> 
i'm try, thanks

553051957, 2023-05-21:
> ```
> <key>bluetoothInternalControllerInfo</key>
> 			<data>AAAAAAAAAAAAAAAAAAA=</data>
> 			<key>bluetoothExternalDongleFailed</key>
> 			<data>AA==</data>.    
> ```
> 
> Thx TiMb3rLanD for find the fix hahahaha

My specific wireless model AX200NGW, commonly known as ax201, has the same result. It seems that I need to install a brand new 13.4 system, and currently my MACOS13.5. Thank you for patiently answering my questions, boss

qcTimb3rlandqc, 2023-05-21:
> My specific wireless model AX200NGW, commonly known as ax201, has the same result. It seems that I need to install a brand new 13.4 system, and currently my MACOS13.5. Thank you for patiently answering my questions, boss

You have nvram problem??? Im on 13.5 and i have  the ax201 and that fix it for me first time i update to 13.4 bêta 3 full instal lost my Bluetooth and i have see i lost some Line in the nvram so i instal 13.3.1 copy the Line i lost in nvram then instal 13.4 and this fix the problem so maybe  something or not good in your config or you have nvram problem

553051957, 2023-05-21:
> You have nvram problem??? Im on 13.5 and i have the ax201 and that fix it for me first time i update to 13.4 bêta 3 full instal lost my Bluetooth and i have see i lost some Line in the nvram so i [...]

Yes, I also cannot start from 13.4bete3, I will not find the bug, I am just a user, Apple bugs will not be caught, but thank you, boss, I also cleared the nvram, the result is the same

553051957, 2023-05-21:
> [image]
> try that
Successfully driven, thank you for the perfect solution to the problem I mentioned. I used OpenCore to add the two lines of code that you mentioned were out of sequence. I used OCAuxiliaryTools to correct the error, and my Bluetooth started perfectly
[image]

553051957, 2023-05-21:
> 我的 macOS 13.4（22F66）OpenCorev0.9.2 ，英特尔 板载蓝牙 BCM_4350C2，还是无法打开🥵 [image] [image]

现在可以，顺序要对的，现在我的就是可以，谢谢TiMb3rLanD大神

xjh169, 2023-05-22:
按照 @553051957 你上面说的操作，开机蓝牙会自动打开 也可以正常使用了🤓，但是 关闭蓝牙之后不能再次开启，需要重启它才会自动打开蓝牙。依然有bug,只要我不关闭就正常使用。
[image]
[image]

xjh169, 2023-05-22:
又回到了这个问题 https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/434

553051957, 2023-05-22:
> 又回到了这个问题 #434

这是个bug,没有办法解决，反正我的蓝牙从来不关，不想重启的话，终端输入 sudo pkill bluetoothd

alphavip, 2023-05-22:
> 
关闭无法打开，可以不必重启，用这个命令sudo pkill bluetoothd

ameenjuz, 2023-05-22:
once again bluetooth break on macOS Ventura 13.5 beta 1
this patch   <key>bluetoothInternalControllerInfo</key>
			<data>AAAAAAAAAAAAAAAAAAA=</data>
			<key>bluetoothExternalDongleFailed</key>
			<data>AA==</data>  is not working on Ventura 13.5 beta 1 but it is working on Ventura 13.4
please help me any solution for this

qcTimb3rlandqc, 2023-05-22:
> once again bluetooth break on macOS Ventura 13.5 beta 1
>
> this patch   <key>bluetoothInternalControllerInfo</key>
>
> 			<data>AAAAAAAAAAAAAAAAAAA=</data> [...]
 
Is working for me but i will instal the 13.5 beta 1 full instal for see i will lets you no

ameenjuz, 2023-05-22:
> > once again bluetooth break on macOS Ventura 13.5 beta 1
> > this patch   bluetoothInternalControllerInfo
> > ```
> >   	<data>AAAAAAAAAAAAAAAAAAA=</data>
> > 
> >   	<key>bluetoothExternalDongleFailed</key>
> > 
> >   	<data>AA==</data>  is not working on Ventura 13.5 beta 1 but it is working on Ventura 13.4
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > please help me any solution for this
> 
> Is working for me but i will instal the 13.5 beta 1 full instal for see i will lets you no
Ok thanks i will wait for your reply i hope that you will find solution

qcTimb3rlandqc, 2023-05-23:
> Is working for me but i will instal the 13.5 beta 1 full instal for see i will lets you no

Still work on 13.5 beta 1 (ax201)

553051957, 2023-05-23:
> Is working for me but i will instal the 13.5 beta 1 full instal for see i will lets you no

Explain the model of your network card, mine is 201 and 200 series, both solutions of qcTimb3rlandqc can be used, and both can be driven in 13.4.4 and 13.5.1

553051957, 2023-05-23:
> qcTimb3rlandqc
I have the same type of card, please check your code sequence is correct, use several config files to compare and see, I also succeeded in many network cards of the same type in the vision forum, and I also succeeded, according to the method of qcTimb3rlandqc. buttocks l...

553051957, 2023-05-23:
> cleared the nvram
You can try the first reply code that God gave me, it is 000, there is a problem with the compilation order of other oc, you need to reverse it, remember to clear the nvram after the update

553051957, 2023-05-23:
> > [image] try that
> 
> This is not working for me. I have gigabyte b550 viison d motherboard with Intel® Wi-Fi 6 AX200 and bluetooth 5.0. I am able to turn on the bluetooth but it is not listing any near by bluetooth devices to connect.
> 
> Thanks in advance for the help
After trying to reverse the order，
cleared the nvram

ameenjuz, 2023-05-23:
> > I
> > > > > > > once again bluetooth break on macOS Ventura 13.5 beta 1
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > > this patch   bluetoothInternalControllerInfo
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > ```
> > > > > > > <data>AAAAAAAAAAAAAAAAAAA=</data>
> > > > > > > ```
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > ```
> > > > > > > <key>bluetoothExternalDongleFailed</key>
> > > > > > > ```
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > > > > > ```
> > > > > > > <data>AA==</data>  is not working on Ventura 13.5 beta 1 but it is working on Ventura 13.4
> > > > > > > ```
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > > please help me any solution for this
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > > Is working for me but i will instal the 13.5 beta 1 full instal for see i will lets you no
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > >
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > > Still work on 13.5 beta 1 (ax201)
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > I have same intel wifi card 6 AX201 but still not work for me 13.5 beta 1 but on 13.4 work fine
> > 
> > 
> > > > I also tried -lilubetaal boot arg but nothing
> > 
> > 
> > > > how do you have work on 13.5 beta 1 and I have no working
> > 
> > 
> > > > please reset your nvram and check once again. work or not
> > 
> > 
> > > > I have no luck
> > 
> > 
> > > 
> > 
> > 
> > > Tcheck if you have the 3 kext you neeed the fix work on 13.5 too I have instal the full 13.5 installer + clear the nvram and all is oker for me
> > 
> > 
> > I have already enabled Bluetoolfixup.kext+IntelBluetoothFirmware.kext+IntelBtpatcher.kext  in Config.plist and kexts folder  tried reset nvram  again and again but no success
> 
> CAN you turn off the Bluetooth reboot and see if is work?

it is already turned off and grayed out how to turn off again
I attached my screenshot and config
[config.plist.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11544512/config.plist.zip)
[image]
[image]

qcTimb3rlandqc, 2023-05-23:
> it is already turned off and grayed out how to turn off again
>
> I attached my screenshot and config [...]

CAN i see the nvram section to?

ameenjuz, 2023-05-23:
@qcTimb3rlandqc  attached

I removed AirportItlwm.kext and IntelBluetoothFirmware.kext from EFI Folder due to big size uploading restriction
[EFI.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11547949/EFI.zip)

qcTimb3rlandqc, 2023-05-23:
> @qcTimb3rlandqc attached
> 
> I removed AirportItlwm.kext and IntelBluetoothFirmware.kext from EFI Folder due to big size uploading restriction [EFI.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11547949/EFI.zip)
[
](https://drive.google.com/file/d/1ykGsBwCKJBk7yuBz_J79IjOkb8n9XDEt/view?usp=sharing).     try that

ameenjuz, 2023-05-23:
didn't work your modified EFI 
same problem
remember I updated macOS Ventura 13.4 to 13.5 beta 1 using OTA update 
I didn't update with full installer of 13.5 beta 1
[image]

[25 more replies]

closed 2023-06-01 by zxystd