OpenIntelWireless/IntelBluetoothFirmware #437: Bluetooth is nonfunctional on macOS 14 - Intel AC 7265

meghan06, 2023-06-07:
### Have you read the docs?

Yes

### macOS Version

macOS 14

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel 7265; 0x8087

### Description

on macOS Sonoma, Bluetooth cannot be enabled. The option to enable it is there, but stays disabled.

How to reproduce:
- Install macOS 14
- Go to SysPref, then Bluetooth
- Try to enable BT 

Note: I am unsure whether the USB Product ID is correct. I will upload a IOReg dump if that makes things easier.

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:    NULL
  State:    Off
  Chipset:    BCM_4350C2
  Discoverable:    Off
  Firmware Version:    v0 c0
  Supported services:    0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:    USB
  Vendor ID:    0x004C (Apple)
```

### Relevant log output

```shell
user@Users-Air ~ % sudo dmesg | grep -i IntelBluetooth
Password:
[3.460320]: IOUSBHostDevice@14300000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
user@Users-Air ~ %
```

### Kernel Panic Logs

```shell
n/a
```

meghan06, 2023-06-07:
I do understand it has only been a day since the announcement of Sonoma, but I am throwing this one out here in case others have the same issue.

dekill93, 2023-06-07:
The problem same for me on intel AC 8260,

zxystd, 2023-06-07:
Intel bluetooth drivers can work pretty well on **Any** version, The needed fix is on `bluetoolfixup.kext` side.

BTW, here is my modified version.
[BlueToolFixup.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11672823/BlueToolFixup.kext.zip)

dekill93, 2023-06-07:
> Intel bluetooth drivers can work pretty well on **Any** version, The needed fix is on `bluetoolfixup.kext` side.
> 
> BTW, here is my modified version. [BlueToolFixup.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11672823/BlueToolFixup.kext.zip)

I try this kext, i can scan an turn on the bluetooth, but i can’t connect bluetooth on my iphone.

zxystd, 2023-06-07:
> I try this kext, i can scan an turn on the bluetooth, but i can’t connect bluetooth on my iphone.

so it is working. You can google why iphone can not pair with macOS.

wuhanxiesong, 2023-06-07:
[image]

qcTimb3rlandqc, 2023-06-07:
> so it is working. You can google why iphone can not pair with macOS.

hello for the kext wifi is that you only have to change the ID for sonoma or it has more repairs to make thank you

meghan06, 2023-06-07:
@zxystd Thanks for the fix. Will test later. Do you have any plans on making a AirportItlwm fork for Sonoma any time soon?

meghan06, 2023-06-07:
@zxystd Got it. Although in 2-3 years, Intel would probably be dropped from macOS altogether.

If development for Sonoma begins, feel free to let me know if you need testing. I'd be happy to do so.

qcTimb3rlandqc, 2023-06-08:
> @qcTimb3rlandqc @meghan06 Apple delete the whole legacy 80211 stack on Sonoma, I am going to turn to the other IO80211Family which is using by macintosh of late 2018(although it will be also deleted in 2~3 years), it is a big work, almost all of the airport interfaces changed, need to start reversed over again.
> 
> [image]

perfect thx

meghan06, 2023-06-08:
@zxystd Can confirm the culprit is BlueToolFixup. Bluetooth is now working fine with Sonoma, thanks!

closed 2023-06-08 by meghan06

qasim799, 2023-07-07:
Even with this bluetoolfixup.kext and nvram patches the Bluetooth on xps 9360 intel 8260 card is not working I can't see it in usb mapping and intelbluetoothfirmware.kext doesn't load up, during boot i see some bluetoothd failure errors.

maxpicelli, 2023-07-08:
Where is Airportitlwm Beta 2 ??

balajipitchumani, 2023-08-19:
> @zxystd Can confirm the culprit is BlueToolFixup. Bluetooth is now working fine with Sonoma, thanks!

Can please confirm which BT next works for sonoma beta 5? WiFi works for me but not BT.

meghan06, 2023-08-20:
@balajipitchumani the latest **commit, not release** of BlueToolFixup will do. See dortania's build page for a download

https://dortania.github.io/builds/?product=BrcmPatchRAM&viewall=true

richso, 2023-10-13:
Hi @meghan06, I am very new to Opencore, I just have installed macOS Sonoma to my **_Mac_** but with the Bluetooth failed to work, to fix that what file I need to download and **_how to install_**? Note: I already have the Opencore EFI and Sonoma installed and can boot into the Sonoma.

MalikBehloul, 2024-02-04:
> Les pilotes Bluetooth Intel peuvent fonctionner assez bien sur **n'importe quelle** version, le correctif nécessaire est à `bluetoolfixup.kext`côté.
> 
> BTW, voici ma version modifiée. [BlueToolFixup.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11672823/BlueToolFixup.kext.zip)
thank you very much before I "had no stability with the kext especially in return from standby with the modification it works a wonder thanks again