OpenIntelWireless/IntelBluetoothFirmware #438: Bluetooth Doesn't work at all on macOS Ventura

TQMatvey, 2023-06-08:
### Have you read the docs?

Yes

### macOS Version

macOS 13.5

### Kext Version

Latest from CI 592507a

### Wireless Adapter Model and USB Product ID

Intel AX 210 (Unable to obtain even from IORegistry, not enough knowledge)

### Description

After Updating from Mac OS Monterey to Mac OS Ventura, I am unable to turn on bluetooth

i have tried different versions, combinations, etc.

i will also
[config.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11697143/config.txt)
 attach my config.plist and `sudo dmesg | grep -i bluetooth`
[dmesg.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11697116/dmesg.txt)

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
[  263.094932]: Sandbox apply: com.apple.CloudDocs.MobileDocume[985] <bytes>Sandbox: bird(456) deny(1) file-read-data /Users/matvey/Library1 duplicate report for Sandbox: bird(456) deny(1) file-read-data /Users/matvey/LibrarySandbox: WindowManager(432) deny(1) file-read-data /private/var/folders/r1/zcrwh9hs1_g2g22pm_hb398m0000gn/C/com.apple.iconservices/store.index3 duplicate reports for Sandbox: WindowManager(432) deny(1) file-read-data /private/var/folders/r1/zcrwh9hs1_g2g22pm_hb398m0000gn/C/com.apple.iconservices/store.indexSandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-xattr /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBluetoothFirmware.kextSandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-xattr /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBluetoothInjector.kextSandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-xattr /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBTPatcher.kextmemorystatus_update(enter): pid 986, priority 210, dirty=0x0, Active(-1MB NF), Inactive(-1MB, NF)
[  263.410695]: Sandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-data /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBluetoothFirmware.kextSandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-data /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBluetoothInjector.kextSandbox: com.apple.quicklook.ThumbnailsA(520) deny(1) file-read-data /Users/matvey/Downloads/IntelBluetoothFirmware/Release/IntelBluetoothFirmware-2.3.0-RELEASE/IntelBTPatcher.kextCouldn't build index for com.apple.quicklook.satellite: Not eligible for acceleration
```

TQMatvey, 2023-06-09:
basically i just updated from Mac OS Monterey to Mac OS ventura, i dont remember to which exact version, either 13.3 or 13.4.

When i just got my laptop, it had intel model 8265NGW, both bluetooth and wifi worked fine on monterey.
Then i replaced it with a better AX210 that i had laying around, everything still was working
Then i updated to Ventura, and both wifi and bluetooth died
Wifi was fixed by using itlwm + heliport, a few days or weeks ago, you released an update for airportitlwm, and wifi started working well without itlwm and heliport

but i still can't fix bluetooth, the only broken thing in my laptop as of now...
i tried releases, and CI-CD builds, i even used your fork of BrcmPatchRAM, still no luck

Can it be USB Mapping issue? i havent done that since all my ports work fine as i am using 
[this](https://github.com/tylernguyen/x1c6-hackintosh)
I tried doing USB mapping, but no luck, i lack experience for that :<

TQMatvey, 2023-06-09:
oh and i did try switching AX210 back to 8265NGW, didnt fix anything

zxystd, 2023-06-09:
use this command to see if intel bluetooth firmware is working
```
ioreg -f -i -l | grep IntelBluetoothFirmware
```
if nothing shows, you need to map USB.

TQMatvey, 2023-06-09:
yup nothing shows up... s#it

zxystd, 2023-06-09:
[image]
If you can see Bluetooth HCI Controller, then the usb mapping is succeed.

TQMatvey, 2023-06-09:
i dont.... i just tried to map, but no luck, it seems that the repo i used for EFI already has a lot of ACPIs and other stuff, i cant usbmap...

misaka-20002, 2023-06-09:
> [image]
> If you can see Bluetooth HCI Controller, then the usb mapping is succeed.
[image]

I do mapped successfully and IOReg shows a lot of stuff
but in system info still

all kexts are loaded(BlueToolFixup,firmware and BTPatcher)

misaka-20002, 2023-06-09:
btw I'm on 13.4

misaka-20002, 2023-06-09:
Just do the following:
1. Do proper usb mapping
2. load the kexts in correct order(Firmware first, then BlueToolFixup, last IntelBTPatcher)
3. open the config.plist for OpenCore and turn to NVRAM->7C436110-AB2A-4BBB-A880-FE41995C9F82 and add two nvram entry: {bluetoothInternalControllerInfo:00000000 00000000 00000000 0000} {bluetoothExternalDongleFailed:00} (Type is DATA)
4. reset your nvram
5. Bluetooth works well
@zxystd may you update the install instructions on GitHub pages? I think it would help a lot

TQMatvey, 2023-06-09:
[image]

is this correct? @misaka-20002 

and tbh i can't do usb mapping... it just doesn't work for me

TQMatvey, 2023-06-09:
sadly after following all steps, without the usb mapping part, bluetooth still doesnt turn on :(

TQMatvey, 2023-06-09:
the weird thing is everything except bluetooth works perfectly, wifi, usb ports, even type-c ports, all work perfectly, but not bluetooth...

misaka-20002, 2023-06-09:
> sadly
[BrcmPatchRAM-2.6.7-RELEASE.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11705032/BrcmPatchRAM-2.6.7-RELEASE.zip)
 after following all steps, without the usb mapping part, bluetooth still doesnt turn on :(
use this BlueToolFixup and try again?

Later check SystemInfo-USB and see if there's anything called like Bluetooth Host Controller or Bluetooth HCI
if none, you need to redo usb mapping

misaka-20002, 2023-06-09:
and if the bluetooth works well previously and mapping is def correct

closed 2023-06-09 by zxystd

TQMatvey, 2023-06-12:
idk :| i just can't map usb on my laptop, will have to downgrade to monterey i guess :/

TQMatvey, 2023-06-19:
I fixed it... the issue wasn't in usb mapping... it was in the goddamn bios.. after a bios reset, bluetooth works!