OpenIntelWireless/IntelBluetoothFirmware #439: Bluetooth on Intel Centrino Wireless-N 135 on BigSur

Karapuzz, 2023-06-10:
### Have you read the docs?

Yes

### macOS Version

BigSur 11.7.7 (20G1345)

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel Centrino Wireless-N 135; IDVendor 0x8087; Device ID 0x07DA

### Description

Is it possible to get Bluetooth on Intel Centrino Wireless-N 135 on BigSur?
IDVendor 0x8087 Device ID 0x07DA. When added to OpenCore IntelBluetoothInjector.kext, the download slows down on IOUSBHostSevice and the bluetooth "falls off".
Without this kext bluetooth is visible in the system, but the shutdown button is not active.
On MacOS Monterey, this card is fully loaded (Wi-Fi and Bluetooth).
[image]
[image]
[image]
[image]
[image]
[image]

### Info in System Report - Bluetooth

```shell
Версия ПО Apple Bluetooth:	8.0.5d7
  Аппаратура, функции и настройки:
  Имя:	Mac Pro — Home
  Адрес:	0C-D2-92-4C-22-68
  Поддержка Bluetooth Low Energy:	Да
  Поддержка Handoff:	Да
  Поддержка Instant Hotspot:	Да
  Производитель:	Intel
  Передача данных:	USB
  Версия прошивки:	6.7869 (6.64512)
  Сигнал Bluetooth:	Вкл.
  Видимость для других устройств:	Выкл.
  Доступно для подключения:	Нет
  Указание автопоиска:	Вкл.
  Удаленное пробуждение:	Вкл.
  ID производителя:	0x8087
  ID продукта:	0x07DA
  Базовая спецификация Bluetooth:	4.0 (0x6)
  Редакция HCI:	0x1EBD
  Версия LMP:	4.0 (0x6)
  Подверсия LMP:	0xFC00
  Тип устройства (основной):	Computer
  Тип устройства (полный):	Mac Desktop
  Композитный класс устройства:	0x380104
  Класс устройства (основной):	0x01
  Класс устройства (малый):	0x01
  Класс службы:	0x1C0
  Автопоиск клавиатуры:	Вкл.
  Службы:
  Передача файлов по Bluetooth:
  Общая папка для устройств:	~/Public
  Во время принятия объектов:	Принять все без предупреждения
  Состояние:	Отключено
  Обмен файлами по Bluetooth:
  Папка для принятых объектов:	~/Downloads
  Когда приняты другие объекты:	Сохранить в папку
  Во время принятия объектов:	Принять все без предупреждения
  Состояние:	Отключено
  Общий доступ к интернету через Bluetooth:
  Состояние:	Отключено
  Входящие последовательные порты:
  Bluetooth-Incoming-Port:
  Канал RFCOMM:	3
  Требует аутентификации:	Нет
```

### Relevant log output

```shell
Last login: Sat Jun 10 12:29:09 on ttys000
home@Mac-Pro-Home ~ % sudo dmesg | grep -i IntelBluetooth
Password:
home@Mac-Pro-Home ~ %
```

Karapuzz, 2023-06-10:
I understand that Device ID 0x07DA is not in the list of supported devices. But in MacOS Monterey this card works with v2.2.0

zxystd, 2023-06-10:
@Karapuzz `0x07DA ` is a special CSR chip, see https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btusb.c#L479. It is added support after aaf42472824865f553eeb7e17c7fa1c024da1305. I have `Centrino Wireless-N 6235` which have the same bluetooth chip with yours, and it is working normal depending on `com.apple.iokit.CSRBluetoothHostControllerUSBTransport`, maybe some chips should be work on `com.apple.iokit
.BroadcomBluetoothHostControllerUSBTransport`, please try this one.

[IntelBluetoothInjector.kext.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11712839/IntelBluetoothInjector.kext.zip)

Karapuzz, 2023-06-10:
Thanks, shutdown works! IntelBluetoothFirmware.kext+IntelBTPatcher.kext+Your kext.
There are a couple more questions:
1. Why doesn't the injector from v2.2.0 work (https://github.com/OpenIntelWireless/IntelBluetoothFirmware/releases/tag/v2.2.0)?
It turns out that new versions are not suitable for my device (if BigSyur is used)?
2. The settings interface slightly "freezes" when turning on / off and connecting devices. The connection is disconnected spontaneously.
Bluetooth file sharing is not working.
I'm not sure if this is a complete plant. Or is it the maximum?
[image]

111jnicn, 2024-04-06:
Can you create a version that supports Sonoma? My network card is the same as his