OpenIntelWireless/IntelBluetoothFirmware #441: A working AX201 on latest Monterey does not work on latest Ventura (reports BCM_4350C2)

mackonsti, 2023-06-12:
### Have you read the docs?

No

### macOS Version

Ventura 13.4 (22F66)

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AX201 [8087:0026]

### Description

I recently made the jump from Monterey to Ventura, thinking the OS is mature... but I am facing the same early problems reported in #369 but with latest Ventura 13.4 (22F66) where my perfectly working **AX201** [8087:0026] reports as being "BCM_4350C2". Originally I thought _Hackintool_ was wrong!

In **IORegistry** there doesn't seem there's a firmware loaded, string **fw_name** is empty. **USBPorts.kext** is long-time created and validated with each new macOS release, the **HS10** port is the one where BTLE is usually attached (no change):
[image]
All needed 2+1 kexts appear loaded and running (via **kextstat**):

```
as.acidanthera.BlueToolFixup (2.6.6)
com.zxystd.IntelBluetoothFirmware (2.2.0)
com.zxystd.IntelBTPatcher (2.2.0)
```

Finally, this is the order in my OpenCore config:

First in OC config: **BlueToolFixup.kext**
Next in OC config: **IntelBluetoothFirmware.kext**
Final in OC config: **IntelBTPatcher.kext**

Please note: on Monterey it was working very well.

Thank you.

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
(TBC as I have not done it before, will update ticket with the needed info)
```

### Kernel Panic Logs

```shell
No kernel panic, thankfully.
```

HieuNVDSgit, 2023-06-14:
I remap usb port and bluetooth working on Ventura but after that i faced new problem:
- I can't connect to my iPhone via Bluetooth (only success connect with my Android TV, Pixel3XL Phone and Bluetooth BLE Mouse)
- Bluetooth is ON and connected but Discoverable still in OFF state (System Report)
[image]

mackonsti, 2023-06-14:
Hi @HieuNVDSgit how do you "remap" the USB?

Because my hardware on Monterey in my case used USB10 port, on Ventura I do not know why it should or would change... There is no option in the Intel Visual BIOS of my NUC to do such mapping or attempts.

Are you using OpenCore? What is the sequence/order of loading the kexts, in your config, please?

But I see that you, too, get the wrong IDs being reported, just like me. It says Apple as vendor, not sure this is expected @zxystd and @williambj1 ? Even _Hackintool_ reports "Apple" as vendor on Ventura, not the case on Monterey... Thanks in advance for your input.

zxystd, 2023-06-14:
@mackonsti Fimware driver is working well, from the ioreg I don't see the bluetoothd process was attached, so it is probably a bluetoothd related problem. You can try my fork .
https://github.com/zxystd/BrcmPatchRAM

zxystd, 2023-06-14:
@HieuNVDSgit 
> * I can't connect to my iPhone via Bluetooth (only success connect with my Android TV, Pixel3XL Phone and Bluetooth BLE Mouse)
You can use Google or Apple forums to get the answer.

> * Bluetooth is ON and connected but Discoverable still in OFF state (System Report)

Keep open the bluetooth search panel, it will display as ON.

I think you need to Google or experience a real mac first before asking questions.

closed 2023-06-14 by zxystd

mackonsti, 2023-06-14:
> @mackonsti Fimware driver is working well, from the ioreg I don't see the bluetoothd process was attached, so it is probably a bluetoothd related problem. You can try my fork .
> https://github.com/zxystd/BrcmPatchRAM

Thank you @zxystd for your kind assistance.
Your fork points to BrcmPatchRAM so which component do you advise me to use?
Isn't this for Broadcom? 😬 

Do you require further logs from my system? You say that **bluetoothd** process was not attached, perhaps some further logs can provide a reason? What kext is responsible for this process to be attached, please? (so that I can check)

非常感谢！

zxystd, 2023-06-14:
> Thank you @zxystd for your kind assistance. Your fork points to BrcmPatchRAM so which component do you advise me to use? Isn't this for Broadcom? 😬
> 
only replace the bluetoolfixup.kext

> Do you require further logs from my system? You say that **bluetoothd** process was not attached, perhaps some further logs can provide a reason? What kext is responsible for this process to be attached, please? (so that I can check)
> 

log show --last boot | grep bluetooth>log.log

mackonsti, 2023-06-14:
Hello @zxystd for the sake of your information and in case someone else reads here, here is what I did and achieved after your suggestion.

1. In my OpenCore configuration, per ticket #435 I added the 2 new NVRAM keys:
```
	<key>bluetoothExternalDongleFailed</key>
	<data>AA==</data>
	<key>bluetoothInternalControllerInfo</key>
	<data>AAAAAAAAAAAAAAAAAAA=</data>
```
...in config section `NVRAM` → `Add` → `7C436110-AB2A-4BBB-A880-FE41995C9F82`

I rebooted whilst having installed the latest official stable **BlueToolFixup.kext** 2.6.7 and I realized the BTLE of AX201 was now detected as THIRD_PARTY_DONGLE. It could detect and connect to my Original (v1) AirPods, but I had 2 issues:

a) could **not** turn ON the Bluetooth after I turned it OFF.
b) somehow, some frequent refresh of the BTLE module made the AirPods disconnect, like some kind of connection reset. It was obvious during music playback (e.g. MP3).

&nbsp;
2. As I don't have Xcode installed (was a clean installation) thankfully I used the [Actions](https://github.com/zxystd/BrcmPatchRAM/actions/) found on your repository and was able to get the "Artifacts" zip file of your latest commit [Remove unused patch, it is replaced by searchAndPatchWithMask code](https://github.com/zxystd/BrcmPatchRAM/actions/runs/5231249629) from which I extracted **BlueToolFixup.kext** and replaced the recent released one (i.e. 2.6.7 from [acidanthera/BrcmPatchRAM ](https://github.com/acidanthera/BrcmPatchRAM/releases))

Rebooted. This time:

a) I coud well turn OFF and then back ON the Bluetooth, the Magic Mouse could reconnect automatically soon after ✅ 
b) For using audio playback, it seems that a song played well up to 1 minute, then AirPods **disconnected** again... 🛑 

P.S. I did not change my OpenCore loading order of the kexts, it's still in this order:

1. BlueToolFixup.kext
2. IntelBluetoothFirmware.kext
3. IntelBTPatcher.kext

System Information > Hardware > Bluetooth
```
  Bluetooth Controller:
	Address:		<removed>
	State:			On
	Chipset:		THIRD_PARTY_DONGLE
	Discoverable:		Off
	Firmware Version:	v256 c256
	Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
	Transport:		USB
	Vendor ID:		0x004C (Apple)
  Connected:
  	Original AirPods:
		Address:		<removed>
		Vendor ID:		0x004C
		Product ID:		0x2002
		Left Battery Level:	100%
		Right Battery Level:	87%
		Firmware Version:	6.8.8
		Minor Type:		Headphones
		RSSI:			-66
		Serial Number:		<removed>
		Services:		0x880019 < HFP AVRCP A2DP AACP ACL >
	  White Magic Mouse:
		Address:		<removed>
		Vendor ID:		0x004C
		Product ID:		0x0269
		Firmware Version:	1.0.7
		Minor Type:		Mouse
		RSSI:			-41
		Services:		0x800020 < HID ACL >
  Not Connected:
	My iPhone:
		Address:		<removed>
		RSSI:			-57
```

**A few questions with yes/no answers, to help you quickly reply (when you have the time) so we can learn from you, please?** 🙏 

a) Is the above kext order OK for OC config? Do you have a better/correct order to suggest?

b) Do we need to add `bluetoothExternalDongleFailed` and `bluetoothInternalControllerInfo` in the NVRAM config, if we use _your_ patched **BlueToolFixup.kext** ?

c) Is this working fix a _workaround_ for Ventura or a _permanent_ fix?
I mean, are we now always getting THIRD_PARTY_DONGLE solution shown?  (we need to see Ventura 13.5....)

d) Is the solution to try to "fool" i.e. fake macOS Ventura thinking that we're using an external USB BTLE stick instead or treating is as an internal device now? (→ meaning good USB mapping!)

e) I saw that you suggested your fixes in the main repository of [BrcmPatchRAM](https://github.com/acidanthera/BrcmPatchRAM). Any idea of they are going to be merged and officially used, for a next **acidanthera** version? Or we should keep using your **BlueToolFixup.kext** ?

**Thank you** very much for spending time on this.

[ax201_konsti.log.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11750996/ax201_konsti.log.zip)

**P.S.** After searching more, I realized there are a lot of people here (on your repo) reporting this same issue on Ventura; perhaps it makes sense to have a mention/reference on your main README.md for this problem, and inform people of the steps; let me know if I can help write it with you, I'd be happy to do this.

Xie Xie

HieuNVDSgit, 2023-06-16:
> Hi @HieuNVDSgit how do you "remap" the USB?
>
> Because my hardware on Monterey in my case used USB10 port, on Ventura I do not know why it should or would change... There is no option in the Intel [...]

I was remap usb with usbtoolbox in Windows:
- Delete detected data;
- Change Setting to Use native classes and disable Blind companions
- Map all USB2 port to 0 and all companions of it (USB3) to 3
- Map Bluetooth and Webcam port to internal (255)
- Build USBMap.kext and update to OC Kernel (remove USBToolbox.kext if exist, no need to use in case Native classes enabled)

I'm using OC 0.9.3 and my order of kext loading is:
- Bluetoolfixup.kext
- IntelBluetoothFirmware.kext
- IntelBTpatcher.kext
but I think it's not important, I tried to change this order many time but no affect (with my Intel card)

mackonsti, 2023-06-16:
Again, thank you @zxystd for your time and comments. All was very clear.

**A final question:** The audio playback that gets disconnected after approx. 1 minute, you think that it has chances to be resolved? I can try tomorrow a BT speaker, but old 1st generation AirPods seem to have this problem.

As you say, this repo deals with loading Firmware to BTLE for Intel, so perhaps indeed for this audio disconnection issue, the new `BlueToolFixup.kext` can fix things 🙏

maddog860, 2026-09-10:
If anyone has a an AX201 and is having problems with Airplay on Ventura, feel free to try my fix: https://github.com/maddog860/itlwm-Working-Ventura-Airplay/releases/tag/Ventura-Airplay-Fix

lemonz666, 2026-09-13:
Confirmed a very similar issue on macOS Tahoe with Intel AX201.

Environment:

macOS: Tahoe
Darwin: 25.x
Bluetooth device: Intel AX201
USB VID:PID: 8087:0026

BlueToolFixup: 2.7.2
IntelBluetoothFirmware: 2.5.0, built from latest master
IntelBTPatcher: enabled

The required NVRAM variables were already configured:

bluetoothExternalDongleFailed = 00

bluetoothInternalControllerInfo =
0000000000000000000000000000

USB mapping was also correct. The Bluetooth USB device was enumerated normally and IORegistry showed:

idVendor = 0x8087
idProduct = 0x0026
FirmwareLoaded = Yes
UsbExclusiveOwner = bluetoothd

The related components were active:

IntelBluetoothFirmware = 1
IntelBTPatcher = 1
BlueToolFixup = 1

However, Bluetooth initialization still failed.

Before applying the workaround, `system_profiler SPBluetoothDataType` showed:

Bluetooth Controller:

Address: NULL
State: Off
Chipset: BCM_4350C2
Firmware Version: v0 c256
Transport: USB
Vendor ID: 0x004C

Nearby Bluetooth devices could sometimes be discovered, but pairing/connection did not work correctly.

The USB layer and IntelBluetoothFirmware itself appeared to be working. The boot log showed:

IntelBluetoothFirmware selected configuration 1

BlueTool also detected:

USB Bluetooth Host Controller 0x0026

After adding the following boot argument:

-btlfxallowanyaddr

the controller initialized correctly.

After the change:

Bluetooth Controller:

Address: <valid Bluetooth address>
State: On
Chipset: THIRD_PARTY_DONGLE
Firmware Version: v256 c256
Transport: USB

Pairing and connection now work correctly, including normal Bluetooth audio services.

So in this AX201 8087:0026 + macOS Tahoe case, the issue did not appear to be USB mapping or firmware loading. It looks like the controller was blocked later during bluetoothd initialization/address validation.

Adding:

-btlfxallowanyaddr

resolved the issue completely.

This may be useful for other AX201 users seeing:

Address: NULL
State: Off
Chipset: BCM_4350C2

on Tahoe.

Related newer Tahoe Bluetooth issue: #513

maddog860, 2026-09-16:
> Confirmed a very similar issue on macOS Tahoe with Intel AX201.
> 
> So in this AX201 8087:0026 + macOS Tahoe case, the issue did not appear to be USB mapping or firmware loading. It looks like the controller was blocked later during bluetoothd initialization/address validation.
> 
> Adding:
> 
> -btlfxallowanyaddr
> 
> resolved the issue completely.
> 
> This may be useful for other AX201 users seeing:
> 
> Address: NULL State: Off Chipset: BCM_4350C2
> 
> on Tahoe.
> 
> Related newer Tahoe Bluetooth issue: [#513](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/513)

This driver should be tried if having problems in Mac OS Tahoe (my kext is still ideal for Mac OS Ventura)
https://github.com/kgp-macPro/AirportItlwm-Tahoe

Thank you for posting your feedback!

avinashchevala, 2026-11-05:
@lemonz666 , can you share me the link to the  file 

IntelBluetoothFirmware: 2.5.0, built from latest master you used,

 also can you confirm whether these values you used are correct
 bluetoothExternalDongleFailed = 00

bluetoothInternalControllerInfo =
0000000000000000000000000000