OpenIntelWireless/IntelBluetoothFirmware #442: bluetooth for ax203 not work but wifi is ok

bestlufeng, 2023-06-15:
### Have you read the docs?

No

### macOS Version

13.3 and 13.4

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

intel ax203

### Description

my wifi is work 
[image]
but can not see buletooth device info 
[image]
and blue switch is not work
[image]

my kext is ok and version is latest 2.2.0
[image]

sure it soupport  ax203 ？

### Info in System Report - Bluetooth

```shell
-
```

### Relevant log output

```shell
no info about it
```

melxman, 2023-06-15:
你试试：
在nvram的7C436110-AB2A-4BBB-A880-FE41995C9F82下添加两个key
bluetoothInternalControllerInfo     data    0000000000000000000000000000
bluetoothExternalDongleFailed     data    00

bestlufeng, 2023-06-15:
> 你试试： 在nvram的7C436110-AB2A-4BBB-A880-FE41995C9F82下添加两个key bluetoothInternalControllerInfo data 0000000000000000000000000000 bluetoothExternalDongleFailed data 00

这个我加过了，但是USB里面还是看不到对应的设备

misaka-20002, 2023-06-15:
remap your usb

closed 2023-06-15 by zxystd