OpenIntelWireless/IntelBluetoothFirmware #445: AC 3165 - Problem

dicheto, 2023-06-17:
### Have you read the docs?

Yes

### macOS Version

Ventura 13.4

### Kext Version

latest

### Wireless Adapter Model and USB Product ID

Intel AC 3165

### Description

Recently, the bluetooth stopped working. I tried restarting, but no success. I updated the kexts, but again - No success. I am sending the log and my OC folder. Thank you in advance :)
[ACPI.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779754/ACPI.zip)
[config.plist.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779755/config.plist.zip)
[Drivers.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779757/Drivers.zip)
[Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779759/Kexts.zip)
[OpenCore.efi.zip](https://github.com/OpenIntelWireless/Intel
[Tools.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779763/Tools.zip)
BluetoothFirmware/files/11779761/OpenCore.efi.zip)
[Resources.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11779762/Resources.zip)

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
[    1.165967]: IOUSBHostDevice@14600000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

dicheto, 2023-06-17:
I want to Add that I am using airportitlwm with the same card, but it is working properly.

closed 2023-06-18 by zxystd

Dicheto2302, 2023-06-18:
Why you are closing my request?

zxystd, 2023-06-18:
> Recently, the bluetooth stopped working

It means it's your PC's problem, not this repo, and we will not provide any support on debugging Hackintosh problems.

dicheto, 2023-06-18:
It's not my PC problem because I have dual boot and everything is working in windows 10.