OpenIntelWireless/IntelBluetoothFirmware #448: Unable to connect to any Bluetooth devices

TQMatvey, 2023-06-20:
### Have you read the docs?

Yes

### macOS Version

macOS 13.5

### Kext Version

592507a

### Wireless Adapter Model and USB Product ID

Intel AX210

### Description

i am unable to connect to any bluetooth device, lets take my xiaomi headphones as an example, they get discovered, trying to connect to them, it just shows a loading animation and then just asks shows a button to connect again

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	84:5C:F3:redacted
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v13446 c13446
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
Not Connected:
  Matvey’s Ipad Air 3:
  Address:	4C:6B:E8:redacted
```

### Relevant log output

```shell
[    1.140802]: IOUSBHostDevice@14500000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

TQMatvey, 2023-06-20:
also i am unable to connect to or from my phone (Samsung)

TQMatvey, 2023-06-20:
# sudo dmesg | grep Bluetooth

```
sudo dmesg | grep -i Bluetooth
[    0.831891]: **** [IOBluetoothFamily][start] -- completed successfully for HIDShim
[    1.171838]: IOUSBHostDevice@14500000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
**** [BroadcomBluetooth20703USBTransport][init] -- entering ****
[    9.940774]: **** [BroadcomBluetooth20703USBTransport][init] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.940785]: **** [BroadcomBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.940795]: **** [CSRBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][init] -- entering ****
[    9.940813]: **** [IOBluetoothHostControllerPCIeTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][init] -- entering ****
[    9.940820]: **** [IOBluetoothHostControllerTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][init] -- entering ****
[    9.940828]: **** [IOBluetoothHostControllerUARTTransport][init] -- exiting -- success ****
**** [IOBluetoothHostControllerUSBTransport][init] -- entering ****
[    9.940836]: **** [IOBluetoothHostControllerUSBTransport][init] -- exiting -- success ****
**** [IOBluetoothPacketLogger][init] -- entering ****
[    9.940843]: **** [IOBluetoothPacketLogger][init] -- exiting -- success ****
**** [IOBluetoothSerialManager][init] -- entering ****
[    9.940851]: **** [IOBluetoothSerialManager][init] -- exiting -- success ****
**** [BroadcomBluetooth20703USBTransport][start] -- entering ****
[    9.943216]: **** [BroadcomBluetooth20703USBTransport][start] -- exiting -- success ****
**** [BroadcomBluetoothHostControllerUSBTransport][start] -- entering ****
[    9.943226]: **** [BroadcomBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [CSRBluetoothHostControllerUSBTransport][start] -- entering ****
[    9.943235]: **** [CSRBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerPCIeTransport][start] -- entering ****
[    9.943290]: **** [IOBluetoothHostControllerPCIeTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerTransport][start] -- entering ****
[    9.943296]: **** [IOBluetoothHostControllerTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerUARTTransport][start] -- entering ****
[    9.943302]: **** [IOBluetoothHostControllerUARTTransport][start] -- exiting -- success ****
**** [IOBluetoothHostControllerUSBTransport][start] -- entering ****
[    9.943308]: **** [IOBluetoothHostControllerUSBTransport][start] -- exiting -- success ****
**** [IOBluetoothPacketLogger][start] -- entering ****
[    9.943313]: **** [IOBluetoothPacketLogger][start] -- exiting -- success ****
**** [IOBluetoothSerialManager][start] -- entering ****
[    9.943319]: **** [IOBluetoothSerialManager][start] -- exiting -- success ****
[   10.586926]: Sandbox: systemstatusd(307) deny(1) user-preference-read kCFPreferencesAnyApplicationAppleKeyStore:8084:162: unexpected session: 100000 uid: -1 requested by: bluetoothd (162)
[   10.594663]: newUserClient: Invalid userClient Type request: 0IO80211ScanManager::searchForPendingAndClearPid: pid 240 had pending scan request issued 0s ago. Deleting client.IO80211ScanManager::searchForPendingAndClearPid: pid 240 had pending scan request issued 0s ago. Deleting request.**** [IOBluetoothFamily][newUserClient] -- created userclient for HIDShim
[   10.625511]: **** [IOBluetoothFamily][AddHCIEventNotification] -- Added notification port for HIDShim
[   10.625576]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created input shared memory for HIDShim
[   10.625615]: **** [IOBluetoothFamily][CreateHIDShimSharedMemoryRingBuffer] -- Created output shared memory for HIDShim
[   10.771900]: Sandbox apply: bluetoothd[162] <bytes>AppleKeyStore:8084:162: unexpected session: 100000 uid: -1 requested by: bluetoothd (162)
[   10.783061]: 1 duplicate report for Sandbox: systemstatusd(307) deny(1) user-preference-read kCFPreferencesAnyApplicationSandbox: bluetoothd(162) deny(1) mach-lookup com.apple.private.corewifi-xpcnewUserClient: Invalid userClient Type request: 0Sandbox apply: secinitd[333] <bytes>newUserClient: Invalid userClient Type request: 0Sandbox: bluetoothd(162) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plist1 duplicate report for Sandbox: bluetoothd(162) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistSandbox: locationd(147) deny(1) mach-lookup com.apple.PowerManagement.controlSandbox: locationd(147) deny(1) file-issue-extension target:/private/var/db/locationd/Library/Preferences/com.apple.CFNetwork.plist extension-class:com.apple.cfprefsd.readPMRD: setAggressiveness(0) kPMMinutesToSleep = 0
[   10.831460]: Sandbox apply: com.apple.AmbientDisplayAgent[331] <bytes>Sandbox: locationd(147) deny(1) file-issue-extension target:/private/var/db/locationd/Library/Preferences/ByHost/com.apple.CFNetwork.7B5E22F0-A5DB-5E53-93AC-6995CDE05358.plist extension-class:com.apple.cfprefsd.readSandbox: searchpartyd(310) deny(1) mach-lookup com.apple.timed.xpcSandbox: locationd(147) deny(1) file-read-data /private/var/db/nsurlstoraged/dafsaData.binSandbox apply: distnoted[332] <bytes>Sandbox: lo [... cut ...] : passd(546) deny(1) file-read-data /Library/Preferences/com.apple.networkd.plistAppleSSE::hasBridgeSSE - SSE not present (neither KR nor SEPMgr)
[   69.456187]: connectx() - failed necp_set_socket_domain_attributesSandbox: bluetoothd(162) deny(1) file-read-metadata /Library/Application SupportSandbox: bluetoothd(162) deny(1) file-read-metadata /Library/Application Support/BTServerconnectx() - failed necp_set_socket_domain_attributesALUC:clientDied(PID: 878), lifsd_entitled 1 koio_entitled 0
```

qhuyduong, 2023-06-20:
@TQMatvey can you try this patch?
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/446

TQMatvey, 2023-06-20:
Already did, doesn't help, and none of my devices are BLE

TQMatvey, 2023-06-20:
reverting to v2.2.0 fixed the issue.... but why????

TQMatvey, 2023-06-20:
sadly "Mirror or extend display" still doesn't work for me :( i wanted universal control with my iPad

zxystd, 2023-06-23:
Also can not do anything on this thing, the bluetooth driver is written by Apple.

closed 2023-06-23 by zxystd