OpenIntelWireless/IntelBluetoothFirmware #449: How to detect used firnware? (Not a Bug)

5T33Z0, 2023-06-23:
### Have you read the docs?

Yes

### macOS Version

macOS 14

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel AC-9560 (0xAAA)

### Description

Sorry to bother you. I want to compile a slimmed version of the IntelBluetoothFirmware.kext that only contains the firmware for my card. I followed this guide: https://github.com/dreamwhite/ChonkyIntelBluetoothFirmware-Build 

But in my case, the used firmware is not revealed by IORegistryExplorer:

[image]

Can you tell me which firmware it uses or is there a list of which firmware is used for which model?

Thank you

### Info in System Report - Bluetooth

```shell
No Sys Report.
```

### Relevant log output

```shell
No log.
```

### Kernel Panic Logs

```shell
No Panic
```

williambj1, 2023-06-23:
There is a known bug where the fw name disappears after waking up from sleep, try booting directly to the system and see if it is there.

5T33Z0, 2023-06-23:
Unfortunately, this also happens when booting normally.

Is there another way to figure out the firmware? A list or a terminal command or something in WIndows that I can llok for. I also checked in Device Manager already. I couldn'tf find anything that points to the firmware used by this card.

zxystd, 2023-06-23:
You can also use ubuntu live to see the correct firmware name (using dmesg)

closed 2023-06-23 by zxystd

5T33Z0, 2023-06-23:
Okay, I can try that. What's the command I need to enter? Sorry, I am not a programmer.

zxystd, 2023-06-23:
@5T33Z0 `dmesg` or `dmesg | grep ibt` is enough.