OpenIntelWireless/IntelBluetoothFirmware #452: intel ax211 broken in  Sonoma betas

Jeditekk, 2023-07-07:
### Have you read the docs?

Yes

### macOS Version

macOS14 betas 1-3

### Kext Version

v2.2.0 bu have loadedt the one you posted to fix 13.5 beta

### Wireless Adapter Model and USB Product ID

intel ax211

### Description

works in Ventura but not Sonoma betas.understand a issue with bluetool..

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
which is not correct!!!!
```

### Relevant log output

```shell
[    1.352731]: IOUSBHostDevice@14c00000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1

I do have a shit.log ready :)
```

### Kernel Panic Logs

```shell
NO KP's
```

Jeditekk, 2023-07-07:
[shit.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11974824/shit.log)

zxystd, 2023-07-07:
I don't see your bluetooth from the log, have you mapped the USB? can you see it in the System Info - USB?

Jeditekk, 2023-07-07:
[image]
[image]

Jeditekk, 2023-07-07:
this config works in monty, and Ventura ..

Jeditekk, 2023-07-07:
[image]
[shit.log](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11975182/shit.log)
 using the latest bluetool kext

Jeditekk, 2023-07-07:
bluetooth is constantly cycling off/on with this config

zxystd, 2023-07-07:
[image]
You should use the latest IntelBTPatcher.kext to fix the bluetoothd crash issue.

Jeditekk, 2023-07-07:
ive tried building 2.3.0 but getting build error on Xcode beta...its only building the injector kext .can u post a compiled 2.3.0?

Jeditekk, 2023-07-07:
boot arg -lilubetaall is a instant KP

zxystd, 2023-07-07:
you can get it from github CI build, don't need to build by yourself.

zxystd, 2023-07-07:
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/suites/13462196536/artifacts/738622442

Jeditekk, 2023-07-07:
That's the fix!!! 2.3.0 IntelBluetoothFirmware and latest bluetool !!
![Screenshot 2023-07-07 at 9 27 57 AM](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/assets/138839
[image]
226/9f342256-a96c-4bfd-8ad3-9f4f0a8f90e5)

Jeditekk, 2023-07-07:
everything working but iPad sidecar with never has!!! too good togo here 2.3.0

closed 2023-07-08 by williambj1