OpenIntelWireless/IntelBluetoothFirmware #453: Apple Keyboard doesn't connect after reboot

damiandudycz, 2023-07-15:
### Have you read the docs?

Yes

### macOS Version

macOS 13.4.1

### Kext Version

v2.2.0

### Wireless Adapter Model and USB Product ID

Intel 8265

### Description

My bluetooth and WiFi Seems to work, but I have one issue with bluetooth. For some reason, Apple keyboard doesn't connect back, after system boot. For start it shows as connected for about 30s, but doesn't react to keystrokes. And after that it just shows as disconnected, and I need to pair it again to work.

### Info in System Report - Bluetooth

```shell
Kontroler Bluetooth:
  Adres:	74:70:FD:71:DE:4A
  Stan:	Wł.
  Zestaw układów:	THIRD_PARTY_DONGLE
  Wykrywalne:	Wył.
  Wersja oprogramowania sprzętowego:	v256 c256
  Obsługiwane usługi:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  ID producenta:	0x004C (Apple)
  Połączone:
  Apple Keyboard - Home Dudycz:
  Adres:	44:2A:60:E1:53:59
  ID producenta:	0x05AC
  ID produktu:	0x0255
  Wersja oprogramowania sprzętowego:	0.5.0
  Typ dodatkowy:	Keyboard
  RSSI:	-58
  Usługi:	0x800020 < HID ACL >
  Magic Mouse - Home Dudycz:
  Adres:	AC:BC:32:E3:A7:53
  ID producenta:	0x004C
  ID produktu:	0x0269
  Wersja oprogramowania sprzętowego:	1.9.2
  Typ dodatkowy:	Mouse
  Usługi:	0x800020 < HID ACL >
  Nie połączono:
  AirPods (Damian):
  Adres:	F8:4D:89:38:73:CA
  ID producenta:	0x004C
  ID produktu:	0x2013
  Wersja etui:	1.0.8
  Wersja oprogramowania sprzętowego:	5​E135
  Typ dodatkowy:	Headphones
  Numer seryjny:	MTXDXG2G21
  Numer seryjny (L):	HQDGKA5H035Y
  Numer seryjny (R):	HNRGJMLC0360
  iPhone (Damian):
  Adres:	88:A4:79:04:1C:5D
  RSSI:	-81
  MacBook Pro 2020:
  Adres:	E0:C0:9A:4F:2C:DC
  SRS-XB13:
  Adres:	78:5E:A2:3F:D8:32
  Typ dodatkowy:	Speaker
```

### Relevant log output

```shell
<no output>
```

zxystd, 2023-07-15:
Sorry it should be handled by Apple, not this firmware driver.

closed 2023-07-15 by zxystd

damiandudycz, 2023-07-15:
> Sorry it should be handled by Apple, not this firmware driver.

You sure? The same keyboard works fine on MacBook Pro 2020, on the same version of the system.