OpenIntelWireless/IntelBluetoothFirmware #454: Bluetooth is nonfunctional on macOS 14 beta 3

Odroid13, 2023-07-23:
### Have you read the docs?

No

### macOS Version

MacOS 14 Beta 3

### Kext Version

V 2.2.0

### Wireless Adapter Model and USB Product ID

BCM_4350C2

### Description

Motherboard: Asus ROG STRIX B660-I GAMING WIFI
MacOs Monterey everything works: Wifi, Ethernet, Bluetooth

With Sonoma Beta 3, I had to make changes to get the network to work.

For ethernet to work.
AppleIntelI210Ethernet.kext & SSDT-I225V.aml
Works perfectly ^^

For Wifi, I installed AirportItlwm.kext (Beta Sonoma)
Works perfectly ^^

For Bluetooth
Does not work :(

I took the kext: BlueToolFixup.kext: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/437#issuecomment-1579931908

In my EFI directory:

BlueToolFixup.kext
IntelBluetoothFirmware.kext
IntelBTPatcher.kext

When I turn on the bluetooth, it quickly turns off.
I must be doing something wrong but I don't know what to do, any help would be appreciated. THANKS

### Info in System Report - Bluetooth

```shell
Contrôleur Bluetooth :
  Adresse :	NULL
  État :	Désactivé
  Jeu de puces :	BCM_4350C2
  Détectable :	Désactivé
  Version du programme interne :	v0 c0
  Services pris en charge :	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport :	USB
  ID du fournisseur :	0x004C (Apple)
```

### Relevant log output

```shell
Last login: Sun Jul 23 19:04:53 on ttys000
pazoo@iMacProdePazoO ~ % sudo dmesg | grep -i IntelBluetooth
Password:
pazoo@iMacProdePazoO ~ %
```

zxystd, 2023-07-23:
I think you are using gen3 bluetooth like ax210/ax211? If that, you should upgrade IntelBTPatcher to the latest v2.3.0 version.

Odroid13, 2023-07-24:
> I think you are using gen3 bluetooth like ax210/ax211? If that, you should upgrade IntelBTPatcher to the latest v2.3.0 version.

Thank you for your reply and your advice.
Now the bluetooth works ^^

Does the kext: BlueToolFixup.kext of version 2.6.7 work for Sonoma or I have to use the one you modified on this post: https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/437#issuecomment-1579931908

Thanks

zxystd, 2023-07-24:
@Odroid13 v2.6.8 dev version is available in BrcmPatchRam official repo, you can download there.

Odroid13, 2023-07-24:
> @Odroid13 v2.6.8 dev version is available in BrcmPatchRam official repo, you can download there.

Thanks

closed 2023-07-24 by Odroid13