OpenIntelWireless/IntelBluetoothFirmware #458: Bluetooth cannot be enabled anymore after disabling it.

charliejimi, 2023-08-06:
### Have you read the docs?

Yes

### macOS Version

macOS 13.3

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

Intel Dual Band Wireless-AC 8260

### Description

Bluetooth can work properly initally. When disabling it, Bluetooth cannot be enabled anymore. Bluetooth can be used again when rebooting PC. 
[after_disabling_BT_it_cannot_enable_anymore.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/12268893/after_disabling_BT_it_cannot_enable_anymore.txt)
[once_PC_rebooted_BT_can_use.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/12268894/once_PC_rebooted_BT_can_use.txt)

### Info in System Report - Bluetooth

```shell
-----------------------------------------------------------------------
System Info
-----------------------------------------------------------------------
OS                             macOS Ventura Version 13.3 (Build 22E252)
Kernel                         Darwin 22.4.0 x86_64
RAM                            8.00 GB
Model Identifier               MacBookPro14,1
CPU                            Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz
Intel Generation               Kaby Lake
....(skip).....
FW Version                     512.0.0.0.0
....(skip).....
VDA Decoder                    Fully Supported
```

### Relevant log output

```shell
as shown in the attached files in the description
```

williambj1, 2023-08-07:
Please use the latest version of BluetoolFixup, https://github.com/acidanthera/BrcmPatchRAM/releases/tag/2.6.8

closed 2023-08-07 by williambj1

charliejimi, 2023-08-07:
> <OpenIntelWireless/IntelBluetoothFirmware/issue/458/issue_event/10023984105
> @github.com>

Hi,

Thanks for your prompt response. Is it possible to have the ".kext" file
for BluetoolFixup to download from the github ?

charliejimi, 2023-08-07:
Hi,

Thanks for help and i downloaded the updated ".kext" file and it can work
properly..
Appreciated it.