OpenIntelWireless/IntelBluetoothFirmware #459: AX201蓝牙不支持LEA服务

maoxikun, 2023-08-08:
### Have you read the docs?

Yes

### macOS Version

macOS 12.6.8

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

英特尔 AX201 (0x0026)

### Description

AX201蓝牙不支持LEA服务，OC及其他驱动都是目前最新release版本

### Info in System Report - Bluetooth

```shell
蓝牙控制器：
  地址：	3C:58:C2:FC:54:6C
  状态：	打开
  芯片组：	THIRD_PARTY_DONGLE
  可被发现：	关闭
  固件版本：	v256 c256
  产品ID：	0x0001
  支持的服务：	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
  传输：	USB
  供应商ID：	0x004C (Apple)
```

### Relevant log output

```shell
无
```

### Kernel Panic Logs

```shell
无
```

Tang061022Tang, 2023-08-13:
好像没有太大的关系。。。吧

maoxikun, 2023-09-27:
更新到macOS14又出现了

  蓝牙控制器：
  地址：	NULL
  状态：	关闭
  芯片组：	BCM_4350C2
  可被发现：	关闭
  固件版本：	v0 c0
  支持的服务：	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  传输：	USB
  供应商ID：	0x004C (Apple)

closed 2023-09-27 by maoxikun

Drk152, 2023-10-16:
请问怎么解决Sonoma蓝牙无法搜索设备！
蓝牙已驱动，usb已定制，mpg z490edge主板，ax201芯片，蓝牙参数加了也不行，本人现在也束手无策，前来求助！
（如图）
[image]

maoxikun, 2023-10-18:
> 请问怎么解决Sonoma蓝牙无法搜索设备！
> 蓝牙已驱动，usb已定制，mpg z490edge主板，ax201芯片，蓝牙参数加了也不行，本人现在也束手无策，前来求助！
> （如图）
> [image]

不用任何参数，驱动用最新的就行