OpenIntelWireless/IntelBluetoothFirmware #460: Broken bluetooth in macOS Sonoma

gorouflex, 2023-09-16:
### Have you read the docs?

Yes

### macOS Version

macOS 14.0 (23A339)

### Kext Version

v.2.3.0

### Wireless Adapter Model and USB Product ID

Intel AX200 , 0x004C

### Description

When i tried to connect my Airpods and some BT devices, my bluetooth will off then it on back..., also my bluetooth can be broken and never be turned on after wake and sleep

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	68:54:5A:DB:14:1E
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v256 c256
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
  Not Connected:
  iPhone của Khánh Bình:
  Address:	F4:06:16:B3:8C:5D
  RSSI:	-61
```

### Relevant log output

```shell
[    1.292418]: IOUSBHostDevice@80200000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

gorouflex, 2023-09-16:
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/assets/98001973/1c89034d-b690-452e-904c-cf4c01e4b6d3

gorouflex, 2023-10-05:
> You need to use BrcmPatchRAM3.kext (Why isn't this mentioned more often?)
> 
> Without it I also have this problem on IntelAX211

I give up on this shit , gonna buy a BCM94360NG

ShroomKing, 2023-10-05:
> > You need to use BrcmPatchRAM3.kext (Why isn't this mentioned more often?)
> > Without it I also have this problem on IntelAX211
> 
> I give up on this shit , gonna buy a BCM94360NG

After adding BrcmPatchRAM3.kext bluetooth starts working even on intel chips, so maybe give that a try first?

KaushikDuddala, 2023-10-05:
Trying that now, will send an update afterward, thanks.

KaushikDuddala, 2023-10-05:
straight off the bat, Bluetooth is way smoother. I was especially having this issue when on calls from my Mac so I'll test it with a call later.

KaushikDuddala, 2023-10-05:
Thanks! My Bluetooth seems to work flawlessly and it even sounds better and connects way smoother it practically feels perfectly like a native card. They need to include this in the OpenCore guide or somewhere in the brcmpatch repo.

ShroomKing, 2023-10-06:
> Thanks! My Bluetooth seems to work flawlessly and it even sounds better and connects way smoother it practically feels perfectly like a native card. They need to include this in the OpenCore guide or somewhere in the brcmpatch repo.

You're welcome! I really think this should be mentioned in the [IntelBluetoothFirmware README](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/blob/master/README.md)

It's not just required for Sonoma, it also fixes a bunch of issues on Ventura. Bluetooth sometimes stopped working after wake from sleep, with BrcmPatchRAM3.kext that's fixed also.

darrikonn, 2023-10-09:
Experienced the same regarding using Web Bluetooth. Adding BrcmPatchRAM3.kext is not really an option for our end-users (on a web application).

laris, 2023-10-09:
report issue here.
macOS 14.0 sonoma + kext v2.3.0
connect bt mice and bt kbd, after open AirPods (gen1) and connect to macOS, the bt crash
crash log here

[sudo-dmesg-grep-i-bluetooth.txt](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/12847485/sudo-dmesg-grep-i-bluetooth.txt)

laris, 2023-10-09:
I swap with below kext and works fine now.
my mx master bt still cannot connect
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/446

lzipant, 2023-10-16:
> Same issue here - only difference is that I have an Intel AX201

same ax201, have you solved it? I tried to add BrcmPatchRAM3.kext and nothing works!

lzipant, 2023-10-16:
@ShroomKing Besides BrcmPatchRAM3.kext, does any other ktext need to be added? I add BrcmPatchRAM3.kext for my ax201 but it don't work. 
[image]

alikaylan, 2023-10-20:
Intel 7265NGW Bluetooth side does not work either in Sonoma. All was good before. WiFi works fine.

0xfeedface1993, 2023-10-20:
Intel AX200 Bluetooth not working in Sonoma, but  WiFi works fine.

System Information screenshot:

[image]

[image]

Hacktoshtool can't detect any bluetooth loaded:

[image]

`OpenCore` log:
```sh
02:011 00:000 OC: Prelinked injection AGPMInjector.kext () - Success
02:012 00:000 OC: Prelinked injection AGPMInjector.kext v111.1.18
02:018 00:005 OCAK: Local relocs 314 on FFFFFF8003F74000
02:019 00:001 OC: Prelinked injection BlueToolFixup.kext (Bluetooth support for macOS 12-) - Success
02:019 00:000 OC: Prelinked injection BlueToolFixup.kext v2.6.8
02:025 00:005 OCAK: Local relocs 499 on FFFFFF8003F7E000
02:027 00:001 OC: Prelinked injection BrcmFirmwareData.kext (Bluetooth firmware) - Success
02:027 00:000 OC: Prelinked injection BrcmFirmwareData.kext v2.6.8
02:030 00:002 OCAK: Patching invalid size 78FF4 with 1188000 for com.apple.iokit.IOUSBHostFamily
02:031 00:000 OCAK: Patching invalid size 4000 with 2504000 for com.apple.driver.AppleBusPowerController
02:033 00:001 OCAK: Patching invalid size 1C000 with 2110000 for com.apple.driver.AppleSMC
02:035 00:002 OCAK: Patching invalid size 2FFF with C12000 for com.apple.driver.watchdog
02:037 00:002 OCAK: Patching invalid size 2000 with 19B4000 for com.apple.iokit.IOACPIFamily
02:041 00:003 OCAK: Patching invalid size 36FF1 with 21F8000 for com.apple.driver.AppleMobileFileIntegrity
02:043 00:001 OCAK: Patching invalid size 9FF7 with 1E1A000 for com.apple.iokit.CoreAnalyticsFamily
02:047 00:003 OCAK: Patching invalid size 7BFD0 with CD1000 for com.apple.kec.corecrypto
02:048 00:001 OCAK: Patching invalid size 9FFA with 1D3E000 for com.apple.kext.CoreTrust
02:049 00:001 OCAK: Patching invalid size 1F000 with 2353000 for com.apple.security.AppleImage4
02:053 00:003 OCAK: Patching invalid size 3FFC with 1F1E000 for com.apple.driver.usb.AppleUSBCommon
02:056 00:003 OCAK: Dependency com.apple.driver.usb.AppleUSBHostPlatformProperties was not found for kext com.apple.iokit.IOUSBHostFamily
02:081 00:025 OCAK: Local relocs 357 on FFFFFF8004229000
02:083 00:001 OC: Prelinked injection BrcmPatchRAM3.kext (Bluetooth support for macOS 10.15-) - Success
02:084 00:000 OC: Prelinked injection BrcmPatchRAM3.kext v2.6.8
02:085 00:000 OC: Prelinked injection skips CryptexFixup.kext (Support for non-AVX2 CPUs in Ventura) kext at 18 due to version 220100 <= 230000 <= 229999
02:095 00:010 OCAK: Local relocs 729 on FFFFFF8004239000
02:097 00:001 OC: Prelinked injection IntelBluetoothFirmware.kext (Intel Bluetooth Firmware) - Success
02:098 00:000 OC: Prelinked injection IntelBluetoothFirmware.kext v2.3.0
02:104 00:006 OCAK: Local relocs 324 on FFFFFF8004990000
02:105 00:001 OC: Prelinked injection IntelBTPatcher.kext (IntelBTPatcher) - Success
02:106 00:000 OC: Prelinked injection IntelBTPatcher.kext v2.3.0
02:111 00:004 OCAK: Patching invalid size 22FF5 with D02000 for com.apple.driver.corecapture
02:122 00:011 OCAK: Patching invalid size 23BFF9 with 1C27000 for com.apple.iokit.IO80211Family
02:124 00:001 OCAK: Patching invalid size 17FF4 with 17A8000 for com.apple.iokit.IONetworkingFamily
02:133 00:009 OCAK: Patching invalid size 2EFF8 with 145A000 for com.apple.iokit.IOSkywalkFamily
02:135 00:001 OCAK: Patching invalid size 5000 with 154E000 for com.apple.driver.mDNSOffloadUserClient
02:252 00:116 OCAK: Local relocs 3218 on FFFFFF800499F000
02:257 00:005 OC: Prelinked injection AirportItlwm.kext (Intel Wi-Fi adaptor) - Success
02:258 00:000 OC: Prelinked injection AirportItlwm.kext v2.3.0
02:259 00:000 OC: Prelink size 92577792 kext offset 62226432 reserved 30408704
02:260 00:000 OCAK: KC TEXT is 65536 bytes with 62104 Mach-O headers need 848
02:276 00:016 OCAK: Forcing sb scheme on 230000
```

systemlog:
[https://mobile.magicbean.top/bluetooth.log](url)

andresiraola, 2023-10-21:
> Intel AX200 Bluetooth not working in Sonoma, but WiFi works fine.

I'm having the exact the same issue. I went back to Ventura in the meantime.

ShroomKing, 2023-10-24:
try adding the bluetoolfixup bootarg -btlfxallowanyaddr

0xfeedface1993, 2023-10-25:
> try adding the bluetoolfixup bootarg -btlfxallowanyaddr

It's still not functioning properly. While it seems that the kext isn't loading, the logs indicate that it has been loaded.

alikaylan, 2023-10-25:
> try adding the bluetoolfixup bootarg -btlfxallowanyaddr

Indeed, does not solve this particular issue.

ShroomKing, 2023-10-25:
Alright, it was just a guess based on the system info output for the bluetooth controller (Address = NULL)

I use AX211 and it works for me(Also works for @KaushikDuddala but idk which chip he is using)

Guess BrcmPatchRAM doesn’t work for all intel chips.

hugonascimento, 2023-10-26:
It worked for me in Sonoma with:
[https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/446#issuecomment-1729711030](url)

chinalujw, 2023-11-02:
My AX201 bluetooth address was null, After added BlueToolFixup.kext.zip, it's ok.
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/11672823/BlueToolFixup.kext.zip
Refer to https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/437#issuecomment-1579958516

sdkdimon, 2023-11-15:
Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/446).
[Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

Deihmos, 2023-11-24:
> Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(#446). [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

This works. Thanks.

Sturmkater, 2023-12-16:
I'm working on a fix for a day and think I miss something ovious, just too long in this topic. My problem is the Mac does not see any BT devices.

What I did so far, connecting a antena, trying out @Deihmos new Kexts, playing around with BrcmPatchRAM,...
`[    1.750504]: IntelFirmware: Driver init()
[    1.750510]: IntelFirmware: Driver Probe()
[    1.750514]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0033
[    1.750521]: IntelFirmware: Driver Start()
[    1.750547]: IntelFirmware: virtual bool BtIntel::initWithDevice(IOService *, IOUSBHostDevice *)
[    1.750550]: IntelFirmware: virtual bool USBDeviceController::init(IOService *, IOUSBHostDevice *)
[    1.750773]: IntelFirmware: virtual bool USBDeviceController::initConfiguration()
[    1.750782]: IntelFirmware: set configuration to 1
[    1.751175]: IntelFirmware: virtual bool USBDeviceController::findInterface()
[    1.751179]: IntelFirmware: Found interface!!!
[    1.751234]: IntelFirmware: virtual bool USBDeviceController::findPipes()
[    1.751238]: IntelFirmware: Found Interrupt endpoint!
[    1.751818]: IntelFirmware: Found Bulk out endpoint!
[    1.752245]: IntelFirmware: Found Bulk in endpoint!
[    1.752611]: IntelFirmware: BT init succeed
[    1.755385]: Sandbox: locationd(134) deny(1) mach-lookup com.apple.PowerManagement.controlIntelFirmware: Firmware timestamp 2021.39 buildtype 1 build 31878
[    1.757982]: IntelFirmware: Found device firmware ibt-1040-0041.sfi
[    1.761864]: IntelFirmware: Found device firmware: ibt-1040-0041.sfi
[    1.761901]: IntelFirmware: Boot Address: 0x100800
[    1.761903]: IntelFirmware: Firmware Version: 134-39.21
[    1.761904]: IntelFirmware: Firmware already loaded
[    1.762016]: IntelFirmware: Frimware is already running, finishing
[    1.763734]: IntelFirmware: Clean up...
[    1.763737]: IntelFirmware: virtual void BtIntel::free()
[    1.763739]: IntelFirmware: virtual void USBDeviceController::free()`
Sandbox message wasn't there from the start.
[image]
[image]
[image]

tuancuong92, 2024-01-30:
> Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(#446). [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

Your build saved my life. Now I can connect to my MX Mouse flawlessly.

Fagner3g, 2024-02-19:
> Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(#446). [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

Thanks, this solved my problem with mx master

eclipselu, 2024-05-10:
> Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(#446).
> [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

Thanks, I this solved my problem with sleep. 

###  My Spec

OS: Sonoma 14.4.1
Bluetooth:
```
  Bluetooth Controller:
  Address:	E4:60:17:CA:E0:F8
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v13446 c13446
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Problem:

When computer sleep and wake, bluetooth keyboard (Leopold FC750RBT) was disconnected, cannot unlock the machine in lock screen. Even if I use another usb keyboard to unlock the machine, bluetooth is disabled state and nothing works by toggling it, need to restart the machine to make it work. 

### What I did
Used the above IntelBluetoothFirmware kext above along with the latest BlueToolFixup, IntelBTPatcher kext. But still need a USB keyboard to unlock the machine as bluetooth is not available in lock screen. 

I then created a shortcut with Shortery to kill bluetoothd and run every time the machine wakes up from sleep, tried a few times, the bluetooth keyboard works now in lock screen.

alikaylan, 2024-08-08:
> Also my AX201 works in Sonoma. I've using latest BlueToolFixup kext, and myself build of IntelBluetoothFirmware kext with updated firmware and patch for LE devices(#446). [Kexts.zip](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/files/13364626/Kexts.zip)

sdkdimon, would you be able to kindly build the latest firmware version with the patch for LE devices? The only thing that seems to be working with MX mouse is the kext you had built with patch and it is no longer the latest firmware version. Thanks in advance. Much appreciated.

AOE33, 2024-09-25:
After many many experiments, my wifi module Intel 8260NGW with 2.4G/5 Ghz Wi-Fi and 867M Bluetooth 4.2NGFF M.2 fine worked with BlueToolFixup.kext, BrcmPatchRAM2.kext IntelBluetoothFirmware.kext and with NVRAM keys bluetoothInternalControllerInfo:000000000000000000000000000, bluetoothExternalDongleFailed:00 without IntelBTPatcher.kext

SherifMEldeeb, 2024-10-15:
BrcmPatchRAM3.kext did not work for me any ideas It's annoying.
CPU: Skylake Core i7-6500U
RAM: 16GB
Audio: ALC255
iGPU: Spoofed KBL 0x5916
Used itlwm + Heliport for wifi Intel AC3160
Bluetooth IntelFirmware + Blutoolfixup
MAC Os Sonoma 14.4.1
Thanks in advance

Vinhts, 2025-11-18:
<Sorry to bring this up, I actually have exactly the same keyboard pairing issue as you with the same AX201 and tried all the combinations I can think of but still no luck, hope someone with AX201 that has no issue in Sonoma can share what should be done there.>>>>>>
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/460#issuecomment-3448679720

Please read #51. It may help you. Bt ax201 with ble keyboard and mouse connect ok!

Vinhts, 2025-11-19:
_> I finally managed to build it by having MacKernelSDK and the Lilu I used in the EFI folder but seems still no luck with keyboard. Would you mind sharing what you have done to make keyboard working?_

Which macos do you use? Sonoma or Sequoia? I use Sonoma and Tahoe (I think Sequoia will be OK too). The key is the 2 kexts IntelBluetoothFirmware_FixLE_Tahoe.kext and IntelBTPatcher_FixLE_Tahoe.kext built from the above repo. Try to create a config with the order of kexts as in my picture.

[image]

emulatorchen, 2025-11-19:
> _> I finally managed to build it by having MacKernelSDK and the Lilu I used in the EFI folder but seems still no luck with keyboard. Would you mind sharing what you have done to make keyboard working?_
> 
> Which macos do you use? Sonoma or Sequoia? I use Sonoma and Tahoe (I think Sequoia will be OK too). The key is the 2 kexts IntelBluetoothFirmware_FixLE_Tahoe.kext and IntelBTPatcher_FixLE_Tahoe.kext built from the above repo. Try to create a config with the order of kexts as in my picture.
> 
> <img alt="Image" width="1019" height="956" src="https://private-user-images.githubusercontent.com/98160638/516056623-1aea14bd-4b05-4ebf-9c98-ad020de4236e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjM1MjY1NTAsIm5iZiI6MTc2MzUyNjI1MCwicGF0aCI6Ii85ODE2MDYzOC81MTYwNTY2MjMtMWFlYTE0YmQtNGIwNS00ZWJmLTljOTgtYWQwMjBkZTQyMzZlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTE5VDA0MjQxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ5ZWQ4MGNiY2M2ODJjNWNlNmQzOTQzMjVjODBjODc4YWJhZWQwZWU4ZmIzNDY5NGFhMWExMjg0MTU4NmNkNWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FZFYpwHMaLhGmtN4gCqd1p4Iw_S87AQcHaKEyj1edfk">

I am in Sequoia 15.7.2 now. Just want to clarify what you wanted me to do is:
* Clone this repo https://github.com/plooshi/IntelBluetoothFirmware
* Build it with Xcode 
* Copy both two kexts to replace existing one

Because I cannot find any prebuilt kexts in that repo and seems no such kexts with "_FixLE_" in that thread too.
I also tried to arrange the order as yours just I ignore the part of wifi as I don't really need it this moment and leave others as is but still no luck.

alexdesigncorp, 2025-11-29:
I've just updated my kexts, and can confirm it works with AX201 and Logitech MX Anywhere 3S on 15.7 Sonoma. Thanks a lot, much appreciated.

dariuxzy, 2025-12-02:
Still not working for me on Tahoe, I used these kext https://github.com/lshbluesky/IntelBluetoothFirmware + BlueToolFixup.kext, added NVRAM keys, but nothing. I'm trying to connect Logitech M196 Mouse. Thank for the help

emulatorchen, 2026-10-10:
> 30\. Still not working for me on Tahoe, I used these kext https://github.com/lshbluesky/IntelBluetoothFirmware + BlueToolFixup.kext, added NVRAM keys, but nothing. I'm trying to connect Logitech M196 Mouse. Thank for the help

I got my solution works with claude code(I don't know anything about that detail but it is also a patch to the above repo). You are free to pick it up.

[image]

Release: https://github.com/emulatorchen/IntelBluetoothFirmware/releases#release-2.5.2-dlc-fix

https://github.com/lshbluesky/IntelBluetoothFirmware/pull/1
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/pull/515