OpenIntelWireless/IntelBluetoothFirmware #461: 13.5.2 Cannot connect Airpods

syakiba, 2023-09-16:
### Have you read the docs?

No

### macOS Version

13.5.2

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

intel ac3165

### Description

bluetooth can be opened or closed, and it can connect the keyboard, but my two Airpods neight can be connected.

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	A0:C5:89:D5:2E:69
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v4096 c4096
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
nothing
```

TheCGCy, 2023-09-17:
Have you read the docs?
Yes

macOS Version
13.5.2

Kext Version
v2.3.0

Wireless Adapter Model and USB Product ID
Intel Wi-Fi AC 9462

Description
I have the same issue too, Airpods Pro 2 only works when using hands-free mode or 8000Hz sample rate. 48KHz won't output sound at all. It was after the 13.5.2 update, it worked for a few seconds, then glitched out with a robotic sound, and then total silence. Tried to forget the device and pair it again, no output ever since.

Info in System Report - Bluetooth
 ``` 
  Bluetooth Controller:
  Address:	A8:7E:EA:DE:2A:23
  State:	On
  Chipset:	THIRD_PARTY_DONGLE
  Discoverable:	Off
  Firmware Version:	v256 c256
  Supported Devices:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
  
 Connected:
  ----------
  Address:	----------
  RSSI:	-65
  Services:	0x400000 < BLE >
  Cy’s AirPods Pro:
  Address:	----------
  Vendor ID:	0x004C
  Product ID:	0x2014
  Left Battery Level:	100%
  Right Battery Level:	100%
  Case Version:	60.82.0
  Firmware Version:	6A301
  Minor Type:	Headphones
  RSSI:	-58
  Serial Number:	----------
  Serial Number (Left):	----------
  Serial Number (Right):	----------
  Services:	0x980019 < HFP AVRCP A2DP AACP GATT ACL >
  ```
 Relevant log output
 ``` nothing ``` 
 
Kernel Panic Logs
 ``` No response ```

gorouflex, 2023-09-17:
Try to add `BcrmPatchRam3.kext` , it worked for me

TheCGCy, 2023-09-17:
Hmm, doesn't work for me, it is connected like usual, but the sound still doesn't go through. 

With or without the BcrmPatchRam3.kext and BrcmBluetoothInjector.kext. The OS has to be playing sound for about 2~3 minutes for the AirPods to receive sound, but it's very delayed and robotic, flickering sound even. The Airpods play for a few seconds then go silent again. Changing playback apps also immediately stop sound transmission.

Interestingly, my left Airpod drains the battery quicker than the right one. Just a few minutes become L92%, R97%. The battery drain problem goes away when connected to the iPad. I think at this time, I better off using other earbuds.

ctopher7, 2023-09-19:
> Try to add `BcrmPatchRam3.kext` , it worked for me

works for me with ac9260 (airpods pro paired successfully) , but my logitech devices (mouse, keyboard) still having issue when pairing. It is discovered, connected, but not paired

do I need to apply another BcrmPatchRam ?

Tim-Masuda, 2023-09-21:
Airpods pro 2 also don't work, they connect, but there is no sound, the sound only works on 16 bits

Satori18282, 2023-09-21:
Airpods pro 2 do not work, the sound is issued only with 16 bits, please fix it please

Satori18282, 2023-09-21:
Airpods pro 2 dont work, sound is issued only with 16 bits, please fix it please

zxystd, 2024-03-11:
Not this repo driver issue.

closed 2024-03-11 by zxystd