OpenIntelWireless/IntelBluetoothFirmware #462: No audio output on Airpods Pro 2

BloodLancet, 2023-10-11:
After my airpods pro 2 firmware unintentionally updated to latest firmware, i got no audio output but detected in mac audio ouput selection. On earlier firmware the airpods just works fine. I've tried the alpha version Airportitlwm 2.3.0 and got similar result.
My device is Lenovo T580 on ventura 13.4 with factory default intel wifi card (8265/8275). i'm not planning to upgrade to sonoma because don't want to risk my daw and audio plugin compatibility.

Icee666, 2023-10-18:
I have the same problem on AX210 on ventura, still no fix?

Icee666, 2023-10-18:
I have same problem (Airpods pro 2) and AX210 on Mac 13.4

garishdp1502, 2023-11-16:
I have same problem with intel wifi card 8265. Ventura 13.6.1

wxxsfxyzm, 2023-12-13:
Accidentally found that when using Safari to play videos, the sound is just fine for Airpods, but not for other apps.

VincenzoXu, 2023-12-26:
Ｉhad the same problem with ax200 on Sonoma 14.2. Same as described above, normal in safari, abnormal in audio player.

VincenzoXu, 2023-12-26:
By the way,AirPods work fine.

monkeyunyun, 2024-01-12:
I have same problem (Airpods pro 2) and AC 9560 on Ventura 13.6.1
[intel蓝牙连接AirPods Pro无声音或者断断续续](urlhttps://bbs.pcbeta.com/forum.php?mod=viewthread&tid=1990931)

Robert-PF, 2024-01-14:
I have same problem , the  pods’s sound quality is terrible

dannyCaiHaoming, 2024-03-02:
The same problem, at first I thought it was the problem of my newly added usb hub and usb3.1 interface. But what I found later was that when I switched from the mac sound to the phone, I couldn't switch back to the mac properly. The first time the sound on mac is all right. But when we switch it back, it's completely lost.

dannyCaiHaoming, 2024-03-02:
> 同样的问题，一开始我以为是我新添加的usb hub和usb3.1接口的问题。但后来我发现，当我从mac声音切换到手机时，无法正常切换回mac。第一次在mac上声音还可以。但当我们把它切换回来时，它就完全丢失了。
Additional description: There is a problem with the music player, the web video playback is normal after switching back. It's embarrassing.

wan-qiyuan, 2024-03-04:
AirPods Pro 2 Sonoma14.3.1 AX200
only Singleside with 8 khz

dannyCaiHaoming, 2024-03-05:
> AirPods Pro 2 Sonoma14.3.1 AX200 仅单面 8 khz
My Intel network card is AX210, Bluetooth function via usb connection. The first time you connect, everything should be fine. But when the AirPods pro automatically switches to the phone and back to the mac, there's a problem

zxystd, 2024-03-11:
Not this repo drivers issue.

Robert-PF, 2024-03-11:
who can provide the repo drivers issue? where we can find it?

geitir, 2024-04-02:
Same issue. I also have OnePlus Buds Pro which default to 44.1 kHz and work fine. I believe the 48 kHz is the cause of the problem, but AirPods Pro 2 don't support 44.1. 

Can you tell us what the likely root cause is if you know that this repo is not the cause?

JonaUCB, 2024-05-16:
Facing the exact same problems, sometimes it even works while playing videos in the browser. Never works on spotify.
Another addition to this "bug", when connected to airpods and trying to use the microphone, the bluetooth crashes entirely.

kidxiang, 2024-06-02:
> Accidentally did something which temporarily make 48khz work:
> 
> 1. Open Safari and play a video, no sound will be delivered through AirPods Pro 2
> 2. Open Audio MIDI Setting, set AirPods' output format to "1 channel 16bit 8khz", now you will hear the sound at call quality
> 3. while keep video playing, change format back to "2 channel 32bit 48khz", you can hear the sound normally in a few seconds.
> 4. Now open some audio app they should work as well. Now you can close Safari.
>    Note that when all audio session is ended, this trick will no longer work, next time you need to hear a sound, you should do it all over again. Still can't figure out why it will happen.

This method does work, but it’s troublesome to do it every time.

wxxsfxyzm, 2024-06-11:
Find another workaround is to use Roon to play local music. Since the problem lies in system side, Roon has a private audio framework called RAAT which cloud bypass system audio framework to send audio to audio devices.
[image]
I believe any music apps have this ability can work flawlessly, I only use Roon for now. Although this is not a simple and easy method and it only works for local music, at least it is functional and may be a temporary workaround for those who wants to listen to local music. In the meantime, I hope this can provide everyone with new ideas.
Based on my testing, the multi-device connection between Roon playing music and the iPhone, music switching, and the conversation awareness feature on the Mac side all work properly.

ThxGiving, 2024-07-03:
Much easier workaround:

If your hardware uses Bluetooth 4.0 instead of 4.2 or above, this solution is highly likely to work for you.

Open System Preferences > Sound > Input

Change sound input from Airpods to Internal Microphone, when Airpods 2 connected.

VincenzoXu, 2024-08-23:
钟大真的不准备修复吗，不从驱动上解决的话，有别的可行的解决办法吗，上面提到的切换还是太过繁琐了些

Cabedon, 2025-01-07:
I used this command sudo nvram bluetoothHostControllerSwitchBehavior=always. This helped me get back 48kz audio. Not perfect, and sometimes I need to toggle them on off, or re-pair them, but it fixed audio to my AirPods Pro2. Apple Music tested only.
[image]

mihavo, 2025-01-17:
@Cabedon I tried this on AX210 on Sequoia and unfortunately there's still no audio output. When opening a video on Safari  audio works flawlessly, however switching to  any other app will keep the audio working for a few minutes and then when the audio session closes it stops working. The bluetoothHostControllerSwitchBehavior flag enables the external bluetooth adapter, correct?

mebaddy, 2025-02-03:
临时解决方案，借助 “BlackHole” 让系统可以正常输出 48.0KHZ。
[https://github.com/ExistentialAudio/BlackHole](url) 
下载安装好 “BlackHole 2ch ”，在 “音频MIDI设置” 创建 “多输出设备”，将 
“BlackHole 2ch ”跟 “AirPods Pro” 添加进去，主设备为 “BlackHole 2ch ”，采样率为48.0KHZ，如图所示。

如果没有声音：

方法1.请在菜单栏的喇叭📢图标里尝试切换到   其他输出设备（例如切换到“BlackHole 2ch ”或者“你的AirPods Pro” ），然后切回到我们创建的 “多输出设备”。

方法2.请在 “音频MIDI设置” 软件里面修改我们创建的  “多输出设备” 采样率为44.1kHZ，你会听到几秒的声音，然后再切回48.0KHZ。

以上两种方法可以同时使用，一般切换几次 AirPods Pro 就可以正常工作（最终的输出设备选择你创建的 “多输出设备”，并不是AirPods Pro）。 每次重新连接后都需要切换一下。

[image]

[image]

AjaxR0, 2025-11-03:
Not gonna open another issue for this but the newly released AirPods Pro Gen 3 have this issue, none of the workarounds work except have poor quality with call quality audio by changing the sample rate in MIDI setup.

hexxyan, 2026-04-08:
This issue should now be resolved.
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/462#issuecomment-4198992107

Could everyone who participated here please help test and confirm whether it is fixed on your side?

@BloodLancet @AjaxR0 @Cabedon @Icee666 @JonaUCB @Rallo00 @Robert-PF @ThxGiving @VincenzoXu @dannyCaiHaoming @garishdp1502 @geitir @hexxyan @isKoi @kidxiang @martinhung1002 @mebaddy @mihavo @monkeyunyun @sambow23 @wan-qiyuan @wxxsfxyzm @zxystd

ia74, 2026-10-11:
Agree, bluetooth works on my machine! Instant connect + instant audio playback