OpenIntelWireless/IntelBluetoothFirmware #465: AX200 even could not open the BT toggle

echokk11, 2023-11-23:
### Have you read the docs?

Yes

### macOS Version

macOS 14.1.1 

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

AX200 (0x004C)

### Description

[image]
[image]
[image]

### Info in System Report - Bluetooth

```shell
Bluetooth USB Host Controller:

  Product ID:	0x0029
  Vendor ID:	0x8087  (Intel Corporation)
  Version:	0.01
  Speed:	Up to 12 Mb/s
  Location ID:	0x20100000 / 2
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
  Built-In:	Yes

  Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
[    0.855245]: IOUSBHostDevice@20100000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

LucaBarrella, 2023-11-27:
Change the order of bluetoolfixup and the firmware.

echokk11, 2023-11-28:
> Change the order of bluetoolfixup and the firmware.

nothing change

echokk11, 2023-11-28:
normally after clean nvram