OpenIntelWireless/IntelBluetoothFirmware #469: AX200 bluetooth not working on Ventura 13.6.3

agustinusst, 2024-01-03:
### Have you read the docs?

Yes

### macOS Version

macOS 13.6.3

### Kext Version

v2.3.0

### Wireless Adapter Model and USB Product ID

intel ax200 (0x004C)

### Description

wifi is good, but bt is not work.

[image]

[image]

[image]

[image]

[image]

[image]

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:	NULL
  State:	Off
  Chipset:	BCM_4350C2
  Discoverable:	Off
  Firmware Version:	v0 c0
  Supported services:	0x382039 < HFP AVRCP A2DP HID Braille AACP GATT SerialPort >
  Transport:	USB
  Vendor ID:	0x004C (Apple)
```

### Relevant log output

```shell
IntelBluetoothFirmware is not found
```

SvenShi, 2024-01-12:
maybe you can try remove BrcmPatchRAM3.kext

za-creature, 2024-01-18:
Can confirm this, on both an older stack (~Jan 2023) and the latest acidanthera (as of today)

Might be related to https://github.com/skysafe/reblog/tree/main/cve-2023-45866 in which case it will presumably affect all supported MacOS versions eventually

zxystd, 2024-01-28:
Confirmed it is working.

visigaurav-oss, 2026-10-02:
It's the same card, same macOS version, same kext version, and the same symptoms. whats the solution