OpenIntelWireless/IntelBluetoothFirmware #470: Intel 7265 not working even tho its supported

zett1201, 2024-01-06:
### Have you read the docs?

Yes

### macOS Version

macOS 13

### Kext Version

The Latest One Available

### Wireless Adapter Model and USB Product ID

Intel (R) Dual-Band Wireless-AC 7265

### Description

Internet doesnt work even tho the driver is supported not even ubuntu can help.. (deactivating bluetooth)

### Info in System Report - Bluetooth

```shell
Activate switch exists but i cant activate it
```

### Relevant log output

```shell
I dont have internet (im on a ipad rn)
```

### Kernel Panic Logs

```shell
Again. I dont have internet
```

zett1201, 2024-01-06:
It says no variant specified, falling back to release

zett1201, 2024-01-06:
When uwing sudo wdutil it says *no wifi hardware installed*

Robert-PF, 2024-01-13:
I have a  problem on Intel 7265!
My Bluetooth connection to my airpos pro 2 sound intermittent, and with the signal is very bad, connected Bluetooth keyboard is still able to use. I try V2.1.0 ,there still have the save problem.
how to solve this, someone can help me?

Robert-PF, 2024-01-13:
> I have a problem on Intel 7265! My Bluetooth connection to my airpos pro 2 sound intermittent, and with the signal is very bad, connected Bluetooth keyboard is still able to use. I try V2.1.0 ,there still have the save problem. how to solve this, someone can help me?

[image]

the blance (“平衡”) appear ,the voice is very diffrent;some time I don‘t tun on the  voice of system-config ,there has no voice on my airpos

zxystd, 2024-01-28:
This is Intel bluetooth firmware driver repository.

closed 2024-01-28 by zxystd