OpenIntelWireless/IntelBluetoothFirmware #471: Intel AC 7265 Dual Band Bluetooth Doesn't Work in Sonoma 14.2.1

momenabdulrazek, 2024-01-25:
### Have you read the docs?

Yes

### macOS Version

macOS 14.2.1

### Kext Version

v2.4.0 Dortania Builds Version

### Wireless Adapter Model and USB Product ID

Intel (R) Dual-Band Wireless-AC 7265

### Description

Although of Adding Intel Bluetooth Firmware , Intel BTpatcher & BluetoolFixup .. Bluetooth Doesn't Work at all only Wi-Fi works with Airportitlwm Alpha Version v2.3.0 I'm running macOS Sonoma 14.2.1 
Tried to Swap Airportitlwm with itlwm and still bluetooth doesn't work 
USB mapped with USBToolBOX and Bluetooth Appears as USB device with no problems (related to mapping)

### Info in System Report - Bluetooth

```shell
Bluetooth Controller:
  Address:    NULL
  State:    Off
  Chipset:    BCM_4350C2
  Discoverable:    On
  Vendor ID:    0x004C (Apple)
```

### Relevant log output

```shell
IOUSBHostDevice@14300000: IOUSBHostDevice::setConfigurationGated: IntelBluetoothFirmware selected configuration 1
```

### Kernel Panic Logs

```shell
There's No Panics At All.
```

zxystd, 2024-01-28:
Seems like a user mistake.

closed 2024-01-28 by zxystd

momenabdulrazek, 2024-01-28:
What do you mean with user mistake ?