OpenIntelWireless/IntelBluetoothFirmware #474: Bluetooth is not working properly in Sonoma 14.2.1

xiaowenqiluo, 2024-02-17:
### Have you read the docs?

Yes

### macOS Version

mac os 14.2.1

### Kext Version

v.2.2.3

### Wireless Adapter Model and USB Product ID

intel AX101(0x8086,0x244)

### Description

wifi normal work. However, Bluetooth is not working properly, and the message shown in the figure is displayed.
[image]
[image]

### Info in System Report - Bluetooth

```shell
None yet
```

### Relevant log output

```shell
None yet
```

### Kernel Panic Logs

```shell
None yet
```

xiaowenqiluo, 2024-02-17:
sorry，The kernel version is v2.3.0

bathIncense, 2024-02-18:
[image]

closed 2024-02-20 by zxystd